LudovicRousseau/CCID #1: Too many file using softlink

FeitianSmartcardReader, 2015-04-26:
Hi Ludovic

I try to build CCID driver, it seems missing to many files, please check and submit to github as well.

11490059 lrwxr-xr-x   1 Ben  staff    26B Apr 25 14:19 parser.h -> ../../../PCSC/src/parser.h
11490060 -rw-r--r--   1 Ben  staff   439B Apr 25 14:19 reader.conf.in
11490061 lrwxr-xr-x   1 Ben  staff    28B Apr 25 14:19 simclist.c -> ../../../PCSC/src/simclist.c
11490062 lrwxr-xr-x   1 Ben  staff    28B Apr 25 14:19 simclist.h -> ../../../PCSC/src/simclist.h
11490063 lrwxr-xr-x   1 Ben  staff    27B Apr 25 14:19 strlcpy.c -> ../../../PCSC/src/strlcpy.c
11490064 lrwxr-xr-x   1 Ben  staff    30B Apr 25 14:19 strlcpycat.h -> ../../../PCSC/src/strlcpycat.h
11490065 lrwxr-xr-x   1 Ben  staff    31B Apr 25 14:19 tokenparser.l -> ../../../PCSC/src/tokenparser.l
11490048 lrwxr-xr-x   1 Ben  staff    24B Apr 25 14:19 misc.h -> ../../../PCSC/src/misc.h
11489673 lrwxr-xr-x  1 Ben  staff    33B Apr 25 14:19 debuglog.h -> ../../../PCSC/src/PCSC/debuglog.h
11489674 lrwxr-xr-x  1 Ben  staff    35B Apr 25 14:19 ifdhandler.h -> ../../../PCSC/src/PCSC/ifdhandler.h
11489675 -rw-r--r--  1 Ben  staff    54B Apr 25 14:19 pcsclite.h
11489676 lrwxr-xr-x  1 Ben  staff    31B Apr 25 14:19 reader.h -> ../../../PCSC/src/PCSC/reader.h
11489673 lrwxr-xr-x  1 Ben  staff    33B Apr 25 14:19 debuglog.h -> ../../../PCSC/src/PCSC/debuglog.h
11489674 lrwxr-xr-x  1 Ben  staff    35B Apr 25 14:19 ifdhandler.h -> ../../../PCSC/src/PCSC/ifdhandler.h

LudovicRousseau, 2015-04-26:
Exact. I will fix that.

LudovicRousseau, 2015-04-26:
Should be fixed with https://github.com/LudovicRousseau/CCID/commit/c9d8a914754ab52ecee7ec7732f4fd0a6c8bf299

Follow the build instructions at https://pcsclite.alioth.debian.org/ccid.html

closed 2015-04-26 by LudovicRousseau

FeitianSmartcardReader, 2015-04-29:
> Should be fixed with c9d8a91 <https://github.com/LudovicRousseau/CCID/commit/c9d8a914754ab52ecee7ec7732f4fd0a6c8bf299>
> Follow the build instructions at https://pcsclite.alioth.debian.org/ccid.html <https://pcsclite.alioth.debian.org/ccid.html>

Hi 

It seems not fix..

I tried new one, doesn’t work

Best Regard

Ben |Technical Specialist (GLBAL SUPPORT)
Skype: Hooben@FEITIAN | Tel: +86 (10) 62304466 ext. 575
Feitian, Inc. Floor 17th, Tower B, Huizhi Mansion, No.9 Xueqing Road, Haidian District Beijing 100085, P.R China
Smartcard Readers->http://www.ftsafe.com/product/Smart_Reader <http://www.ftsafe.com/product/Smart_Reader>
Online Shop->http://www.ftsafe.com/product/reader_shop <http://www.ftsafe.com/product/reader_shop>

This email message (including any attachments) is for the sole use of the intended recipient and may contain confidential and privileged information. Any unauthorized review, use, disclosure or distribution of this information is prohibited. If you are not the intended recipient, please contact the sender by reply email and destroy the original message and all of its copies.

LudovicRousseau, 2015-04-29:
Please provide a trace of:
- how you fetched the source code
- what error your get

FeitianSmartcardReader, 2015-04-29:
Hi

I using
git clone https://github.com/LudovicRousseau/CCID <https://github.com/LudovicRousseau/CCID>
And 
cd CCID

and do check the file in MacOSX

ls -ilh MacOSX

12083914 lrwxr-xr-x  1 Ben  staff    28B Apr 30 04:16 reader.h -> ../../PCSC/src/PCSC/reader.h
12083912 lrwxr-xr-x  1 Ben  staff    32B Apr 30 04:16 ifdhandler.h -> ../../PCSC/src/PCSC/ifdhandler.h

Did I do something wrong? I always using this way to download code from github.
Best Regard

Ben |Technical Specialist (GLBAL SUPPORT)
Skype: Hooben@FEITIAN | Tel: +86 (10) 62304466 ext. 575
Feitian, Inc. Floor 17th, Tower B, Huizhi Mansion, No.9 Xueqing Road, Haidian District Beijing 100085, P.R China
Smartcard Readers->http://www.ftsafe.com/product/Smart_Reader <http://www.ftsafe.com/product/Smart_Reader>
Online Shop->http://www.ftsafe.com/product/reader_shop <http://www.ftsafe.com/product/reader_shop>

This email message (including any attachments) is for the sole use of the intended recipient and may contain confidential and privileged information. Any unauthorized review, use, disclosure or distribution of this information is prohibited. If you are not the intended recipient, please contact the sender by reply email and destroy the original message and all of its copies.

LudovicRousseau, 2015-04-30:
Please follow the instructions at https://pcsclite.alioth.debian.org/ccid.html#download