LudovicRousseau/CCID #3: Use submodules instead of hardcoded symlinks

martinpaljak, 2015-05-02:
I have "~/projects" that usually is a huge flat folder with various development projects and/or repositories. Right now CCID instructions suggest to clone "contrib" which is a generic name and a conflict. While it could be an easy workaround to have a separate subfolder for PC/SC+CCID, why not make PCSC a submodule for CCID so that a single `git clone --recursive` would work ?

LudovicRousseau, 2015-05-02:
I don't think submodules will work for me.

The official repos (for me) are at ``ssh://git.debian.org/git/pcsclite/*.git`` but I also want to be able to use repos at github. So I can't use a specific reposority name for a submodules.

Maybe you can create a ``pcsclite/`` directory and clone the different needed project inside?

martinpaljak, 2015-05-02:
The trick is "relative submodules": http://blog.tremily.us/posts/Relative_submodules/

LudovicRousseau, 2015-05-02:
Nice. I will have a look.

LudovicRousseau, 2015-05-10:
Fixed in 28a44ae58b5adc1508b78de58686aa64cb4e1f17

closed 2015-05-10 by LudovicRousseau