LudovicRousseau/CCID #8: Build ccid driver on yosemite got error

FeitianSmartcardReader, 2015-07-31:
error:
sh-3.2# ./MacOSX/configure 

*************************
No Universal Binary build
*************************

+ ./configure 'CFLAGS= -DRESPONSECODE_DEFINED_IN_WINTYPES_H' PCSC_CFLAGS=-I/Users/Ben/Documents/sourcecode/ludovic.ccid/CCID/MacOSX 'PCSC_LIBS=-framework PCSC' 'LIBUSB_CFLAGS=-I/usr/local/include/libusb-1.0 ' 'LIBUSB_LIBS=-L/usr/local/lib -lusb-1.0 -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation ' LDFLAGS= --enable-usbdropdir=/usr/libexec/SmartCardServices/drivers --disable-dependency-tracking --enable-syslog --disable-static --disable-pcsclite --enable-composite-as-multislot
configure: WARNING: unrecognized options: --disable-dependency-tracking, --disable-static
./configure: line 2229: syntax error near unexpected token `1.8'
./configure: line 2229: `AM_INIT_AUTOMAKE(1.8 dist-bzip2 no-dist-gzip subdir-objects)'
+ rm -f src/Info.plist
sh-3.2# 

sh-3.2# uname -a
Darwin localhost 14.3.0 Darwin Kernel Version 14.3.0: Mon Mar 23 11:59:05 PDT 2015; root:xnu-2782.20.48~5/RELEASE_X86_64 x86_64
sh-3.2#

LudovicRousseau, 2015-08-01:
what version of the CCID driver?

Have you generated the configure script yourself?

FeitianSmartcardReader, 2015-08-02:
The latst version. I download using "git clone https://github.com/LudovicRousseau/CCID"
./bootstrap
and run "./MacOSX/configure", get error

My system is yosemite

LudovicRousseau, 2015-08-02:
I guess ./bootstrap reported an error.
It is a autoconf/automake/etc issue.

FeitianSmartcardReader, 2015-08-02:
right, i saw some errors, let me check and try to solve these errors, thanks, Luc

Best Regard 

Ben |Technical Specialist (GLBAL SUPPORT) | 国际部 王红宾
Skype: Hooben@FEITIAN | Tel: +86 (10) 62304466 ext. 575
Feitian, Inc. Floor 17th, Tower B, Huizhi Mansion, No.9 Xueqing Road, Haidian District Beijing 100085, P.R China
Smartcard Readers->http://www.ftsafe.com/product/Smart_Reader
Online Shop->http://www.ftsafe.com/product/reader_shop

This email message (including any attachments) is for the sole use of the intended recipient and may contain confidential and privileged information. Any unauthorized review, use, disclosure or distribution of this information is prohibited. If you are not the intended recipient, please contact the sender by reply email and destroy the original message and all of its copies.

From: Ludovic Rousseau
Date: 2015-08-02 17:56
To: LudovicRousseau/CCID
CC: Feitian Technologies
Subject: Re: [CCID] Build ccid driver on yosemite got error (#8)
I guess ./bootstrap reported an error.
It is a autoconf/automake/etc issue.

LudovicRousseau, 2015-08-27:
I guess this problem is solved on your side now. Closing.

closed 2015-08-27 by LudovicRousseau