LudovicRousseau/CCID #9: Watchdata Usb Key not working

giroberto, 2015-08-26:
Tried with latest ccid 1.4.20 on archlinux

pcsc_scan log:
```
04677683 hotplug_libudev.c:614:HPEstablishUSBNotifications() USB Device add
00000230 hotplug_libudev.c:296:get_driver() Looking for a driver for VID: 0x163C, PID: 0x0407, path: /dev/bus/usb/004/003
00000013 hotplug_libudev.c:435:HPAddDevice() Adding USB device: Watchdata USB Key
00000111 readerfactory.c:1012:RFInitializeReader() Attempting startup of Watchdata USB Key [USB CD-ROM] 00 00 using /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000223 readerfactory.c:897:RFBindFunctions() Loading IFD Handler 3.0
00000028 ifdhandler.c:1927:init_driver() Driver version: 1.4.20
00000627 ifdhandler.c:1944:init_driver() LogLevel: 0x0003
00000008 ifdhandler.c:1955:init_driver() DriverOptions: 0x0000
00000161 ifdhandler.c:96:CreateChannelByNameOrChannel() Lun: 0, device: usb:163c/0407:libudev:0:/dev/bus/usb/004/003
00000019 ccid_usb.c:283:OpenUSBByName() Using: /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist
00000581 ccid_usb.c:301:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000007 ccid_usb.c:302:OpenUSBByName() ifdProductString: Generic CCID driver
00000005 ccid_usb.c:303:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00005850 ccid_usb.c:573:OpenUSBByName() Found interface 0 but expecting 1
00000025 ccid_usb.c:575:OpenUSBByName() Wrong interface for USB device 4/3. Checking next one.
00000049 ccid_usb.c:190:close_libusb_if_needed() libusb_exit
00000276 ccid_usb.c:702:OpenUSBByName() Device not found?
00000015 ifdhandler.c:130:CreateChannelByNameOrChannel() failed
00000008 readerfactory.c:1043:RFInitializeReader() Open Port 0x200000 Failed (usb:163c/0407:libudev:0:/dev/bus/usb/004/003)
00000005 readerfactory.c:335:RFAddReader() Watchdata USB Key [USB CD-ROM] init failed.
00000008 readerfactory.c:558:RFRemoveReader() UnrefReader() count was: 1
00000005 readerfactory.c:1063:RFUnInitializeReader() Attempting shutdown of Watchdata USB Key [USB CD-ROM] 00 00.
00000003 readerfactory.c:934:RFUnloadReader() Unloading reader driver.
00000103 hotplug_libudev.c:614:HPEstablishUSBNotifications() USB Device add
00000157 hotplug_libudev.c:296:get_driver() Looking for a driver for VID: 0x163C, PID: 0x0407, path: /dev/bus/usb/004/003
00000009 hotplug_libudev.c:435:HPAddDevice() Adding USB device: Watchdata USB Key
00000058 readerfactory.c:1012:RFInitializeReader() Attempting startup of Watchdata USB Key [USB CCID Key] 00 00 using /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000175 readerfactory.c:897:RFBindFunctions() Loading IFD Handler 3.0
00000025 ifdhandler.c:1927:init_driver() Driver version: 1.4.20
00000563 ifdhandler.c:1944:init_driver() LogLevel: 0x0003
00000008 ifdhandler.c:1955:init_driver() DriverOptions: 0x0000
00000164 ifdhandler.c:96:CreateChannelByNameOrChannel() Lun: 0, device: usb:163c/0407:libudev:1:/dev/bus/usb/004/003
00000012 ccid_usb.c:283:OpenUSBByName() Using: /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist
00000615 ccid_usb.c:301:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000006 ccid_usb.c:302:OpenUSBByName() ifdProductString: Generic CCID driver
00000005 ccid_usb.c:303:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00005086 ccid_usb.c:588:OpenUSBByName() Can't claim interface 4/3: -6
00000028 ccid_usb.c:190:close_libusb_if_needed() libusb_exit
00000094 ifdhandler.c:130:CreateChannelByNameOrChannel() failed
00000013 readerfactory.c:1043:RFInitializeReader() Open Port 0x200001 Failed (usb:163c/0407:libudev:1:/dev/bus/usb/004/003)
00000005 readerfactory.c:335:RFAddReader() Watchdata USB Key [USB CCID Key] init failed.
00000005 readerfactory.c:558:RFRemoveReader() UnrefReader() count was: 1
00000004 readerfactory.c:1063:RFUnInitializeReader() Attempting shutdown of Watchdata USB Key [USB CCID Key] 00 00.
00000004 readerfactory.c:934:RFUnloadReader() Unloading reader driver.
00000047 hotplug_libudev.c:501:HPAddDevice() Failed adding USB device: Watchdata USB Key
```

LudovicRousseau, 2015-08-27:
```
00005086 ccid_usb.c:588:OpenUSBByName() Can't claim interface 4/3: -6
```
-6 is LIBUSB_ERROR_BUSY

Your device CCID interface is *already* used by another program. You have to stop this other program first.

giroberto, 2015-08-27:
Thank you! The wdtoken module still was on my system, now i got this

```
PC/SC device scanner
V 1.4.23 (c) 2001-2011, Ludovic Rousseau <ludovic.rousseau@free.fr>
Compiled with PC/SC lite version: 1.8.12
Using reader plug'n play mechanism
Scanning present readers...
0: Watchdata USB Key [USB CCID Key] 00 00

Thu Aug 27 10:19:02 2015
Reader 0: Watchdata USB Key [USB CCID Key] 00 00
  Card state: Card inserted, 
  ATR: 3B 6E 00 00 57 44 36 19 69 86 93 02 21 18 15 50 0E 2B

ATR: 3B 6E 00 00 57 44 36 19 69 86 93 02 21 18 15 50 0E 2B
+ TS = 3B --> Direct Convention
+ T0 = 6E, Y(1): 0110, K: 14 (historical bytes)
  TB(1) = 00 --> VPP is not electrically connected
  TC(1) = 00 --> Extra guard time: 0
+ Historical bytes: 57 44 36 19 69 86 93 02 21 18 15 50 0E 2B
  Category indicator byte: 57 (proprietary format)

Possibly identified card (using /home/umberto/.cache/smartcard_list.txt):
        NONE

Updating /home/umberto/.cache/smartcard_list.txt using http://ludovic.rousseau.free.fr/softwares/pcsc-tools/smartcard_list.txt
--2015-08-27 10:19:02--  http://ludovic.rousseau.free.fr/softwares/pcsc-tools/smartcard_list.txt
Resolvendo ludovic.rousseau.free.fr (ludovic.rousseau.free.fr)... 212.27.63.159
Conectando-se a ludovic.rousseau.free.fr (ludovic.rousseau.free.fr)|212.27.63.159|:80... conectado.
A requisição HTTP foi enviada, aguardando resposta... 200 OK
Tamanho: 228069 (223K) [text/plain]
Salvando em: “/home/umberto/.cache/smartcard_list.txt”

/home/umberto/.cache/smartcard_list.txt      100%[=============================================================================================>] 222,72K  --.-KB/s   em 0,02s  

2015-08-27 10:19:04 (9,65 MB/s) - “/home/umberto/.cache/smartcard_list.txt” salvo [228069/228069]

Possibly identified card (using /home/umberto/.cache/smartcard_list.txt):
        NONE
```

LudovicRousseau, 2015-08-27:
Bug was on the user configuration.

closed 2015-08-27 by LudovicRousseau

YogeshSeralia, 2017-01-16:
Its possibly be **pn533** which blocks the access to reader for pcscd.

what I done was 

1) check if pn533 is loaded by :  
  `lsmod | grep pn533`

if your output is like : 

```
pn533                  40960  0 
nfc                   102400  1 pn533

```
2) If output is similar to above then unload **pn533** from kernel : 

`sudo modprobe -r pn533 pn533`

3) Now start your pcscd and then start pcsc_scan , it should now show your card status.