LudovicRousseau/CCID #11: --enable-udev switch mentioned in INSTALL, but not available in configure.ac

shearl, 2016-01-15:
The --enable-udev switch mentioned in the INSTALL file is not available from the configure script.
Perhaps the INSTALL file just needs to be updated?

But, if I use --disable-libusb, pcscd doesn't work. How do I make it use udev?

LudovicRousseau, 2016-01-15:
Fixed in https://github.com/LudovicRousseau/CCID/commit/43ccf87ea24ade4c5603e0e719383779fe7201bf

LudovicRousseau, 2016-01-15:
The CCID driver do not use udev anymore, only libusb.

pcsc-lite already uses libudev (and not udev)

closed 2016-01-15 by LudovicRousseau

shearl, 2016-01-15:
Ok. One more question. If the ccid driver doesn't use udev, why does ccid include src/92_pcscd_ccid.rules and tell you to copy it to /etc/udev/rules.d/?

(Sorry for my ignorance of the interaction between ccid & pcsc-lite.)

LudovicRousseau, 2016-01-15:
The CCID driver uses udev to allow selective suspend of the USB readers.
It just installs a configuration script. This file is not needed to make the software work. Maybe I should remove the red colour in the message (it was important at some time but not any more).