LudovicRousseau/CCID #12: Can't add id3 Semiconductors CL1356T device

mib4fun, 2016-01-25:
Hi,

i have installed ccid 1.4.20 on archlinux
logs :
<pre>
00000000 debuglog.c:291:DebugLogSetLevel() debug level=debug
00000206 configfile.l:283:DBGetReaderListDir() Parsing conf directory: /etc/reader.conf.d
00000026 configfile.l:355:DBGetReaderList() Parsing conf file: /etc/reader.conf.d/libccidtwin
00000019 configfile.l:355:DBGetReaderList() Parsing conf file: /etc/reader.conf.d/libccidtwin                                                 [20/1801]
00000040 configfile.l:317:DBGetReaderListDir() Skipping non regular file: ..
00000003 configfile.l:317:DBGetReaderListDir() Skipping non regular file: .
00000006 pcscdaemon.c:569:main() pcsc-lite 1.8.14 daemon ready.
00005480 hotplug_libudev.c:296:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/003/001
00000168 hotplug_libudev.c:296:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/003/001
00000157 hotplug_libudev.c:296:get_driver() Looking for a driver for VID: 0x0B81, PID: 0x0200, path: /dev/bus/usb/003/057
00000009 hotplug_libudev.c:435:HPAddDevice() Adding USB device: id3 Semiconductors CL1356T
00000066 readerfactory.c:1012:RFInitializeReader() Attempting startup of id3 Semiconductors CL1356T (01754179) 00 00 using /usr/lib/pcsc/drivers/ifd-cc
id.bundle/Contents/Linux/libccid.so
00000220 readerfactory.c:897:RFBindFunctions() Loading IFD Handler 3.0
00000032 ifdhandler.c:1927:init_driver() Driver version: 1.4.20
00000622 ifdhandler.c:1944:init_driver() LogLevel: 0x0003
00000006 ifdhandler.c:1955:init_driver() DriverOptions: 0x0000
00000137 ifdhandler.c:96:CreateChannelByNameOrChannel() Lun: 0, device: usb:0b81/0200:libudev:0:/dev/bus/usb/003/057
00000027 ccid_usb.c:283:OpenUSBByName() Using: /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist
00000637 ccid_usb.c:301:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000007 ccid_usb.c:302:OpenUSBByName() ifdProductString: Generic CCID driver
00000004 ccid_usb.c:303:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your 
option) any later version.
00081906 ccid_usb.c:595:OpenUSBByName() Found Vendor/Product: 0B81/0200 (id3 Semiconductors CL1356T)
00000016 ccid_usb.c:597:OpenUSBByName() Using USB bus/device: 3/57
00000286 ccid_usb.c:1125:get_data_rates() declared: 106000 bps
00000011 ccid_usb.c:1125:get_data_rates() declared: 212000 bps
00000003 ccid_usb.c:1125:get_data_rates() declared: 424000 bps
00000002 ccid_usb.c:1125:get_data_rates() declared: 847000 bps

61101765 ccid_usb.c:786:ReadUSB() read failed (3/57): -7 LIBUSB_ERROR_TIMEOUT
00100313 ccid_usb.c:786:ReadUSB() read failed (3/57): -7 LIBUSB_ERROR_TIMEOUT
00100263 ccid_usb.c:786:ReadUSB() read failed (3/57): -7 LIBUSB_ERROR_TIMEOUT
00000018 ifdhandler.c:171:CreateChannelByNameOrChannel() failed
00000010 readerfactory.c:1043:RFInitializeReader() Open Port 0x200000 Failed (usb:0b81/0200:libudev:0:/dev/bus/usb/003/057)
00000003 readerfactory.c:335:RFAddReader() id3 Semiconductors CL1356T (01754179) init failed.
00000007 readerfactory.c:558:RFRemoveReader() UnrefReader() count was: 1
00000003 readerfactory.c:1063:RFUnInitializeReader() Attempting shutdown of id3 Semiconductors CL1356T (01754179) 00 00.
00000003 readerfactory.c:934:RFUnloadReader() Unloading reader driver.
00000066 hotplug_libudev.c:507:HPAddDevice() Failed adding USB device: id3 Semiconductors CL1356T
</pre>

LudovicRousseau, 2016-01-26:
- It is reproducible?
- Does the reader works on another system (like Windows)?
- What is the output of parse as described in http://pcsclite.alioth.debian.org/ccid.html#CCID_compliant

mib4fun, 2016-01-26:
* yes
* yes, but he uses a commercial driver
* yes, it seems
<pre>
sudo ./src/parse > output.txt
Parsing USB bus/device: 0B81:0200 (bus 1, device 2)
 idVendor:  0x0B81  iManufacturer: id3 Semiconductors
 idProduct: 0x0200  iProduct: Contactless Reader
  Found a CCID/ICCD device at interface 0
Parsing USB bus/device: 1D6B:0001 (bus 1, device 1)
 idVendor:  0x1D6B  iManufacturer: Linux 3.16.0-4-amd64 ohci_hcd
 idProduct: 0x0001  iProduct: OHCI PCI host controller
  NOT a CCID/ICCD device
</pre>
 i have attached the file : 
[output.txt](https://github.com/LudovicRousseau/CCID/files/104784/output.txt)

LudovicRousseau, 2016-01-26:
Your reader uses an old firmware: 1.08 instead of 1.12 from http://pcsclite.alioth.debian.org/ccid/readers/id3_CL1356T.txt.
That is surprising since my entry is also quiet old, from 2010.

Maybe id3 Semiconductors provides a new firmware. I could not find anything on http://www.id3.eu/

mib4fun, 2016-01-26:
Wow, my device is that old.
Ok, thanks ! I will look for an upgrade, then.

closed 2016-01-26 by LudovicRousseau