LudovicRousseau/CCID #13: Can't get data from Token

giroberto, 2016-02-03:
Hello, I had reported while ago that I can't connect to my token using CCID and with your help I could make the computer recognize it, but now I can't get the certificate.
The ATR is not present on smartcard_list.txt, I added it manually to it.

Is a usb token from watchdata. I will attach a log using your code, versions
pcsc-perl 1.4.14-1
pcsc-tools 1.4.25-1
pcsclite 1.8.14-1
opensc 0.15.0-1
ccid 1.4.20-1 
and another working log using watchdata driver.

[log_pcsclite.txt](https://github.com/LudovicRousseau/CCID/files/115704/log_pcsclite.txt)
[log_wd.txt](https://github.com/LudovicRousseau/CCID/files/115712/log_wd.txt)

[parse_pcsclite.txt](https://github.com/LudovicRousseau/CCID/files/115714/parse_pcsclite.txt)
[parse_wd.txt](https://github.com/LudovicRousseau/CCID/files/115713/parse_wd.txt)

[getinfo_pcsclite.txt](https://github.com/LudovicRousseau/CCID/files/115715/getinfo_pcsclite.txt)
[getinfo_wd.txt](https://github.com/LudovicRousseau/CCID/files/115716/getinfo_wd.txt)

LudovicRousseau, 2016-02-03:
According to ``getinfo_wd.txt`` the token is correctly working at the PKCS#11 level.
It is not a CCID issue.

You should read PKCS#11 documentation or contact Watch data if is a bug in ``/usr/lib/watchdata/ICP/lib/libwdpkcs_icp.so``.

closed 2016-02-03 by LudovicRousseau

giroberto, 2016-02-03:
_wd logs are  not using your ccid,  are samples using watchdata supplied
driver

LudovicRousseau, 2016-02-03:
Why not use the watchdata driver if it works for you?

It looks like it is a fork or my CCID driver:
```
05007151 ccid_usb.c:697:OpenUSBByName() ifdManufacturerString: WatchData
05007154 ccid_usb.c:698:OpenUSBByName() ifdProductString: WatchData CCID driver
05007156 ccid_usb.c:699:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
```
Where can I get the driver?

giroberto, 2016-02-03:
Because I work developing a application using token authentication
accepting a variety o models and using the pcsclite will be easier to
maintain and deploy.

yes, it is a fork I believe
I attached the driver, but you can download fom
http://www.watchdata.com/brazil/watchkey/index.htm

LudovicRousseau, 2016-02-03:
I already contacted watchdata last year about this forked driver but got no answer.
Maybe you should NOT work with token providers like this that do not respect the software licenses.

giroberto, 2016-02-03:
You're right, thank you for your support.

alexandreaguiar, 2016-06-07:
Hi giroberto,
I facing the same problem with watchdata tokens, do find some solution?
Thanks

giroberto, 2016-06-07:
I did not found any solution using CCID, I had to keep using manufacturer's provided driver.

eferro70, 2016-06-09:
Watchdata driver works but it bug if there is a safenet aladdin driver installed. The bug occurs when the processes pcscd and pcscd_wd are running and watchdata token is inserted. If the pcscd is stoped everything gets ok.