LudovicRousseau/CCID #14: MacOS Yosemite Problem

gerdogdu, 2016-03-16:
Hi 
I have installed the MILKO SmartCard Reader via the attached sh script provider provided but smart card is not working. Can you help me how to fix it? 
When I opened the System Report it shows that 
MILKO v.2.4
Current Required (mA):	Unknown (Device has not been configured)**

Best
Gurkan
[Setup.txt](https://github.com/LudovicRousseau/CCID/files/176256/Setup.txt)

LudovicRousseau, 2016-03-16:
You should report the problem to the provider of the script.

Or you can follow https://pcsclite.alioth.debian.org/ccid.html#CCID_compliant

closed 2016-03-16 by LudovicRousseau

gerdogdu, 2016-03-16:
I downloaded source code but it gives the following error when configure:

configure: error: install pcsc-lite 1.8.3 or later, or use ./configure PCSC_CFLAGS=...

LudovicRousseau, 2016-03-16:
For Mac OS X use ``./MacOSX/configure``
see https://pcsclite.alioth.debian.org/ccid.html#MacOSX

gerdogdu, 2016-03-16:
Thanks but now I get the following error:
configure: error: libusb.h not found, install libusb or use ./configure LIBUSB_CFLAGS=...
+ rm -f src/Info.plist

gerdogdu, 2016-03-16:
I run parse, it gives the following output

Parsing USB bus/device: 2A17:0001 (bus 20, device 26)
 idVendor:  0x2A17  iManufacturer: udea
 idProduct: 0x0001  iProduct: MILKO V2.4
  Can't get config descriptor: -5

gerdogdu, 2016-03-16:
Do you have any comment on this?

LudovicRousseau, 2016-03-16:
My comment is that your device is not a valid CCID device.
You should use a supported reader https://pcsclite.alioth.debian.org/ccid/supported.html

gerdogdu, 2016-03-16:
But in release notes, it says support the Milko 
1.4.17 - 11 June 2014, Ludovic Rousseau
    - Add support of
      . udea MILKO V1.

LudovicRousseau, 2016-03-16:
Your device is "iProduct: MILKO V2.4".
The one from my list is "iProduct: MILKO V1." https://pcsclite.alioth.debian.org/ccid/readers/udea_MILKO_V1.txt

It looks like udea (the manufacturer) changed the firmware. I don't know if your version if older or newer than the one I have in the list.

gerdogdu, 2016-03-16:
How can I inspect logs ? Is there any log file to look further diagnose

gerdogdu, 2016-03-16:
I noticed that there also exists /usr/libexec/SmartCardServices/drivers/ifd-acsccid.bundle/Contents/Info.plist but in this file, no milko found. Can it be a problem?