LudovicRousseau/CCID #17: Incorporate the OSX installer ?

martinpaljak, 2016-05-03:
I'm 50/50 split between wanting to keep the overall installer outside of the CCID codebase vs wanting to build one straight from the CCID sources.

https://github.com/martinpaljak/osx-ccid-installer

LudovicRousseau, 2016-05-03:
I prefer to separate the source code from the packaging.
For example the source code does not contain the ``debian/`` directory or files to build the ``.rpm`` packages.

I do not plan to merge the OSX installer. But continue maintaining it :-)

closed 2016-05-03 by LudovicRousseau

martinpaljak, 2016-05-03:
OK, at least it is clear. At the same time, things like SDK target versions that somewhat closely are bound to the distribution, are currently conveniently in MacOSX folder. Is it a read-only area or OK to sync patches between the two ?

 (to get rid of this: https://github.com/martinpaljak/osx-ccid-installer/blob/master/ccid-patches/00_osx1011.patch) ?

Maybe you can add a link to somewhere for other OSX users in the README (or should I make a PR)?

By the way, what about moving the default readme to markdown for a "nicer" github page?

LudovicRousseau, 2016-05-03:
It may be simpler to just remove the possibility to build a Universal Binary driver. I have not build a UB CCID driver since a long time.
Unless you have a Universal Binary 32/64 bits libusb this line is not used. Or I am wrong?

Feel free to convert the Readme into a Readme.md to make it nice for github.