LudovicRousseau/CCID #23: VMWare's shared CCID reader doesn't work

mickflemm, 2016-10-22:
The behavior is a bit strange, it seems that some reads/writes succeed and others fail. For example pcsc_scan works and recognises the card but opensc doesn't work. Note that gnupg's scd internal ccid driver (https://github.com/mtigas/gnupg/blob/mtigas/scd/ccid-driver.c) works just fine so this is an issue with ccid and/or pcscd.

This is what I see in the logs:
ifdwrapper.c:369:IFDStatusICC() Card not transacted: 612
eventhandler.c:333:EHStatusHandlerThread() Error communicating to: VMware Virtual USB CCID 00 00
ccid_usb.c:797:WriteUSB() write failed (2/4): -4 LIBUSB_ERROR_NO_DEVICE
ccid_usb.c:836:ReadUSB() read failed (2/5): -7 LIBUSB_ERROR_TIMEOUT
ifdwrapper.c:369:IFDStatusICC() Card not transacted: 612
eventhandler.c:333:EHStatusHandlerThread() Error communicating to: VMware Virtual USB CCID 00 00
ccid_usb.c:836:ReadUSB() read failed (2/5): -1 LIBUSB_ERROR_IO
ifdwrapper.c:369:IFDStatusICC() Card not transacted: 612
eventhandler.c:333:EHStatusHandlerThread() Error communicating to: VMware Virtual USB CCID 00 00
ccid_usb.c:797:WriteUSB() write failed (2/5): -4 LIBUSB_ERROR_NO_DEVICE

Please let me know if you want me to test any patches or at least give me a hint on where to start debugging this.

Thanks for your time

LudovicRousseau, 2016-10-22:
These are errors reported by libusb.
Please follow https://pcsclite.alioth.debian.org/ccid.html#support to generate logs with more details.

mickflemm, 2016-10-22:
I attached the log as you said, below is the console output of the commands I issued. Note that I also recompiled gpg scd to use pcsc instead of the built-in ccid driver and it also fails so it's not due to card handling, it's something wrong with the reader driver.

Please let me know if I can help in any other way.

```
#pcsc_scan 
PC/SC device scanner
V 1.4.27 (c) 2001-2011, Ludovic Rousseau <ludovic.rousseau@free.fr>
Compiled with PC/SC lite version: 1.8.16
Using reader plug'n play mechanism
Scanning present readers...
0: VMware Virtual USB CCID 00 00

Sat Oct 22 13:02:22 2016
Reader 0: VMware Virtual USB CCID 00 00
  Card state: Card inserted, 
  ATR: 3B FC 13 00 00 81 31 FE 15 59 75 62 69 6B 65 79 4E 45 4F 72 33 E1

ATR: 3B FC 13 00 00 81 31 FE 15 59 75 62 69 6B 65 79 4E 45 4F 72 33 E1
+ TS = 3B --> Direct Convention
+ T0 = FC, Y(1): 1111, K: 12 (historical bytes)
  TA(1) = 13 --> Fi=372, Di=4, 93 cycles/ETU
    43010 bits/s at 4 MHz, fMax for Fi = 5 MHz => 53763 bits/s
  TB(1) = 00 --> VPP is not electrically connected
  TC(1) = 00 --> Extra guard time: 0
  TD(1) = 81 --> Y(i+1) = 1000, Protocol T = 1 
-----
  TD(2) = 31 --> Y(i+1) = 0011, Protocol T = 1 
-----
  TA(3) = FE --> IFSC: 254
  TB(3) = 15 --> Block Waiting Integer: 1 - Character Waiting Integer: 5
+ Historical bytes: 59 75 62 69 6B 65 79 4E 45 4F 72 33
  Category indicator byte: 59 (proprietary format)
+ TCK = E1 (correct checksum)

Possibly identified card (using /usr/share/pcsc/smartcard_list.txt):
3B FC 13 00 00 81 31 FE 15 59 75 62 69 6B 65 79 4E 45 4F 72 33 E1
	YubiKey NEO (PKI)
	http://www.yubico.com/
^C
# opensc-tool -l
# Detected readers (pcsc)
Nr.  Card  Features  Name
0    Yes             VMware Virtual USB CCID 00 00
# opensc-tool --list-algorithms
Using reader with a card: VMware Virtual USB CCID 00 00
Failed to connect to card: Unresponsive card (correctly inserted?)
# opensc-tool -a
Using reader with a card: VMware Virtual USB CCID 00 00
Failed to connect to card: Unresponsive card (correctly inserted?)
```

[log.txt](https://github.com/LudovicRousseau/CCID/files/545902/log.txt)

LudovicRousseau, 2016-10-22:
I can't find any error in your log file.

Maybe you should increase the debug level of opensc-tool.

mickflemm, 2016-10-22:
Seems like it detects the card is present but then it doesn't respond.

```
0x385b940e700 14:59:28.037 [opensc-tool] sc.c:259:sc_detect_card_presence: called
0x385b940e700 14:59:28.038 [opensc-tool] reader-pcsc.c:397:pcsc_detect_card_presence: called
0x385b940e700 14:59:28.038 [opensc-tool] reader-pcsc.c:303:refresh_attributes: VMware Virtual USB CCID 00 00 check
0x385b940e700 14:59:28.039 [opensc-tool] reader-pcsc.c:323:refresh_attributes: returning with: 0 (Success)
0x385b940e700 14:59:28.039 [opensc-tool] reader-pcsc.c:402:pcsc_detect_card_presence: returning with: 1
0x385b940e700 14:59:28.039 [opensc-tool] sc.c:264:sc_detect_card_presence: returning with: 1
Using reader with a card: VMware Virtual USB CCID 00 00
0x385b940e700 14:59:28.039 [opensc-tool] sc.c:259:sc_detect_card_presence: called
0x385b940e700 14:59:28.039 [opensc-tool] reader-pcsc.c:397:pcsc_detect_card_presence: called
0x385b940e700 14:59:28.039 [opensc-tool] reader-pcsc.c:303:refresh_attributes: VMware Virtual USB CCID 00 00 check
0x385b940e700 14:59:28.039 [opensc-tool] reader-pcsc.c:323:refresh_attributes: returning with: 0 (Success)
0x385b940e700 14:59:28.039 [opensc-tool] reader-pcsc.c:402:pcsc_detect_card_presence: returning with: 1
0x385b940e700 14:59:28.039 [opensc-tool] sc.c:264:sc_detect_card_presence: returning with: 1
Connecting to card in reader VMware Virtual USB CCID 00 00...
0x385b940e700 14:59:28.039 [opensc-tool] card.c:196:sc_connect_card: called
0x385b940e700 14:59:28.039 [opensc-tool] reader-pcsc.c:479:pcsc_connect: called
0x385b940e700 14:59:28.039 [opensc-tool] reader-pcsc.c:303:refresh_attributes: VMware Virtual USB CCID 00 00 check
0x385b940e700 14:59:28.040 [opensc-tool] reader-pcsc.c:323:refresh_attributes: returning with: 0 (Success)
0x385b940e700 14:59:28.084 [opensc-tool] reader-pcsc.c:501:pcsc_connect: VMware Virtual USB CCID 00 00:SCardConnect failed: 0x80100066
0x385b940e700 14:59:28.084 [opensc-tool] card.c:325:sc_connect_card: returning with: -1113 (Unresponsive card (correctly inserted?))
Failed to connect to card: Unresponsive card (correctly inserted?)
0x385b940e700 14:59:28.084 [opensc-tool] ctx.c:842:sc_release_context: called
0x385b940e700 14:59:28.084 [opensc-tool] reader-pcsc.c:794:pcsc_finish: called

```

LudovicRousseau, 2016-10-22:
I found the problem.
It is a known vmware bug. See the comment in https://pcsclite.alioth.debian.org/ccid/unsupported.html#0x0E0F0x0004

This vmware reader is in the "Unsupported or partly supported CCID readers" list.
I can't help. Report the problem to vmware.

closed 2016-10-22 by LudovicRousseau

mickflemm, 2016-10-22:
But it works with Gnupg's built-in ccid driver and it also works on windows. Anyway I'll try to debug this further and if possible come back with a patch or something. Thanks for your help.

Alex1130, 2017-03-16:
@mickflemm did you get anywhere with this issue?
I get on average each 100th signature
"The smart card has been reset, so any shared state information is invalid."
on average every 1000th signature
"Protocol not supported"
I suspected [Auto Power On Off](https://ludovicrousseau.blogspot.co.at/2010/10/card-auto-power-on-and-off.html) (pcscd Compiler options) as the culprit.
I'm using pcsclite 1.8.20  libccid 1.4.22 with Gemalto Twin CT40 and Cherry TC1100.

mickflemm, 2017-03-16:
https://github.com/LudovicRousseau/CCID/pull/24