LudovicRousseau/CCID #26: CCID ID USB = Reader

joaoguariglia, 2016-12-19:
Good Morning!
I have a question about CCID, in the pcsc_scan program I can see the READER: number.
I would like to know if it is possible to identify the READER with the usb ID, when you give the command
# Luscious
Bus 001 Device 004: ID 1059: 0017
If it is possible to identify the Reader with this ID, or any other ID that is unique to that reader.

I know the newer readers come with serial number, but there are some older CCIDs that have nothing to identify, so I doubt it.

arbego, 2016-12-19:
Hello,

sadly the USB ID cannot be used to identify a reader. As soon as you plug your reader into an other USB port the USB ID will change. There is no way to distinguish between two identical readers when there is no serial number in the USB descriptor.
You can try to fetch the reader's friendly name ([SCardGetAttrib()](https://ludovicrousseau.blogspot.co.at/2011/06/scardgetattrib-by-pcsc-lite.html), take a look at [SCARD_ATTR_DEVICE_FRIENDLY_NAME](https://ludovicrousseau.blogspot.co.at/2010/05/what-is-in-pcsc-reader-name.html)). But also in this case it is likely that your readers get a differnt friendly name each time it reconnects from/to a USB port. You will have to try it.

Regards.

joaoguariglia, 2016-12-19:
Hi! Thank you very much for the quick response.
Just one more question, it is possible to identify where READER: number
Is it pointing to a physical USB?

arbego, 2016-12-19:
> Reader 0: HID Global OMNIKEY 5022 Smart Card Reader [OK 5022 Smart Card Reader] 00 00

As far as I know "Reader **0**" is only an internal enumeration index of the pcsc_scan tool. You can verify this by plugging the reader into USB port 1 and then into USB port 2. Both times it will get index 0.

LudovicRousseau, 2016-12-19:
Hello,
You can find information in 2 blog articles:
- "What is in a PC/SC reader name?" https://ludovicrousseau.blogspot.fr/2010/05/what-is-in-pcsc-reader-name.html
- "Identifying a reader model (part 2)" https://ludovicrousseau.blogspot.fr/2012/04/identifying-reader-model-part-2.html

The USB ID "Bus 001 Device 004", as in your example, is *not* available at the PC/SC level.

joaoguariglia, 2016-12-19:
Thank you very much for the answer, I'm going to read it!

closed 2017-01-31 by LudovicRousseau