LudovicRousseau/CCID #28: Cannot compile CCID

DerTomm, 2017-01-31:
Hi Ludovic,

today I tried to compile the most recent versions of pcsc-lite and the ccid driver on my CentOS 7 machine.

This was my procedure:
- git clone --recursive https://github.com/LudovicRousseau/CCID.git
- bootstrap, configure, make and make install of PCSC
- then in CCID:
`PCSC_LIBS="-L/usr/local/lib -lpcsclite" ./configure PCSC_CFLAGS=-I/usr/local/include/PCSC --enable-usbdropdir=/usr/local/lib/pcsc/drivers`

Until here everything worked fine. But when I start __make__ I get the following error:
```
make[2]: Entering directory `/home/user/temp/CCID/src'
  CCLD     parse
/usr/bin/ld: parse-ccid_usb.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
/usr/lib64/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [parse] Error 1
make[2]: Leaving directory `/home/user/temp/CCID/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/user/temp/CCID'
make: *** [all] Error 2
```
Do you have any idea what could cause this behaviour?

The hint you gave [here](http://musclecard.996296.n3.nabble.com/can-t-quot-make-quot-ccid-on-Ubuntu-14-04-4-LTS-td5904.html) did not work for me. The package _pkgconfig.x86_64_ is already installed.

In relation to this error I often read that the linker flag "-lpthread" might be missing - could that be the cause of this problem?

My environment:
- CentOS 7.3
- x64
- gcc 4.8.5

Regards,
Thomas

LudovicRousseau, 2017-01-31:
Try with this patch:
```
--- /var/folders/jb/2mvc64nx74b76qjg_5yk8zs00000gn/T//XPXUSb_Makefile.am    2017-01-31 17:50:51.000000000 +0100
+++ src/Makefile.am 2017-01-31 17:48:50.000000000 +0100
@@ -67,8 +67,9 @@ libccidtwin_la_LIBADD = $(PTHREAD_LIBS)
 libccidtwin_la_LDFLAGS = -avoid-version
 
 parse_SOURCES = parse.c debug.c ccid_usb.c $(TOKEN_PARSER)
-parse_LDADD = $(LIBUSB_LIBS)
-parse_CFLAGS = $(PCSC_CFLAGS) $(LIBUSB_CFLAGS) -DSIMCLIST_NO_DUMPRESTORE
+parse_LDADD = $(LIBUSB_LIBS) $(PTHREAD_LIBS)
+parse_CFLAGS = $(PCSC_CFLAGS) $(LIBUSB_CFLAGS) $(PCSC_CFLAGS)\
+   -DSIMCLIST_NO_DUMPRESTORE
 
 EXTRA_DIST = Info.plist.src create_Info_plist.pl reader.conf.in \
    towitoko/COPYING towitoko/README openct/LICENSE openct/README \
```

DerTomm, 2017-02-01:
Thanks for the patch. I just applied it - unfortunately it didn't solve the problem. I still get the same "undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'" error message.

This is the __configure__ file which is created by `./bootstrap`:
[configure.txt](https://github.com/LudovicRousseau/CCID/files/744085/configure.txt)

And this the the __config.log__ file:
[config.log.txt](https://github.com/LudovicRousseau/CCID/files/744081/config.log.txt)

(I appended .txt to both files for Github compliance)
Maybe these files can help finding the cause...

LudovicRousseau, 2017-02-01:
I just installed a CentOS 7 system and I have no problem rebuilding the CCID.
I used the ``pcsc-lite-devel`` package provided by CentOS. I don't know why you use your own version of pcsc-lite installed in ``/usr/local/``.

The difference with you is that I have:
```
$ pkg-config --libs libpcsclite
-lpcsclite  
$ pkg-config --cflags libpcsclite
-pthread -I/usr/include/PCSC  
```

You should add the missing ``-pthread`` in your ``PCSC_CFLAGS``.

closed 2017-02-01 by LudovicRousseau

DerTomm, 2017-02-01:
Adding "-pthread" to PCSC_CFLAGS solved the problem. Thanks for the great support :-)

The reason why I compiled pcsc-lite and CCID myself is that we encounter erratic crashes of the pcscd process which comes in version 1.8.8 in the CentOS 7 respository. I hope that installing the current releases of pcsc-lite and the driver solve this problem.

LudovicRousseau, 2017-02-01:
pcsc-lite 1.8.8 is from January 2013 (4 years ago).

CentOS/RedHat is not really up to date :-)