LudovicRousseau/CCID #29: Issues compiling CCID 1.4.26 on Ubuntu 16.04

acidrop, 2017-02-07:
Hello,

I have a Dell Latitude E7270 with 0a5c:5832 Broadcom Corp. builtin smartcard. I can see that this card *is* supported on the latest versions on CCID. When trying to compile CCID on my system (Ubuntu 16.04) I'm getting the errors below.

```
~/git/CCID-ccid-1.4.26$ ./bootstrap
+ LIBTOOLIZE=libtoolize
+ which glibtoolize
+ aclocal -I m4
aclocal: error: aclocal: file 'm4/ax_pthread.m4' does not exist
+ libtoolize --copy --force --automake
+ autoheader --force
+ autoconf --force
configure.ac:10: error: possibly undefined macro: AM_INIT_AUTOMAKE
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:26: error: possibly undefined macro: AM_MAINTAINER_MODE
configure.ac:30: error: possibly undefined macro: AM_PROG_CC_C_O
configure.ac:35: error: possibly undefined macro: AM_PROG_LEX
configure.ac:36: error: possibly undefined macro: AM_PROG_AR
configure.ac:45: error: possibly undefined macro: AC_MSG_ERROR
configure.ac:47: error: possibly undefined macro: AC_MSG_WARN
configure.ac:132: error: possibly undefined macro: AM_CONDITIONAL
configure.ac:141: error: possibly undefined macro: AC_DEFINE
configure.ac:274: error: possibly undefined macro: AS_AC_EXPAND
+ automake --add-missing --copy --force --foreign
configure.ac: error: no proper invocation of AM_INIT_AUTOMAKE was found.
configure.ac: You should verify that configure.ac invokes AM_INIT_AUTOMAKE,
configure.ac: that aclocal.m4 is present in the top-level directory,
configure.ac: and that aclocal.m4 was recently regenerated (using aclocal)
contrib/Kobil_mIDentity_switch/Makefile.am: installing './depcomp'
/usr/share/automake-1.15/am/depend2.am: error: am__fastdepCC does not appear in AM_CONDITIONAL
/usr/share/automake-1.15/am/depend2.am:   The usual way to define 'am__fastdepCC' is to add 'AC_PROG_CC'
/usr/share/automake-1.15/am/depend2.am:   to 'configure.ac' and run 'aclocal' and 'autoconf' again
/usr/share/automake-1.15/am/depend2.am: error: AMDEP does not appear in AM_CONDITIONAL
/usr/share/automake-1.15/am/depend2.am:   The usual way to define 'AMDEP' is to add one of the compiler tests
/usr/share/automake-1.15/am/depend2.am:     AC_PROG_CC, AC_PROG_CXX, AC_PROG_OBJC, AC_PROG_OBJCXX,
/usr/share/automake-1.15/am/depend2.am:     AM_PROG_AS, AM_PROG_GCJ, AM_PROG_UPC
/usr/share/automake-1.15/am/depend2.am:   to 'configure.ac' and run 'aclocal' and 'autoconf' again
contrib/Makefile.am:1: error: WITH_LIBUSB does not appear in AM_CONDITIONAL
src/Makefile.am:10: error: WITH_LIBUSB does not appear in AM_CONDITIONAL
src/Makefile.am:16: error: WITH_TWIN_SERIAL does not appear in AM_CONDITIONAL
src/Makefile.am:52: error: WITHOUT_PCSC does not appear in AM_CONDITIONAL
src/Makefile.am:7: error: Libtool library used but 'LIBTOOL' is undefined
src/Makefile.am:7:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/Makefile.am:7:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/Makefile.am:7:   If 'LT_INIT' is in 'configure.ac', make sure
src/Makefile.am:7:   its definition is in aclocal's search path.
src/Makefile.am:34: warning: source file 'towitoko/atr.c' is in a subdirectory,
src/Makefile.am:34: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
automake: this behaviour will change in future Automake versions: they will
automake: unconditionally cause object files to be placed in the same subdirectory
automake: of the corresponding sources.
automake: You are advised to start using 'subdir-objects' option throughout your
automake: project, to avoid future incompatibilities.
src/Makefile.am:34: warning: source file 'towitoko/pps.c' is in a subdirectory,
src/Makefile.am:34: but option 'subdir-objects' is disabled
src/Makefile.am:34: warning: source file 'openct/buffer.c' is in a subdirectory,
src/Makefile.am:34: but option 'subdir-objects' is disabled
src/Makefile.am:34: warning: source file 'openct/checksum.c' is in a subdirectory,
src/Makefile.am:34: but option 'subdir-objects' is disabled
src/Makefile.am:34: warning: source file 'openct/proto-t1.c' is in a subdirectory,
src/Makefile.am:34: but option 'subdir-objects' is disabled
src/Makefile.am: error: Lex source seen but 'LEX' is undefined
src/Makefile.am:   The usual way to define 'LEX' is to add 'AM_PROG_LEX'
src/Makefile.am:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
yannis@ubuntu:~/git/CCID-ccid-1.4.26$ ./configure
./configure: line 2241: syntax error near unexpected token `1.8'
./configure: line 2241: `AM_INIT_AUTOMAKE(1.8 dist-bzip2 no-dist-gzip subdir-objects)'
```

Could you please suggest what I'm doing wrong?

Thanks,
Yannis

LudovicRousseau, 2017-02-07:
Install the automake package.

Also install the following packages:
Build-Depends: debhelper (>= 9~), libusb-1.0-0-dev, libpcsclite-dev (>= 1.8.3~), flex, pkg-config

xaqq, 2017-02-14:
Hello,

I ran into the exact same issue. However, installing debhelper, libusb-1.0-0-dev, libpcsclite-dev, flex and pkg-config didn't help. The automake / libtool packages are also installed.

It turns out some files in `m4` are symlink that expects a `PCSC` folder in the `CCID` directory.

LudovicRousseau, 2017-02-14:
As documented in https://pcsclite.alioth.debian.org/ccid.html do
```
git clone --recursive git://anonscm.debian.org/pcsclite/CCID.git
cd CCID
./bootstrap
./configure
make
```

xaqq, 2017-02-15:
Yes, maybe it's worth updating the INSTALL file. It simply says to get the release tarball and run ./bootstrap && ./configure and doesn't mention submodules.

Thank you.

LudovicRousseau, 2017-02-15:
The INSTALL file is correct.
Getting the tarball is NOT the same as getting the code from git.

closed 2017-02-15 by LudovicRousseau

xaqq, 2017-02-15:
Mmm I must be missing something wrt INSTALL file then. INSTALL says go get tarball and run commands, but the tarball when extracted contains an empty PCSC folder so the commands fail.

Anyway it works fine when cloning from git and pulling submodule, so thanks !

LudovicRousseau, 2017-02-15:
The official project web site is https://pcsclite.alioth.debian.org/ccid.html
tarballs are available from https://alioth.debian.org/frs/?group_id=30105#title_ccid

github does **not** provide valid tarballs.

jl2035, 2019-08-05:
How about now in 2019 when those links are no longer available?

snakehand, 2020-07-18:
I also had to install flex, which is something configure did not pick up on.