LudovicRousseau/CCID #31: New reader: ACR1255U-J1 ® Secure Bluetooth NFC Reader

frankmorgner, 2017-03-17:
[output.txt](https://github.com/LudovicRousseau/CCID/files/850366/output.txt)

Successfully tested the reader's USB interface with libccid-1.4.26 and pcsc-lite-1.8.13.

The [manufacturer's page](http://www.acs.com.hk/en/products/403/acr1255u-j1-secure-bluetooth%C2%AE-nfc-reader/) also includes a proprietary linux driver (which I didn't try).

LudovicRousseau, 2017-03-17:
ACS provides a forked version https://github.com/acshk/acsccid of my CCID driver.
I guess you used the reader connected using USB only, not Bluetooth?

frankmorgner, 2017-03-17:
Yes, USB only

LudovicRousseau, 2017-03-17:
Add in the "Should work but untested by me" list
https://pcsclite.alioth.debian.org/ccid/shouldwork.html#0x072F0x223F

closed 2017-03-17 by LudovicRousseau