LudovicRousseau/CCID #32: Fujitsu Siemens Computers Smar tCard Keyboard USB 2A init failed

nobanzai, 2017-03-19:
Hello,

I have a Fujitsu Siemens Computers SmartCard Keyboard and get an error when starting pcscd, that initialization is failing.

```
00000000 ifdhandler.c:144:CreateChannelByNameOrChannel() failed
00000024 readerfactory.c:1108:RFInitializeReader() Open Port 0x200000 Failed (usb:0bf8/1005:libudev:0:/dev/bus/usb/004/006)
00000005 readerfactory.c:373:RFAddReader() Fujitsu Siemens Computers SmartCard Keyboard USB 2A init failed.
00163282 ifdhandler.c:144:CreateChannelByNameOrChannel() failed
00000015 readerfactory.c:1108:RFInitializeReader() Open Port 0x200001 Failed (usb:0bf8/1005:libudev:2:/dev/bus/usb/004/006)
00000005 readerfactory.c:373:RFAddReader() Fujitsu Siemens Computers SmartCard Keyboard USB 2A init failed.
Mär 19 16:45:14 susi42.hirmke.de systemd[1]: Stopped PC/SC Smart Card Daemon.
Mär 19 16:45:14 susi42.hirmke.de systemd[1]: Started PC/SC Smart Card Daemon.
00000000 debuglog.c:289:DebugLogSetLevel() debug level=debug
00000026 pcscdaemon.c:461:main() Started by systemd
00000099 configfile.l:282:DBGetReaderListDir() Parsing conf directory: /etc/reader.conf.d
00000023 configfile.l:320:DBGetReaderListDir() Skipping non regular file: ..
00000005 configfile.l:358:DBGetReaderList() Parsing conf file: /etc/reader.conf.d/libccidtwin
00000055 configfile.l:320:DBGetReaderListDir() Skipping non regular file: .
00000005 configfile.l:358:DBGetReaderList() Parsing conf file: /etc/reader.conf.d/reader.conf
00000039 pcscdaemon.c:672:main() pcsc-lite 1.8.17 daemon ready.
00008324 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/001/001
00000206 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0003, path: /dev/bus/usb/002/001
00000182 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/003/001
00000158 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/003/001
00000149 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x8087, PID: 0x0024, path: /dev/bus/usb/003/002
00000161 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x1130, PID: 0x6801, path: /dev/bus/usb/003/003
00000237 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x8087, PID: 0x0024, path: /dev/bus/usb/003/002
00000281 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x058F, PID: 0x6366, path: /dev/bus/usb/003/004
00000319 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x8087, PID: 0x0024, path: /dev/bus/usb/003/002
00000368 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/004/001
00000486 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/004/001
00000374 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x8087, PID: 0x0024, path: /dev/bus/usb/004/002
00000188 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x0451, PID: 0x8142, path: /dev/bus/usb/004/003
00000168 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x0BF8, PID: 0x1005, path: /dev/bus/usb/004/006
00000009 hotplug_libudev.c:433:HPAddDevice() Adding USB device: Fujitsu Siemens Computers SmartCard Keyboard USB 2A
00000058 readerfactory.c:1077:RFInitializeReader() Attempting startup of Fujitsu Siemens Computers SmartCard Keyboard USB 2A 00 00 using /usr/lib64/readers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000195 readerfactory.c:952:RFBindFunctions() Loading IFD Handler 3.0
00000024 ifdhandler.c:1953:init_driver() Driver version: 1.4.25
00000593 ifdhandler.c:1970:init_driver() LogLevel: 0x0003
00000005 ifdhandler.c:1981:init_driver() DriverOptions: 0x0000
00000193 ifdhandler.c:110:CreateChannelByNameOrChannel() Lun: 0, device: usb:0bf8/1005:libudev:0:/dev/bus/usb/004/006
00000023 ccid_usb.c:287:OpenUSBByName() Using: /usr/lib64/readers/ifd-ccid.bundle/Contents/Info.plist
00000660 ccid_usb.c:305:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000006 ccid_usb.c:306:OpenUSBByName() ifdProductString: Generic CCID driver
00000003 ccid_usb.c:307:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00009458 ccid_usb.c:599:OpenUSBByName() Found interface 1 but expecting 0
00000009 ccid_usb.c:601:OpenUSBByName() Wrong interface for USB device 4/6. Checking next one.
00000133 ccid_usb.c:189:close_libusb_if_needed() libusb_exit
00000077 ccid_usb.c:749:OpenUSBByName() Device not found?
00000005 ifdhandler.c:144:CreateChannelByNameOrChannel() failed
00000006 readerfactory.c:1108:RFInitializeReader() Open Port 0x200000 Failed (usb:0bf8/1005:libudev:0:/dev/bus/usb/004/006)
00000003 readerfactory.c:373:RFAddReader() Fujitsu Siemens Computers SmartCard Keyboard USB 2A init failed.
00000005 readerfactory.c:606:RFRemoveReader() UnrefReader() count was: 1
00000003 readerfactory.c:1128:RFUnInitializeReader() Attempting shutdown of Fujitsu Siemens Computers SmartCard Keyboard USB 2A 00 00.
00000002 readerfactory.c:989:RFUnloadReader() Unloading reader driver.
00000224 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x0BF8, PID: 0x1005, path: /dev/bus/usb/004/006
00000007 hotplug_libudev.c:433:HPAddDevice() Adding USB device: Fujitsu Siemens Computers SmartCard Keyboard USB 2A
00000078 readerfactory.c:1077:RFInitializeReader() Attempting startup of Fujitsu Siemens Computers SmartCard Keyboard USB 2A [SmartCard Reader USB] 00 00 using /usr/lib64/readers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000138 readerfactory.c:952:RFBindFunctions() Loading IFD Handler 3.0
00000019 ifdhandler.c:1953:init_driver() Driver version: 1.4.25
00000516 ifdhandler.c:1970:init_driver() LogLevel: 0x0003
00000005 ifdhandler.c:1981:init_driver() DriverOptions: 0x0000
00000195 ifdhandler.c:110:CreateChannelByNameOrChannel() Lun: 0, device: usb:0bf8/1005:libudev:1:/dev/bus/usb/004/006
00000018 ccid_usb.c:287:OpenUSBByName() Using: /usr/lib64/readers/ifd-ccid.bundle/Contents/Info.plist
00000715 ccid_usb.c:305:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000010 ccid_usb.c:306:OpenUSBByName() ifdProductString: Generic CCID driver
00000005 ccid_usb.c:307:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00011219 ccid_usb.c:621:OpenUSBByName() Found Vendor/Product: 0BF8/1005 (Fujitsu Siemens Computers SmartCard Keyboard USB 2A)
00000008 ccid_usb.c:623:OpenUSBByName() Using USB bus/device: 4/6
[... 140 lines cut ...]
00000267 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x0451, PID: 0x8142, path: /dev/bus/usb/004/007
00000308 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x045E, PID: 0x0745, path: /dev/bus/usb/004/009
00000304 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x045E, PID: 0x0745, path: /dev/bus/usb/004/009
00000301 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x045E, PID: 0x0745, path: /dev/bus/usb/004/009
00000281 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x0451, PID: 0x8142, path: /dev/bus/usb/004/007
00000247 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x0451, PID: 0x8142, path: /dev/bus/usb/004/003
00000205 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x8087, PID: 0x0024, path: /dev/bus/usb/004/002
00000260 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x04A9, PID: 0x1093, path: /dev/bus/usb/004/004
00000181 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x8087, PID: 0x0024, path: /dev/bus/usb/004/002
00000206 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x17CC, PID: 0x1001, path: /dev/bus/usb/004/005
00000234 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x17CC, PID: 0x1001, path: /dev/bus/usb/004/005
00000228 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x17CC, PID: 0x1001, path: /dev/bus/usb/004/005
00000230 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x17CC, PID: 0x1001, path: /dev/bus/usb/004/005
00000213 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x17CC, PID: 0x1001, path: /dev/bus/usb/004/005
00000233 hotplug_libudev.c:294:get_driver() Looking for a driver for VID: 0x8087, PID: 0x0024, path: /dev/bus/usb/004/002
04997228 ifdhandler.c:1146:IFDHPowerICC() action: PowerDown, usb:0bf8/1005:libudev:1:/dev/bus/usb/004/006 (lun: 0)
00000706 eventhandler.c:479:EHStatusHandlerThread() powerState: POWER_STATE_UNPOWERED
```

This is an openSuSE Leap 42.2 system.
Any idea, what might be wrong here?

TIA.
Bye.
Michael

LudovicRousseau, 2017-03-19:
You get an error message in the logs but does the reader works?
Follow https://ludovicrousseau.blogspot.fr/2014/03/level-1-smart-card-support-on-gnulinux.html

nobanzai, 2017-03-19:
According to your blog the reader is working and it recognizes the card, although the card itself is reported as unknown. Perhaps I should have said, why I thought the problem is the reader: I try to use it with ICAClient/wfica from Citrix and this program refuses to recognize the reader/card combination. Did you ever hear about that problem?

LudovicRousseau, 2017-03-19:
No. It is a ICAClient/wfica from Citrix issue.

closed 2017-03-19 by LudovicRousseau

nobanzai, 2017-03-19:
Ok. Thx for your answer.

nobanzai, 2019-03-29:
In the meantime Citrix released a few new versions and changed the product name from Receiver to Workplace last year.
Now the card (reader) seems to work in general, i.e. I can view the certificates in the Windows client started inside the Citrix session, the passphrase window for authentication is shown, ...
Nevertheless decrypting a mail or authenticating against windows at least are failing. In the logs I get:
...
Mär 29 18:32:15 thunder pcscd[7534]: 00000041 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612
Mär 29 18:32:15 thunder pcscd[7534]: 00000014 winscard.c:1626:SCardTransmit() Card not transacted: 0x80100016
Mär 29 18:32:22 thunder pcscd[7534]: 06691816 commands.c:1771:CmdXfrBlockTPDU_T0() Command too long (266 bytes) for max: 261 bytes
Mär 29 18:32:22 thunder pcscd[7534]: 00000037 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612
Mär 29 18:32:22 thunder pcscd[7534]: 00000013 winscard.c:1626:SCardTransmit() Card not transacted: 0x80100016
Mär 29 18:32:33 thunder pcscd[7534]: 11426723 commands.c:1771:CmdXfrBlockTPDU_T0() Command too long (266 bytes) for max: 261 bytes
...

Is this still a problem with Citrix?
Running pcsc_scan hangs at:

Possibly identified card (using /home/hirmkem/.cache/smartcard_list.txt):
3B DF 96 FF 81 31 FE 45 80 5B 44 45 2E 42 41 5F 53 43 33 35 32 81 05 B5
 \   

Thx and bye.
Michael.

LudovicRousseau, 2019-03-29:
This time it is a problem with the reader.
`Command too long (266 bytes) for max: 261 bytes` indicates that the application is sending a extended APDU command. But your reader https://ccid.apdu.fr/ccid/shouldwork.html#0x0BF80x1005 does not support it.

See https://ccid.apdu.fr/ccid_extended_apdu.html and https://ludovicrousseau.blogspot.com/2011/05/extended-apdu-status-per-reader.html

nobanzai, 2019-03-29:
Thx for your answer.
This is Interesting.
If the reader doesn't support it, why does it work from Windows?
If I just start Windows and use the same hardware and connect via Citrix to my workplace just as I do from Linux, everything works as expected.

Bye.
Michael.

LudovicRousseau, 2019-03-30:
Maybe the Windows and Linux application are not the same?
Maybe the Citrix client on Linux is not the same as the Citrix client on Windows?

Generate a pcscd log as described on https://ccid.apdu.fr/#support

nobanzai, 2019-03-30:
The Citrix client for Windows is for sure different from the one for Linux, the applications within the Citrix client are identical.
Where do you want me to send the log file to? It seems to be too large to attach it here.

LudovicRousseau, 2019-03-30:
My email is listed on https://ccid.apdu.fr/#support

nobanzai, 2019-03-30:
you should have got mail

LudovicRousseau, 2019-03-30:
The PC/SC application tried to send the APDU: 00 2A 80 86 00 01 01 81 73 41 93 44 C1 78 E7 71 7C B6 4B 6D 0B 5A 0B 38 ED B1 58 ED B5 23 3A AF 54 2A 13 73 18 48 E9 01 01 FB E9 52 49 B4 4D 57 8E 95 1E EA 34 F0 0C 75 08 25 8D 80 C4 72 05 EA C3 6B F2 8F DC C1 7E 33 DE 51 75 37 DC 55 88 29 10 4C CC 01 82 02 AD 80 14 E0 06 58 58 83 03 2E 79 93 2E C8 89 33 47 BD CC 16 1C 9F 36 C1 FA 73 D1 62 06 96 0E 9A 4B B8 B2 6F F7 CC 06 68 05 AF 55 E2 2B 50 71 9F AD 57 FF CC 78 04 36 56 7A DF D2 F5 4F 00 2D 77 AC 8B 10 E1 47 CE 64 0D 06 84 CA C6 94 07 DE D4 CD BF FF 29 1F 76 64 9E 71 BD B8 A4 FF 9D 87 FD 63 98 E8 F9 4A 8B 58 86 8B 2B 16 9D 2F 55 2F E7 39 96 CC 6F C3 7C 3E C5 CF F1 05 9F 1E A1 F6 45 C9 55 A4 2D 19 2F A2 82 FC 58 C1 1D 97 36 B6 43 B9 C9 29 40 0D CF B3 82 E8 37 55 6E 32 4F 3B B7 C7 CD A3 3F D1 F3 34 EC C8 EF FC 0F 25 55 DF 13 E2 0F 01 05
which is an extended APDU.

Maybe the Windows driver knows a proprietary command to switch the reader in extended APDU mode.
The best I can suggest is to find a reader with extended APDU support.

nobanzai, 2019-03-31:
Thx for your patience. I'll try a different smartcard keyboard.

nobanzai, 2019-04-10:
I tested
- CHERRY SmartTerminal ST-1144UB USB cardreader
- Fujitsu SmartCard KB100 SCR Black D

Both devices work within a Citrix VDI session.
When logging in to a different machine via RDP from within the VDI session, the smartcard information is transmitted to the RDP session, too. I have to retry authentication two or three times, though, until the information from the smartcard container is known to the RDP client. After that, decrypting mails in Outlook works like a charme.
Is there anything I can do to avoid the retries on authentication?
Besides that, thanks for the hint regarding APDU!

LudovicRousseau, 2019-04-10:
* The "Fujitsu SmartCard KB100 SCR" has support of extended APDU https://ccid.apdu.fr/ccid/readers/FujitsuTechnologySolutions_Keyboard_KB100_SCR.txt
* I don't know what the "CHERRY SmartTerminal ST-1144UB USB cardreader" is.

No idea about authentication.

nobanzai, 2019-04-10:
Ok, thx again.
I stay with the actual state, which is usable for me.