LudovicRousseau/CCID #35: No smart card inserted. [SCM Microsystems Inc. SCR 3310]

fredericosar, 2017-07-29:
Hello,

First, my apologies if this is the incorrect place for this question.

I am using a SCM Microsystems Inc. SCR 3310 and trying to read an AT88SC0404C.

When I run pcsctest I get the following results:

Enter the reader number          : 1
Waiting for card insertion         
                                                   : Command successful.
Testing SCardConnect             : No smart card inserted.

Any suggestion of what could be happening?

Thank you

LudovicRousseau, 2017-07-29:
You should use the Pcsclite-muscle mailing list for support questions https://lists.alioth.debian.org/mailman/listinfo/pcsclite-muscle

It looks like the AT88SC0404C is a *memory* card and not a *smart* card. I am not surprised you can't use it.

closed 2017-07-29 by LudovicRousseau

P-Xavier, 2017-07-31:
Memory card support would need tools from the reader vendors
http://support.identiv.com/developer-tools-for-smart-card-readers/