LudovicRousseau/CCID #38: The checking of `bInterfaceNumber` is skipped at the first time

godfreychung, 2017-11-08:
The `static_interface` is initialized with -1 and the `interface_number` is initialized with `static_interface`.
https://github.com/LudovicRousseau/CCID/blob/373a0af0f91fde530ddd782e991a78bf42e7124d/src/ccid_usb.c#L429
https://github.com/LudovicRousseau/CCID/blob/373a0af0f91fde530ddd782e991a78bf42e7124d/src/ccid_usb.c#L472

Therefore, the checking of `bInterfaceNumber` is skipped at the first time.
https://github.com/LudovicRousseau/CCID/blob/373a0af0f91fde530ddd782e991a78bf42e7124d/src/ccid_usb.c#L628-L629

Will it be an issue if wrong interface returns at the first time?

LudovicRousseau, 2017-11-08:
No.
Is it an issue for you?

godfreychung, 2017-11-08:
The driver will not load the next interface if the first interface is wrong. If the libusb returns correct order of interfaces, it will be no problem. Actually, I tried many times and I couldn't reproduce the problem on macOS.

According to your driver code, it seems that the checking of interface number is to guard the incorrect order of interfaces. Therefore, I would like to know why the first interface will not be checked.

LudovicRousseau, 2017-11-08:
Why do you write: "the first interface will not be checked."?
The first CCID interface found is always used.

godfreychung, 2017-11-09:
The `interface_number` is initialized with `static_interface`. i.e. The value is -1 at the first time. Therefore, the checking of interface number for the first interface at line 629 is skipped.

LudovicRousseau, 2017-11-09:
Please provide a log as described in https://pcsclite.alioth.debian.org/ccid.html#support and the output of ``lsusb -l`` for the device with the problem.

godfreychung, 2017-11-09:
Thank for your information.

Currently, I'm doing code review and find the potential issue.
- Is it safe to not checking the interface number for the first interface in composite device?

According to your driver code, it always checks for single interface device.

closed 2017-11-18 by LudovicRousseau