LudovicRousseau/CCID #40: HP Smartcard keyboards & SPE

vrmp, 2017-11-17:
HP CCID keyboard  - 0x03F0,0x164A - offers secure pin entry, but does not work in this mode.

Works fine without  SPE (disabled via src/ccid.c like HPSMARTCARDKEYBOARD/HP_CCIDSMARTCARDKEYBOARD)

Can you consider the the disabling SPE for this reader/or all HP keyboards/or making it configurable?

LudovicRousseau, 2017-11-17:
How do you know the device does NOT support SPE?

vrmp, 2017-11-17:
Tried to use it in SPE mode - entering correct PIN in SPE mode  results red light blinking  on reader and transaction failure.

vrmp, 2017-11-17:
possible problems with this reader are also documented at: 
https://www.id.ee/?id=34282

-----
USB\VID_03F0&PID_1624&REV_0535&MI_01

This version of the keyboard might not work as a card reader in Linux and Mac OS X environment because of missing drivers.

LudovicRousseau, 2018-02-09:
I don't know where problem is documented at https://www.id.ee/?id=34282

Also the driver for this reader is **not** missing. You just used it.

closed 2018-02-09 by LudovicRousseau