LudovicRousseau/CCID #42: Yubikey gives "OpenUSBByName() Found interface 2 but expecting 0"

j4cko, 2018-01-11:
Hi, I am trying to use pcscd and ccid together with my yubikey to work with gpg but so far without success. I am unsure if the reason is some recent upgrade of my linux distribution (It had worked before, but I am confused if gpg used its own "scdaemon" or I was talking to pcscd).

The errors I am getting can be seen in the log I attached.

Does someone know what is going on here?
Thanks...
[pcscd.log](https://github.com/LudovicRousseau/CCID/files/1622358/pcscd.log)

LudovicRousseau, 2018-01-11:
Your token ATR is visible in your logs. So the driver and reader are working as expected.

See also https://ludovicrousseau.blogspot.fr/2014/03/level-1-smart-card-support-on-gnulinux.html

closed 2018-01-11 by LudovicRousseau

j4cko, 2018-01-12:
Thanks. It was indeed some deadlock on the gpg-agent ipc; I think systemd starts gpg-agent before pcscd. Anyway, thanks for looking at the log file and sorry for bothering you with this.