LudovicRousseau/CCID #43: Can this Watchdata token be supported?

mndar, 2018-01-31:
There are these Trustkey tokens in the Indian market which seem to be using the Watchdata drivers (pcscd_wd). They distribute a Java application and a pkcs11wrapper with the tokens. There's also a binary called P11ModelTool. I am unaware of its purpose.

However, I would like to use them with opensc and gnutls (p11tool to load certificates).
I added the USB ID (163c:0418) to Info.plist and the Init fails. I'm attaching the output of pcscsd, pcsc_scan and lsusb -v.

Please have a look.

[lsusb.txt](https://github.com/LudovicRousseau/CCID/files/1680585/lsusb.txt)
[pcscd.txt](https://github.com/LudovicRousseau/CCID/files/1680586/pcscd.txt)
[pcsc_scan.txt](https://github.com/LudovicRousseau/CCID/files/1680587/pcsc_scan.txt)

LudovicRousseau, 2018-01-31:
What is the result of parse? See https://pcsclite.alioth.debian.org/ccid.html#CCID_compliant

mndar, 2018-01-31:
Attached output.txt
The link to the manufacturer's website is http://trustkeytoken.co.in/ Takes a long time to load.
Here is the link to the product on Amazon.in https://www.amazon.in/Trustkey-USB-Pack-of-5/dp/B01N4I2OSB/ref=sr_1_1?ie=UTF8&qid=1517392483&sr=8-1&keywords=trustkey

[output.txt](https://github.com/LudovicRousseau/CCID/files/1681209/output.txt)

mndar, 2018-01-31:
I installed ccid-1.4.28 and opensc gives this output

$ opensc-tool -l
Detected readers (pcsc)
Nr.  Card  Features  Name
0    Yes             TrustKey [WDIND USB CCID Key] 00 00

$ pkcs11-tool -L
Available slots:
Slot 0 (0xffffffffffffffff): Virtual hotplug slot
  (empty)
Slot 1 (0x1): TrustKey [WDIND USB CCID Key] 00 00
  (empty)

$ opensc-tool -n
Using reader with a card: TrustKey [WDIND USB CCID Key] 00 00
Unsupported card

Does this mean that pcsc-lite supports the token and support needs to be added in OpenSC ?

LudovicRousseau, 2018-01-31:
The driver available at http://trustkeytoken.co.in/ is my CCID driver. I was not able to find the source code (as requested by GNU LGPL v2.1) so I can't tell if they made modifications or not.

I added your device in the "Should work but untested by me" list at https://pcsclite.alioth.debian.org/ccid/shouldwork.html#0x163C0x0418

It looks like the device works fine at the PC/SC level. Maybe OpenSC has no support for the application inside the token. I can't help for that.

closed 2018-01-31 by LudovicRousseau