LudovicRousseau/CCID #46: SCR331 SmartCard Reader not detected - FreeBSD 11.1

ds2k5, 2018-02-16:
Hello,
I got a SCR331 SmartCard Reader. put pcsc_scan did not find it.

```
 # lsusb 
Bus /dev/usb Device /dev/ugen1.3: ID 04e6:e001 SCM Microsystems, Inc. SCR331 SmartCard Reader

```
```
# pcsc_scan 
PC/SC device scanner
V 1.4.27 (c) 2001-2011, Ludovic Rousseau <ludovic.rousseau@free.fr>
Compiled with PC/SC lite version: 1.8.22
Using reader plug'n play mechanism
Scanning present readers...
Waiting for the first reader...

```

What did i wrong ?

pcsc-lite-1.8.22,2 and ccid-1.4.28 are installed

Found somthing about Firmware, how to detect
the Firmware Version of the reader ?
How to upgrade if needed ?

Thanks

ds2k5, 2018-02-16:
Found some Firmware there:

http://militarycac.com/cacdrivers.htm#FIRMWARE_UPDATE

But only Windows Flash tool :-(

LudovicRousseau, 2018-02-17:
Please follow https://pcsclite.alioth.debian.org/ccid.html#CCID_compliant
And https://pcsclite.alioth.debian.org/ccid.html#support

ds2k5, 2018-02-17:
Thanks, with FW 5.25 it works!

closed 2018-02-17 by ds2k5