LudovicRousseau/CCID #47: Yubico Yubikey 4 OTP+U2F+CCID init failed

bigon, 2018-05-01:
Hello,

I just bought a yubiley, the key is detected by the ccid driver but it doesn't work

[yuby_pcscd.txt](https://github.com/LudovicRousseau/CCID/files/1963949/yuby_pcscd.txt)

LudovicRousseau, 2018-05-01:
Why do you say it does not work?
From the log I can see the Yubikey 4 ATR https://smartcard-atr.appspot.com/parse?ATR=3BF81300008131FE15597562696B657934D4

Please see https://ludovicrousseau.blogspot.fr/2014/03/level-1-smart-card-support-on-gnulinux.html

bigon, 2018-05-02:
mmmh, indeed I overlooked the ATR at the end of the log file and got confused by the multiple interfaces on the device

LudovicRousseau, 2018-05-02:
If gnupg does not work then you should ask support from gnupg team.
I am closing this issue.

closed 2018-05-02 by LudovicRousseau