LudovicRousseau/CCID #48: BUG in Chicony HP Skylab USB Smartcard Keyboard

jarek556, 2018-05-16:
rdesktop and xfreerdp are closing while inserting Athena card into Chicony HP Skylab USB Smartcard Keyboard reader.  This same configuration works fine with ACS ACR 38U reader.
There is no error or segfault - application just closes after insertion of card:
rdesktop -r scard <server ip>
xfreerdp -sec-nla /smartcard:"Chicony HP Skylab USB Smartcard Keyboard [HP Skylab Smartcard Reader]" /v:<server ip>:<port>

jarek556, 2018-05-16:
Hello!

>Why do you think the bug is in the CCID driver?

OK. I'm not 100% sure, but I have tested the following configuration:

Debian with pcscd (1.8.23-3) + libccid (1.4.29-2) + rdesktop (1.8.3)

and with ACS card reader it works fine, but with Chicony HP Skylab it
fails (rdesktop closes after insertion of Athena card).
I have repeated the same test with xfreerdp, and there is exactly the
same problem.

As I understand, libccid is responsible for interfacing with card
reader, so it works differently when works with ACS than with Chicony.

Of course there is for sure some problem with rdesktop or libpcsclite -
the application should close in such way. 
best regards
Jaroslaw Tabor

LudovicRousseau, 2018-05-17:
The CCID driver is not in the same process as the application (rdesktop in your case). So the crash can't be in the driver.
Maybe the bug is in libpcsclite used by the application.

Without a gdb trace or some more info I can't help.

LudovicRousseau, 2018-06-13:
It looks like the library ascpkcs.dll does not like long reader names. See https://github.com/LudovicRousseau/PCSC/issues/39#issuecomment-396849801

@jarek556 do you also use this library on Windows?

LudovicRousseau, 2018-10-28:
Please provide a complete log as documented in https://ccid.apdu.fr/#support

lainosantos, 2018-11-14:
@LudovicRousseau, Hello.

these are the bug information context:

*Versions
/usr/sbin/pcscd --version
pcsc-lite version 1.8.23.
Copyright (C) 1999-2002 by David Corcoran <corcoran@musclecard.com>.
Copyright (C) 2001-2015 by Ludovic Rousseau <ludovic.rousseau@free.fr>.
Copyright (C) 2003-2004 by Damien Sauveron <sauveron@labri.fr>.
Report bugs to <pcsclite-muscle@lists.alioth.debian.org>.
Enabled features: Linux x86_64-redhat-linux-gnu libsystemd serial usb libudev usbdropdir=/usr/lib64/pcsc/drivers ipcdir=/var/run/pcscd configdir=/etc/reader.conf.d

*Platform
Fedora 28 Spin Kde
Keyboard with reader: HP USB Slim Smartcard CCID KB

*Log
[pcsc.txt](https://github.com/LudovicRousseau/CCID/files/2580997/pcsc.txt)

lainosantos, 2018-11-19:
[36m00000001[0m [35mreaderfactory.c:376:RFAddReader() Chicony HP Skylab USB Smartcard Keyboard [HP Skylab Smartcard Keyboard] (17100700001229) init failed.

even with this error should work?

LudovicRousseau, 2018-11-19:
yes

closed 2018-11-19 by LudovicRousseau

lainosantos, 2018-11-19:
@LudovicRousseau,

I'm getting this error on scriptor.

$> scriptor 
No reader given: using Chicony HP Skylab USB Smartcard Keyboard [HP Skylab Smartcard Reader] (17100700001229) 00 00
Using T=0 protocol
Reading commands from STDIN
00 A4 00 02 3F 00
> 00 A4 00 02 3F 00
Can't get info: Transaction failed.

LudovicRousseau, 2018-11-20:
https://ccid.apdu.fr/#support