LudovicRousseau/CCID #50: Error in ./bootstrap, error in make

PiedPieper, 2018-07-31:
I tried your suggestion here https://github.com/LudovicRousseau/CCID/issues/29 :

> mangoh@mangoh-virtualbox:~/repos$ git clone --recursive git://anonscm.debian.org/pcsclite/CCID.git
> Cloning into 'CCID'...
> fatal: unable to connect to anonscm.debian.org:
> anonscm.debian.org[0: 2001:648:2ffc:deb::211:202]: errno=Connection refused
> anonscm.debian.org[1: 194.177.211.202]: errno=Connection refused

It looks like the code is no longer available at that location.. So I tried it again here:

> mangoh@mangoh-virtualbox:~/repos$ git clone https://github.com/LudovicRousseau/CCID.git
> Cloning into 'CCID'...
> remote: Counting objects: 11097, done.
> remote: Compressing objects: 100% (5/5), done.
> remote: Total 11097 (delta 0), reused 2 (delta 0), pack-reused 11092
> Receiving objects: 100% (11097/11097), 1.85 MiB | 0 bytes/s, done.
> Resolving deltas: 100% (8349/8349), done.
> Checking connectivity... done.
> mangoh@mangoh-virtualbox:~/repos$ cd CCID/
> mangoh@mangoh-virtualbox:~/repos/CCID$ ./bootstrap 
> + LIBTOOLIZE=libtoolize
> + which glibtoolize
> + aclocal -I m4
> aclocal: error: aclocal: file 'm4/ax_pthread.m4' does not exist
> + libtoolize --copy --force --automake
> + autoheader --force
> + autoconf --force
> configure.ac:10: error: possibly undefined macro: AM_INIT_AUTOMAKE
>       If this token and others are legitimate, please use m4_pattern_allow.
>       See the Autoconf documentation.

And encountered the same error as @xaqq. So, I attempted to work from the tarball, following the directions...:

mangoh@mangoh-virtualbox:~/repos$ wget https://github.com/LudovicRousseau/CCID/archive/ccid-1.4.28.tar.gz
...

in the INSTALL directions, it says:

> $ tar xzvf ccid-x.y.z.tar.gz
> $ cd ccid-x.y.z
> $ ./configure
> $ make
> $ sudo make install

but there is no configure script, so i assume that a ./bootstrap is needed to generate it. When I run bootsrap, I get the same error:

> mangoh@mangoh-virtualbox:~/repos/CCID-ccid-1.4.28$ ./bootstrap 
> + LIBTOOLIZE=libtoolize
> + which glibtoolize
> + aclocal -I m4
> aclocal: error: aclocal: file 'm4/ax_pthread.m4' does not exist
> + libtoolize --copy --force --automake
> + autoheader --force
> + autoconf --force
> configure.ac:10: error: possibly undefined macro: AM_INIT_AUTOMAKE
>       If this token and others are legitimate, please use m4_pattern_allow.
>       See the Autoconf documentation.

Following your advice above though, that PCSC must be in the same directory as CCID, from inside of CCID I ran:

> git clone https://salsa.debian.org/rousseau/PCSC.git
> cd PCSC
> ./bootstrap
> ./configure
> make

This all worked. Following, I re-entered the CCID directory and ran ./bootstrap and ./configure. Worked. 

Then I run make:

> mangoh@mangoh-virtualbox:~/repos/CCID-ccid-1.4.28$ make
> make  all-recursive
> make[1]: Entering directory '/home/mangoh/repos/CCID-ccid-1.4.28'
> Making all in readers
> make[2]: Entering directory '/home/mangoh/repos/CCID-ccid-1.4.28/readers'
> make[2]: Nothing to be done for 'all'.
> make[2]: Leaving directory '/home/mangoh/repos/CCID-ccid-1.4.28/readers'
> Making all in examples
> make[2]: Entering directory '/home/mangoh/repos/CCID-ccid-1.4.28/examples'
>   CC       scardcontrol-scardcontrol.o
> scardcontrol.c:34:26: fatal error: PCSCv2part10.h: No such file or directory
> compilation terminated.
> Makefile:412: recipe for target 'scardcontrol-scardcontrol.o' failed
> make[2]: *** [scardcontrol-scardcontrol.o] Error 1
> make[2]: Leaving directory '/home/mangoh/repos/CCID-ccid-1.4.28/examples'
> Makefile:439: recipe for target 'all-recursive' failed
> make[1]: *** [all-recursive] Error 1
> make[1]: Leaving directory '/home/mangoh/repos/CCID-ccid-1.4.28'
> Makefile:371: recipe for target 'all' failed
> make: *** [all] Error 2

And it errors out. If these build instructions are not up-to-date, can you please point me in the right direction? Thank you!

LudovicRousseau, 2018-07-31:
The project home page is https://ccid.apdu.fr/
The documentation should be up to date.

If you still have problems open a new ticket.

closed 2018-07-31 by LudovicRousseau

PiedPieper, 2018-07-31:
This worked. Thank you