LudovicRousseau/CCID #51: Error: CmdEscape error on byte 10. Gemalto USB Shell Token V2 CentOS 7

draganafcb, 2018-08-17:
I'm trying to get my Gemalto USB Shell Token V2 to run on CentOS 7 but it seems there are bugs and errors with the drivers. Here are some logs:

**/var/log/messages**

Aug 17 04:02:24 localhost systemd: Starting Session 3 of user root.
Aug 17 04:02:47 localhost systemd: Started PC/SC Smart Card Daemon.
Aug 17 04:02:47 localhost systemd: Starting PC/SC Smart Card Daemon...
Aug 17 04:02:48 localhost pcscd: 00000000 utils.c:53:GetDaemonPid() Can't open /var/run/pcscd/pcscd.pid: No such file or directory
Aug 17 04:02:48 localhost pcscd: 00213651 commands.c:905:CmdEscape error on byte 10

**pcscd -adf** 

00000000 pcscdaemon.c:233:main() pcscd set to foreground with debug send to stdout
[36m00000118[0m [34mpcscdaemon.c:238:main() Force colored logs[0m
[36m00000151[0m configfile.l:245:DBGetReaderListDir() Parsing conf directory: /etc/reader.conf.d
[36m00000013[0m configfile.l:257:DBGetReaderListDir() Skipping non regular file: .
[36m00000005[0m configfile.l:257:DBGetReaderListDir() Skipping non regular file: ..
[36m00000009[0m [34mpcscdaemon.c:525:main() pcsc-lite 1.8.8 daemon ready.[0m
[36m00020646[0m hotplug_libudev.c:260:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0001, path: /dev/bus/usb/002/001
[36m00000180[0m hotplug_libudev.c:260:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0001, path: /dev/bus/usb/002/001
[36m00000193[0m hotplug_libudev.c:260:get_driver() Looking for a driver for VID: 0x0E0F, PID: 0x0003, path: /dev/bus/usb/002/002
[36m00000104[0m hotplug_libudev.c:260:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0001, path: /dev/bus/usb/002/001
[36m00000110[0m hotplug_libudev.c:260:get_driver() Looking for a driver for VID: 0x0E0F, PID: 0x0002, path: /dev/bus/usb/002/003
[36m00000112[0m hotplug_libudev.c:260:get_driver() Looking for a driver for VID: 0x08E6, PID: 0x3438, path: /dev/bus/usb/002/004
[36m00000024[0m hotplug_libudev.c:260:get_driver() Looking for a driver for VID: 0x08E6, PID: 0x3438, path: /dev/bus/usb/002/004
[36m00000004[0m [34mhotplug_libudev.c:312:HPAddDevice() Adding USB device: Gemalto USB Shell Token V2[0m
[36m00000069[0m [34mreaderfactory.c:978:RFInitializeReader() Attempting startup of Gemalto USB Shell Token V2 (27306242) 00 00 using /usr/lib64/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so[0m
[36m00000597[0m [34mreaderfactory.c:868:RFBindFunctions() Loading IFD Handler 3.0[0m
[36m00000051[0m [34mifdhandler.c:1840:init_driver() Driver version: 1.4.10[0m
[36m00000784[0m [34mifdhandler.c:1857:init_driver() LogLevel: 0x0003[0m
[36m00000011[0m [34mifdhandler.c:1868:init_driver() DriverOptions: 0x0000[0m
[36m00000094[0m [34mifdhandler.c:83:CreateChannelByNameOrChannel() Lun: 0, device: usb:08e6/3438:libudev:0:/dev/bus/usb/002/004[0m
[36m00000061[0m [34mccid_usb.c:258:OpenUSBByName() Using: /usr/lib64/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist[0m
[36m00000380[0m [34mccid_usb.c:276:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)[0m
[36m00000005[0m [34mccid_usb.c:277:OpenUSBByName() ifdProductString: Generic CCID driver[0m
[36m00000003[0m [34mccid_usb.c:278:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.[0m
[36m00003848[0m [34mccid_usb.c:551:OpenUSBByName() Found Vendor/Product: 08E6/3438 (Gemalto USB Shell Token V2)[0m
[36m00000010[0m [34mccid_usb.c:553:OpenUSBByName() Using USB bus/device: 2/4[0m
[36m00230533[0m [34mccid_usb.c:1054:get_data_rates() declared: 12903 bps[0m
[36m00000023[0m [34mccid_usb.c:1054:get_data_rates() declared: 25806 bps[0m
[36m00000006[0m [34mccid_usb.c:1054:get_data_rates() declared: 51613 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 103226 bps[0m
[36m00000005[0m [34mccid_usb.c:1054:get_data_rates() declared: 206452 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 412903 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 825806 bps[0m
[36m00000005[0m [34mccid_usb.c:1054:get_data_rates() declared: 154839 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 258065 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 17204 bps[0m
[36m00000005[0m [34mccid_usb.c:1054:get_data_rates() declared: 34409 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 68817 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 137634 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 275269 bps[0m
[36m00000005[0m [34mccid_usb.c:1054:get_data_rates() declared: 550538 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 172043 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 12403 bps[0m
[36m00000005[0m [34mccid_usb.c:1054:get_data_rates() declared: 24806 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 49612 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 99225 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 198450 bps[0m
[36m00000005[0m [34mccid_usb.c:1054:get_data_rates() declared: 396899 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 74419 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 124031 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 86022 bps[0m
[36m00000005[0m [34mccid_usb.c:1054:get_data_rates() declared: 38710 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 64516 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 10323 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 20645 bps[0m
[36m00000005[0m [34mccid_usb.c:1054:get_data_rates() declared: 41290 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 82581 bps[0m
[36m00000004[0m [34mccid_usb.c:1054:get_data_rates() declared: 165161 bps[0m
[... 33 lines cut ...]
[36m00010950[0m readerfactory.c:1309:RFWaitForReaderInit() Waiting init for reader: Gemalto USB Shell Token V2 (27306242) 00 00
[36m00010772[0m readerfactory.c:1309:RFWaitForReaderInit() Waiting init for reader: Gemalto USB Shell Token V2 (27306242) 00 00
[36m00010347[0m readerfactory.c:1309:RFWaitForReaderInit() Waiting init for reader: Gemalto USB Shell Token V2 (27306242) 00 00
[36m00010872[0m readerfactory.c:1309:RFWaitForReaderInit() Waiting init for reader: Gemalto USB Shell Token V2 (27306242) 00 00
[36m00011195[0m readerfactory.c:1309:RFWaitForReaderInit() Waiting init for reader: Gemalto USB Shell Token V2 (27306242) 00 00
[36m00010896[0m readerfactory.c:1309:RFWaitForReaderInit() Waiting init for reader: Gemalto USB Shell Token V2 (27306242) 00 00
[36m00010913[0m readerfactory.c:1309:RFWaitForReaderInit() Waiting init for reader: Gemalto USB Shell Token V2 (27306242) 00 00
[36m00010991[0m readerfactory.c:1309:RFWaitForReaderInit() Waiting init for reader: Gemalto USB Shell Token V2 (27306242) 00 00
[36m00010231[0m readerfactory.c:1309:RFWaitForReaderInit() Waiting init for reader: Gemalto USB Shell Token V2 (27306242) 00 00
[36m00002362[0m eventhandler.c:256:EHStatusHandlerThread() powerState: POWER_STATE_POWERED
[36m00000069[0m [34mCard ATR: 3B 16 96 41 73 74 72 69 64 [0m
[36m05700146[0m [34mifdhandler.c:1122:IFDHPowerICC() action: PowerDown, usb:08e6/3438:libudev:0:/dev/bus/usb/002/004 (lun: 0)[0m
[36m00014005[0m eventhandler.c:446:EHStatusHandlerThread() powerState: POWER_STATE_UNPOWERED

**opensc-tool -l**
# Detected readers (pcsc)
Nr.  Card  Features  Name
0    Yes             Gemalto USB Shell Token V2 (27306242) 00 00

**lsusb**
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 08e6:3438 Gemalto (was Gemplus) GemPC Key SmartCard Reader
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

LudovicRousseau, 2018-08-27:
Please regenerate pcscd logs as documented at https://ccid.apdu.fr/#support

draganafcb, 2018-08-27:
[log.txt](https://github.com/LudovicRousseau/CCID/files/2323394/log.txt)

draganafcb, 2018-08-27:
[log.txt](https://github.com/LudovicRousseau/CCID/files/2323451/log.txt)

LudovicRousseau, 2018-08-27:
No error in this log. You can ignore the "CmdEscape error on byte 10" log.

I see no card transaction so I guess you have not executed your application.

draganafcb, 2018-08-29:
I am using Gemalto USB Shell Token V2 for installing License Server on the machine, but I am not able to do that. In the installation guide is required to have this type of log in the /var/log/messages.
```
00000006 ccid_usb.c:1054:get_data_rates() declared: 25000 bps
00000007 ccid_usb.c:1054:get_data_rates() declared: 50000 bps
00000006 ccid_usb.c:1054:get_data_rates() declared: 100000 bps
00000006 ccid_usb.c:1054:get_data_rates() declared: 200000 bps
00000006 ccid_usb.c:1054:get_data_rates() declared: 400000 bps
00000006 ccid_usb.c:1054:get_data_rates() declared: 125000 bps
00000006 ccid_usb.c:1054:get_data_rates() declared: 56250 bps
00000006 ccid_usb.c:1054:get_data_rates() declared: 93750 bps
```

I have done the installation on CentOS 6.9 and it is working properly, but I can't on CentOS 7.
This are the logs in /var/log/messages from both CentOS 6.9 and CentOS 7:
[logCentOS7.txt](https://github.com/LudovicRousseau/CCID/files/2331042/logCentOS7.txt)
[logCentOS6.9.txt](https://github.com/LudovicRousseau/CCID/files/2331043/logCentOS6.9.txt)
I am using pcsc-lite version 1.5.2 on the CentOS 6.9 machine.

LudovicRousseau, 2018-08-29:
First step: https://ludovicrousseau.blogspot.com/2014/03/level-1-smart-card-support-on-gnulinux.html

LudovicRousseau, 2018-10-03:
You can ignore this error message. It is harmless.

draganafcb, 2018-10-04:
Oct  4 02:45:18 localhost pcscd: 99999999 commands.c:1443:CCID_Receive() Not enough data received: 0 bytes
Oct  4 02:45:42 localhost systemd: Got automount request for /proc/sys/fs/binfmt_misc, triggered by 2602 (find)
Oct  4 02:45:42 localhost systemd: Mounting Arbitrary Executable File Formats File System...
Oct  4 02:45:42 localhost systemd: Mounted Arbitrary Executable File Formats File System.
Oct  4 02:46:29 localhost pcscd: 71101815 commands.c:1117:CmdGetSlotStatus() Not enough data received: 0 bytes
Oct  4 02:46:29 localhost pcscd: 00000271 ifdwrapper.c:348:IFDStatusICC() Card not transacted: 612
Oct  4 02:46:29 localhost pcscd: 00000003 eventhandler.c:303:EHStatusHandlerThread() Error communicating to: Gemalto USB Shell Token V2 (539F5EC2) 00 00
Oct  4 02:46:29 localhost LS: 2018.10.04 02:46:29 SmartCard.c (160): Error: pkcs11: Unable to initialize engine
Oct  4 02:46:29 localhost LS: 140370160273328:error:820710E0:PKCS#11 module:pkcs11_init_slot:No PKCS#11 token present:p11_slot.c:416:
Oct  4 02:46:29 localhost LS: 2018.10.04 02:46:29 LicenseServerLog.c (682): Error: LS_Step_1 FAILED!
Oct  4 02:46:29 localhost LS: 2018.10.04 02:46:29 Server.c (249): Error: LOG_Init FAILED!!!
Oct  4 02:46:29 localhost LS: ######################################################
Oct  4 02:46:29 localhost LS: #  License Server Up and running
Oct  4 02:46:29 localhost LS: #  License Server RELEASE configuration
Oct  4 02:46:29 localhost LS: #  License Server Version:  R_1.0.4-1-g3b3f918
Oct  4 02:46:29 localhost LS: #  License Server Build:    Aug  9 2018 16:04:50
Oct  4 02:46:29 localhost LS: #  License Server Build By: Jenkins Payment
Oct  4 02:46:29 localhost LS: ######################################################
Oct  4 02:46:30 localhost LS: 2018.10.04 02:46:30 SmartCard.c (182): Error: Finding token

draganafcb, 2018-10-04:
I tried with the newest version of CCID and it doesn't work. Is it possible that the problem is that we are using VMWare, because everything works fine in VirtualBox ?

LudovicRousseau, 2018-10-04:
So use VirtualBox.
Or report the problem to VMWare.

draganafcb, 2018-10-05:
I need it to run on VMWare ...

this is backtrace.log from gdb pcscd
(gdb)set args --apd
(gdb)run

Thread 4 (Thread 0x7ffff4be3700 (LWP 1384)):
#0  0x00007ffff74dbc73 in select () from /lib64/libc.so.6
#1  0x00005555555632b3 in HPEstablishUSBNotifications ()
#2  0x00007ffff77bae25 in start_thread () from /lib64/libpthread.so.0
#3  0x00007ffff74e4bad in clone () from /lib64/libc.so.6

Thread 3 (Thread 0x7ffff53e4700 (LWP 1383)):
#0  0x00007ffff74d9f0d in poll () from /lib64/libc.so.6
#1  0x00007ffff5bef7a0 in handle_events.part.1 () from /lib64/libusb-1.0.so.0
#2  0x00007ffff5bf0689 in libusb_handle_events_timeout_completed () from /lib64/libusb-1.0.so.0
#3  0x00007ffff5bf076f in libusb_handle_events () from /lib64/libusb-1.0.so.0
#4  0x00007ffff5e0b1d8 in InterruptRead () from /usr/lib64/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
#5  0x00007ffff5e06657 in IFDHPolling () from /usr/lib64/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
#6  0x000055555555a75b in EHStatusHandlerThread ()
#7  0x00007ffff77bae25 in start_thread () from /lib64/libpthread.so.0
#8  0x00007ffff74e4bad in clone () from /lib64/libc.so.6

Thread 2 (Thread 0x7ffff5be5700 (LWP 1382)):
#0  0x00007ffff74d9f0d in poll () from /lib64/libc.so.6
#1  0x00007ffff5bf5a7c in linux_udev_event_thread_main () from /lib64/libusb-1.0.so.0
#2  0x00007ffff77bae25 in start_thread () from /lib64/libpthread.so.0
#3  0x00007ffff74e4bad in clone () from /lib64/libc.so.6

Thread 1 (Thread 0x7ffff7fd0740 (LWP 1378)):
#0  0x00007ffff74dbc73 in select () from /lib64/libc.so.6
#1  0x0000555555565ea7 in ProcessEventsServer ()
#2  0x0000555555557ae5 in main ()

LudovicRousseau, 2018-10-05:
Have you noticed you NEVER indicated what does not work?

closed 2018-10-05 by LudovicRousseau