LudovicRousseau/CCID #56: RFAddReader() init failed

ultrabug, 2018-10-05:
Hi

Since I switched to a new machine with systemd my Yubikey is not working with my browsers any more and I must admit that I fail to understand why. It works good with my GPG-Agent for SSH tho.

All my debug got me to **pcscd** which logs this when I plug in the Yubikey 4

```
kernel: usb 1-1.4.2: new full-speed USB device number 86 using xhci_hcd
kernel: usb 1-1.4.2: New USB device found, idVendor=1050, idProduct=0407, bcdDevice= 4.34
kernel: usb 1-1.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
kernel: usb 1-1.4.2: Product: Yubikey 4 OTP+U2F+CCID
kernel: usb 1-1.4.2: Manufacturer: Yubico
kernel: usb 1-1.4.2: SerialNumber: 0005435106
kernel: input: Yubico Yubikey 4 OTP+U2F+CCID as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.4/1-1.4.2/1-1.4.2:1.0/0003:1050:0407.003B/input/input75
kernel: hid-generic 0003:1050:0407.003B: input,hidraw4: USB HID v1.10 Keyboard [Yubico Yubikey 4 OTP+U2F+CCID] on usb-0000:00:14.0-1.4.2/input0
kernel: hid-generic 0003:1050:0407.003C: hiddev97,hidraw5: USB HID v1.10 Device [Yubico Yubikey 4 OTP+U2F+CCID] on usb-0000:00:14.0-1.4.2/input1
mtp-probe[16754]: checking bus 1, device 86: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.4/1-1.4.2"
mtp-probe[16754]: bus: 1, device: 86 was not an MTP device
systemd-udevd[16753]: Process '/bin/env IN_HOTPLUG=1 /etc/init.d/pcscd --quiet start' failed with exit code 127.
pcscd[15553]: 02862899 ifdhandler.c:150:CreateChannelByNameOrChannel() failed
pcscd[15553]: 00000018 readerfactory.c:1106:RFInitializeReader() Open Port 0x200000 Failed (usb:1050/0407:libudev:0:/dev/bus/usb/001/086)
pcscd[15553]: 00000005 readerfactory.c:376:RFAddReader() Yubico Yubikey 4 OTP+U2F+CCID (0005435106) init failed.
pcscd[15553]: 00008017 ifdhandler.c:150:CreateChannelByNameOrChannel() failed
pcscd[15553]: 00000162 readerfactory.c:1106:RFInitializeReader() Open Port 0x200000 Failed (usb:1050/0407:libudev:1:/dev/bus/usb/001/086)
pcscd[15553]: 00000008 readerfactory.c:376:RFAddReader() Yubico Yubikey 4 OTP+U2F+CCID (0005435106) init failed.
```

I googled things and saw that you recommended having a **pcsc_scan** running as well, which outputs this:

```
Using reader plug'n play mechanism
Scanning present readers...
0: Yubico Yubikey 4 OTP+U2F+CCID (0005435106) 00 00
 
Fri Oct  5 14:23:11 2018
 Reader 0: Yubico Yubikey 4 OTP+U2F+CCID (0005435106) 00 00
  Card state: Card inserted, 
  ATR: 3B F8 13 00 00 81 31 FE 15 59 75 62 69 6B 65 79 34 D4

ATR: 3B F8 13 00 00 81 31 FE 15 59 75 62 69 6B 65 79 34 D4
+ TS = 3B --> Direct Convention
+ T0 = F8, Y(1): 1111, K: 8 (historical bytes)
  TA(1) = 13 --> Fi=372, Di=4, 93 cycles/ETU
    43010 bits/s at 4 MHz, fMax for Fi = 5 MHz => 53763 bits/s
  TB(1) = 00 --> VPP is not electrically connected
  TC(1) = 00 --> Extra guard time: 0
  TD(1) = 81 --> Y(i+1) = 1000, Protocol T = 1 
-----
  TD(2) = 31 --> Y(i+1) = 0011, Protocol T = 1 
-----
  TA(3) = FE --> IFSC: 254
  TB(3) = 15 --> Block Waiting Integer: 1 - Character Waiting Integer: 5
+ Historical bytes: 59 75 62 69 6B 65 79 34
  Category indicator byte: 59 (proprietary format)
+ TCK = D4 (correct checksum)

Possibly identified card (using /usr/share/pcsc/smartcard_list.txt):
3B F8 13 00 00 81 31 FE 15 59 75 62 69 6B 65 79 34 D4
	Yubico Yubikey 4 OTP+CCID
Scanning present readers...
```

I'm running out of ideas, so I decided to open this issue to maybe get some help.

Thanks!

LudovicRousseau, 2018-10-05:
You get the token ATR so the PC/SC layer works as expected.
It may be a problem in the library you use in your browser.

ultrabug, 2018-10-05:
Thanks for your fast answer mate

Ok so those `init failed` mislead me right?

Any hint from your experience before I close this issue? :)

closed 2018-10-05 by ultrabug