LudovicRousseau/CCID #58: Add a new variant of ACR1252U

martinpaljak, 2018-10-12:
```
 idVendor: 0x072F
  iManufacturer: ACS
 idProduct: 0x223E
  iProduct: ACR1252 Reader
 bcdDevice: 2.06 (firmware release?)
 bLength: 9
 bDescriptorType: 4
 bInterfaceNumber: 0
 bAlternateSetting: 0
 bNumEndpoints: 3
  bulk-IN, bulk-OUT and Interrupt-IN
 bInterfaceClass: 0x0B [Chip Card Interface Device Class (CCID)]
 bInterfaceSubClass: 0
 bInterfaceProtocol: 0
  bulk transfer, optional interrupt-IN (CCID)
 iInterface: ACR1252 Reader PICC
 CCID Class Descriptor
  bLength: 0x36
  bDescriptorType: 0x21
  bcdCCID: 1.00
  bMaxSlotIndex: 0x00
  bVoltageSupport: 0x07
   5.0V
   3.0V
   1.8V
  dwProtocols: 0x0000 0x0002
   T=1
  dwDefaultClock: 4.000 MHz
  dwMaximumClock: 4.000 MHz
  bNumClockSupported: 0 (will use whatever is returned)
   IFD does not support GET CLOCK FREQUENCIES request: Resource temporarily unavailable
  dwDataRate: 10752 bps
  dwMaxDataRate: 250000 bps
  bNumDataRatesSupported: 0 (will use whatever is returned)
   IFD does not support GET_DATA_RATES request: Resource temporarily unavailable
  dwMaxIFSD: 254
  dwSynchProtocols: 0x00000000
  dwMechanical: 0x00000000
   No special characteristics
  dwFeatures: 0x000404BA
   ....02 Automatic parameter configuration based on ATR data
   ....08 Automatic ICC voltage selection
   ....10 Automatic ICC clock frequency change according to parameters
   ....20 Automatic baud rate change according to frequency and Fi, Di params
   ....80 Automatic PPS made by the CCID
   ..04.. Automatic IFSD exchange as first exchange (T=1)
   04.... Short and Extended APDU level exchange
  dwMaxCCIDMessageLength: 522 bytes
  bClassGetResponse: 0x00
  bClassEnvelope: 0x00
  wLcdLayout: 0x0000
  bPINSupport: 0x00
  bMaxCCIDBusySlots: 1
```
[output.txt](https://github.com/LudovicRousseau/CCID/files/2472473/output.txt)
[image]

bought from https://www.smartcardfocus.com/shop/ilp/id~841/acr1252u/p/index.shtml

LudovicRousseau, 2018-10-12:
Is it this device https://www.acs.com.hk/en/products/342/acr1252u-usb-nfc-reader-iii-nfc-forum-certified-reader/

If yes it should already be supported by the ACS driver. I don't know if it is a good idea to _also_ add it in my CCID driver.

martinpaljak, 2018-10-12:
It should be - but I really don't know. BUT it seems to work OK (even better on RF aspects than some other regional manufacturer) and I do not have (nor would want to) add the ACS driver for a compliant reader (plug-n-play on macOS/Linux)

LudovicRousseau, 2018-10-12:
Fixed in https://github.com/LudovicRousseau/CCID/commit/5adffb317bcae37144e6fad7840d8c0970851c25

closed 2018-10-12 by LudovicRousseau