LudovicRousseau/CCID #59: Add support for Elatec TWN4 E/B1.06/CKF1.64/SGSS1.01

sguldemond, 2018-10-18:
Hi,

I'm trying to get the Elatec TWN4 E/B1.06/CKF1.64/SGSS1.01 NFC scanner working with this library.
I added the vender-id + product-id (09D8:0410) to Info.plist, this resulted in:
```
00000006 hotplug_libudev.c:311:HPAddDevice() Adding USB device: Elatec TWN4 E/B1.06/CKF1.64/SGSS1.01
00000019 readerfactory.c:936:RFInitializeReader() Attempting startup of Elatec TWN4 E/B1.06/CKF1.64/SGSS1.01 00 00 using /usr/local/lib/pcsc/drivers/ifd-ccid.bundle/Con
00000133 readerfactory.c:826:RFBindFunctions() Loading IFD Handler 3.0
00000018 ifdhandler.c:1961:init_driver() Driver version: 1.4.30
00000474 ifdhandler.c:1978:init_driver() LogLevel: 0x0003
00000007 ifdhandler.c:1989:init_driver() DriverOptions: 0x0000
00000114 ifdhandler.c:110:CreateChannelByNameOrChannel() Lun: 0, device: usb:09d8/0410:libudev:0:/dev/bus/usb/001/009
00000013 ccid_usb.c:302:OpenUSBByName() Using: /usr/local/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist
00000455 ccid_usb.c:320:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000005 ccid_usb.c:321:OpenUSBByName() ifdProductString: Generic CCID driver
00000003 ccid_usb.c:322:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later
00002253 ccid_usb.c:614:OpenUSBByName() Can't find a CCID interface on 1/9
00000131 ccid_usb.c:189:close_libusb_if_needed() libusb_exit
00000185 ccid_usb.c:799:OpenUSBByName() Device not found?
00000007 ifdhandler.c:150:CreateChannelByNameOrChannel() failed
00000003 readerfactory.c:967:RFInitializeReader() Open Port 0x200000 Failed (usb:09d8/0410:libudev:0:/dev/bus/usb/001/009)
00000003 readerfactory.c:276:RFAddReader() Elatec TWN4 E/B1.06/CKF1.64/SGSS1.01 init failed.
00000003 readerfactory.c:987:RFUnInitializeReader() Attempting shutdown of Elatec TWN4 E/B1.06/CKF1.64/SGSS1.01 00 00.
00000002 readerfactory.c:863:RFUnloadReader() Unloading reader driver.
```

I also tried to edit the source code by adding my device in `ccid.h` by mimicking the behavior of the other Elatec TWN4 devices:
```
#define ElatecTWN4_CCID_CDC	0x09D80427
#define ElatecTWN4_CCID	0x09D80428
#define ElatecTWN4_NFC	0x09D80410 /* Elatec TWN4 E/B1.06/CKF1.64/SGSS1.01 */
```

and adding the `ElatecTWN4_NFC` definition to the `ccid.c` next to the other Elatec references,
which after compiling and installing CCID resulted in the same log as before.

Is there anyway to get my device working with this library?

I also have this info about the device if that helps:
```
$ lsusb -D /dev/bus/usb/001/009
Device: ID 09d8:0410  
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x09d8 
  idProduct          0x0410 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              250mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.01
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      71
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               4
  
```

Thanks :)

LudovicRousseau, 2018-10-18:
Follow https://ccid.apdu.fr/#CCID_compliant

sguldemond, 2018-10-19:
```
$ sudo ./src/parse > output.txt
Parsing USB bus/device: 09D8:0410 (bus 1, device 3)
 idVendor:  0x09D8  iManufacturer: OEM
 idProduct: 0x0410  iProduct: RFID Device (Keyboard)
  NOT a CCID/ICCD device
```

and `output.txt` is empty, I guess it's not gonna work then? :(

LudovicRousseau, 2018-10-19:
Exact. I can't help.

closed 2018-10-19 by LudovicRousseau