LudovicRousseau/CCID #60: Fails to compile without pthread

nioncode, 2019-02-19:
I'm trying to compile CCID for an embedded device without autotools by simply creating a custom `config.h`:
```c
#define HAVE_ARPA_INET_H
#define HAVE_ERRNO_H
#define HAVE_STDINT_H
#define HAVE_STDIO_H
#define HAVE_STDLIB_H
#define HAVE_STRING_H
#define HAVE_SYS_TYPES_H
#define HAVE_UNISTD_H
#undef HAVE_PTHREAD
#define NO_LOG
#define VERSION "internal"
#define BUNDLE "unused"
#define PCSCLITE_HP_DROPDIR "unused"
```

When linking the resulting static lib to my application, it fails with `libccid.a(ccid_usb.c.o): undefined reference to symbol 'pthread_create@@GLIBC_2.1'`. In `ccid_usb.c`, pthread gets used without guarding it with `HAVE_PTHREAD` like it is done in `ifdhandler.c`.

Is there any reason why the guards are not used in `ccid_usb.c`? Are they strictly required if it is a multislot reader? If so, then there should be a new `#define` that allows to disable multi slot support so that we don't have to rely on pthread being available.

LudovicRousseau, 2019-02-21:
I think pthread is required for multi slots readers.
`./configure --disable-multi-thread` should be enough to remove dependency on pthread.

I will try to work on a patch.

LudovicRousseau, 2019-02-22:
libccid uses libusb for the USB communication. libusb itself uses pthread.

Do you have a libusb on your embedded system that does **not** use pthread?

nioncode, 2019-02-22:
Actually you are right, pthread gets pulled in by libusb anyways. I cannot reproduce the initial issue at the moment. Let me get back on this next week, then I'll run it again on the CI.

nioncode, 2019-02-26:
So, on my Ubuntu 14.04 CI system the build fails, because `pthread` is not linked. On my nixos developer machine, everything builds fine, although `pthread` is also not linked (the build command line is the same for both:
```
g++ -rdynamic example/CMakeFiles/ex.dir/ex.cpp.o -o example/ex ccid/libccidlib.a -lusb-1.0
```

I don't understand why it works on my machine without adding `-pthread`, but adding it fixes the build on the CI. Would you be open to a PR that allows to remove multislot support so that `pthread` is not needed for CCID (although it might be pulled in by libusb after all)?

LudovicRousseau, 2019-02-26:
Have you checked you can USE the `libccidlib.a` library?
Or that you can generate an complete executable and not just a `.a` lib?

Do you use `libccidlib.a` with pcscd? or something else?

LudovicRousseau, 2019-02-26:
I see you generate a `ex` binary. So ignore the first 2 questions above.

I am surprised you have a libusb without pthread. What version of libusb are you using? It is the same on your machine and for CI?

nioncode, 2019-02-27:
I use `libccidlib.a` directly in the final executable to send CCID commands / APDUs.

I'm using `libusb-1.0.so.0.1.0` on both machines. Actually, when using `readelf -d libusb-1.0.so.0.1.0`, both machines have `pthread` listed as NEEDED. Also, in the libusb pkg-config file, both have `Libs.private: -ludev  -pthread`, but only `Libs: -L${libdir} -lusb-1.0`, so both only link with usb-1.0 and not explicitly with `-pthread`.

I think that my system is at fault here, since `libccidlib.a` uses methods of `pthread` directly, so users (i.e. executable `ex`) should link against `pthread` and not rely on a dependent library to pull this dependency in. What I don't understand is why gcc does not complain on my system, but simply searches for the `pthread` methods in `libusb` transitively. This does not seem right.

nioncode, 2019-03-18:
I'm closing this, since I think the issue is somehow with my system and how gcc searches for libraries.
Sorry for the noise.

closed 2019-03-18 by nioncode