LudovicRousseau/CCID #63: Using the CCID standalone as parser on Android

D4no0, 2019-11-14:
I would like to know if it is possible to use the CCID library as a standalone library for parsing commands over USB on Android. 

From the current ports of PCSC on Android I noticed that all of them require the PCSC [daemon](https://github.com/vicamo/pcsc-lite-android), witch in most cases require either root access or even custom firmware, witch is not scalable to multiple users. I would like to avoid using the service and just generate/parse commands with different protocols and readers.

I managed to make connection to the reader natively on android, the problems are the parsing, I noticed that different readers have different initialization (for example ACS readers send some undocumented commands at the beginning).

I'd like to know if it is worth investing time in porting the library interfaces to native android or if there is a part of functionality that can be done only by the daemon(initialization of the reader), or maybe the ports can be used with slight modifications (any kind of feedback would be appreciated).

PS: the only functionality I care is basic APDU commands, like read binary, generate hash, currently I use it to dump data from tachocards.

LudovicRousseau, 2019-11-14:
Feel free to adapt the CCID driver for your needs. Your only obligation is to respect the license (GNU LGPL v2.1).
You can also have a look at the Android version of libusb.

D4no0, 2019-11-14:
yes I saw the android version of libusb, however there is no point in using
usb communication at C level since it is harder to handle than on native.
What I need to know: is the logic of initialization located in CCID library
or on daemon and what is the advantage of using daemon over straight up
communication?

LudovicRousseau, 2019-11-15:
I provide Free Software, not free support.

If you need my advices please contact me privately.

closed 2019-11-15 by LudovicRousseau