LudovicRousseau/CCID #72: Error with T1 card in CCID Driver

FeitianSmartcardReader, 2020-07-09:
Today had test with T1 card, and found the code enter to a loop and never return.

check the log with pcscd

The loop start from Line 568, and no return, could please take a look, thanks

`
0000010 APDU: 00 A4 04 00 06 11 22 33 44 55 66 
00000012 ifdhandler.c:1303:IFDHTransmitToICC() usb:096e/060d:libudev:1:/dev/bus/usb/002/033 (lun: 10000)
00000010 commands.c:2228:CmdXfrBlockTPDU_T1() T=1: 11 and 65548 bytes
00000010 openct/proto-t1.c:600:t1_build() more bit: 0
00000008 openct/proto-t1.c:210:t1_transceive() line 211****** 
00000163 sending: 00 40 0B 00 A4 04 00 06 11 22 33 44 55 66 9A 
00000010 -> 000001 6F 0F 00 00 00 00 17 00 00 00 00 40 0B 00 A4 04 00 06 11 22 33 44 55 66 9A 
00013173 <- 000001 80 04 00 00 00 00 17 00 00 00 00 82 00 82 
00000035 received: 00 82 00 82 
00000034 openct/proto-t1.c:313:t1_transceive() received: 0, expected: 1, more: 0
00000035 openct/proto-t1.c:327:t1_transceive() R-Block required
00000043 openct/proto-t1.c:210:t1_transceive() line 211****** 
00000006 sending: 00 92 00 92 
00000014 -> 000001 6F 04 00 00 00 00 18 00 00 00 00 92 00 92 
00011650 <- 000001 80 06 00 00 00 00 18 00 00 00 00 00 02 6D 00 6F 
00000036 received: 00 00 02 6D 00 6F 
00000008 openct/proto-t1.c:376:t1_transceive() T1-I-BLOCK
00000004 openct/proto-t1.c:388:t1_transceive() wrong nr
00000003 openct/proto-t1.c:210:t1_transceive() line 211****** 
00000004 sending: 00 92 00 92 
00000006 -> 000001 6F 04 00 00 00 00 19 00 00 00 00 92 00 92 
00012160 <- 000001 80 06 00 00 00 00 19 00 00 00 00 00 02 6D 00 6F 
00000033 received: 00 00 02 6D 00 6F 
00000006 openct/proto-t1.c:388:t1_transceive() wrong nr
00000004 openct/proto-t1.c:210:t1_transceive() line 211****** 
00000004 sending: 00 C0 00 C0 
00000005 -> 000001 6F 04 00 00 00 00 1A 00 00 00 00 C0 00 C0 
00011613 <- 000001 80 04 00 00 00 00 1A 00 00 00 00 E0 00 E0 

`

[pcscd.1.log](https://github.com/LudovicRousseau/CCID/files/4896526/pcscd.1.log)

FeitianSmartcardReader, 2020-07-09:
The select file command works well at first time, but after write wrong data to the card, the driver into loop and never return, had test on macOS, has same issue.

LudovicRousseau, 2020-07-12:
Is it a normal T=1 card or a test card used to check the T=1 automaton?
Can you send me the card so I can reproduce the problem?

I will have to read ISO 7816-3 again :-)

Do you have a patch to propose?

FeitianSmartcardReader, 2020-07-13:
@LudovicRousseau thanks for your reply, we using java card to simulate the test case for T0 and T1, our engineer working to check the card, if we cannot find problem in card, then we may need ship to you demo cards to have a look, I will keep you update today, thanks

FeitianSmartcardReader, 2020-07-13:
Did test with two java cards, and load same applet with same operations, after test, confirm it is card OS issue, thanks

closed 2020-07-13 by FeitianSmartcardReader