LudovicRousseau/CCID #74: Incorrect naming of Yubikey devices

bmomjian, 2020-09-06:
I have installed libccid 1.4.30-1 on Debian Buster 10.  I noticed that the capitalization of the Yubikey devices is inconsistent in the file `/etc/libccid_Info.plist`:

`Yubico Yubikey NEO OTP+CCID`
`Yubico Yubikey NEO CCID`
`Yubico Yubikey NEO U2F+CCID`
`Yubico Yubikey NEO OTP+U2F+CCID`
`Yubico YubiKey CCID`
`Yubico YubiKey OTP+CCID`
`Yubico YubiKey FIDO+CCID`
`Yubico YubiKey OTP+FIDO+CCID`

Notice that the first four entries use "Yubikey" and the last four use "YubiKey" (capital K).  Using the capital K causes the `yubico-piv-tool` application to fail to recognized the Yubico Yubikey OTP+FIDO+CCID device:

`$ yubico-piv-tool -v -a verify-pin -P 000000`
`skipping reader 'Yubico YubiKey OTP+FIDO+CCID 00 00' since it doesn't match 'Yubikey'.`
`error: no usable reader found.`
`Failed to connect to reader.`

Changing the last four entries to lower-case K fixes it.  Patch attached, but I am not sure exactly how to fix it in the libccid sources.
[libccid.txt](https://github.com/LudovicRousseau/CCID/files/5178993/libccid.txt)

LudovicRousseau, 2020-09-06:
The reader name comes from the USB descriptor.
For example see https://ccid.apdu.fr/ccid/readers/Yubico_Yubikey_4_CCID.txt
It contains:
```
 idVendor: 0x1050
  iManufacturer: Yubico
 idProduct: 0x0404
  iProduct: YubiKey CCID
```

It was the choice of Yubico to name the reader `YubiKey CCID` with a capital K.

I propose to patch the `yubico-piv-tool` program to also accept YubiKey.

An even better solution would be to identify a reader using the USB VendorID and ProductID instead of the PC/SC reader name. See https://ludovicrousseau.blogspot.com/2012/04/identifying-reader-model-part-2.html

bmomjian, 2020-09-07:
Thanks, I have created an open yubico-piv-tool issue (listed above) so we can close this one, thanks.

closed 2020-09-07 by bmomjian