LudovicRousseau/CCID #79: OMNIKEY CardMan 3121 fails with OpenPGP card

dangowrt, 2021-02-08:
When using CardMan 3121 reader with OpenPGP (the only card I got for testing) it fails when using ccid:
```
pkcs11-tool -I
P:8532; T:0x139856261707584 19:42:20.113 [opensc-pkcs11] pkcs15-din-66291.c:203:sc_pkcs15emu_din_66291_init_ex: called
Cryptoki version 2.20
Manufacturer     OpenSC Project
Library          OpenSC smartcard framework (ver 0.21)
P:8532; T:0x139856261707584 19:42:20.404 [opensc-pkcs11] pkcs15-din-66291.c:203:sc_pkcs15emu_din_66291_init_ex: called
P:8532; T:0x139856261707584 19:42:20.697 [opensc-pkcs11] pkcs15-din-66291.c:203:sc_pkcs15emu_din_66291_init_ex: called
P:8532; T:0x139856261707584 19:42:20.987 [opensc-pkcs11] pkcs15-din-66291.c:203:sc_pkcs15emu_din_66291_init_ex: called
P:8532; T:0x139856261707584 19:42:21.278 [opensc-pkcs11] pkcs15-din-66291.c:203:sc_pkcs15emu_din_66291_init_ex: called
error: PKCS11 function C_GetSlotInfo failed: rv = CKR_FUNCTION_NOT_SUPPORTED (0x54)
Aborting.
```

Also trying to use GnuPG to sign a message with the key on the card fails, but less verbose:
```
error: gpg failed to sign the data
```

When using the proprietary `ifdokccid.so` driver downloadable from https://www.hidglobal.com/drivers/19364 it all just works fine. So apparently something is still missing in the free ccid driver and in order to prevent people loosing a similar amount of time I did until I realized that I had to use the proprietary driver instead, it'd be nice to have this at least documented.

LudovicRousseau, 2021-02-08:
Please provide a log as described in https://ccid.apdu.fr/#support

dangowrt, 2021-02-08:
This is archlinux on x86_64.
ccid version 1.4.34
```
pcsc-lite version 1.9.0.
Copyright (C) 1999-2002 by David Corcoran <corcoran@musclecard.com>.
Copyright (C) 2001-2018 by Ludovic Rousseau <ludovic.rousseau@free.fr>.
Copyright (C) 2003-2004 by Damien Sauveron <sauveron@labri.fr>.
Report bugs to <pcsclite-muscle@lists.alioth.debian.org>.
Enabled features: Linux x86_64-pc-linux-gnu libsystemd serial usb libudev usbdropdir=/usr/lib/pcsc/drivers ipcdir=/run/pcscd filter configdir=/etc/reader.conf.d
```

Complete log of failing attempt to gpg --sign
```
00000000 debuglog.c:299:DebugLogSetLevel() debug level=debug
00000025 debuglog.c:320:DebugLogSetCategory() Debug options: APDU
[36m00000003[0m [139703017170880] [34mpcscdaemon.c:353:main() Force colored logs[0m
[36m00000133[0m [139703017170880] configfile.l:293:DBGetReaderListDir() Parsing conf directory: /etc/reader.conf.d
[36m00000035[0m [139703017170880] configfile.l:329:DBGetReaderListDir() Skipping non regular file: .
[36m00000005[0m [139703017170880] configfile.l:329:DBGetReaderListDir() Skipping non regular file: ..
[36m00000003[0m [139703017170880] configfile.l:369:DBGetReaderList() Parsing conf file: /etc/reader.conf.d/libccidtwin
[36m00000058[0m [139703017170880] [34mpcscdaemon.c:663:main() pcsc-lite 1.9.0 daemon ready.[0m
[36m00005368[0m [139703017170880] hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/001/001
[36m00000201[0m [139703017170880] hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/001/001
[36m00000208[0m [139703017170880] hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x8087, PID: 0x0020, path: /dev/bus/usb/001/002
[36m00000220[0m [139703017170880] hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x413C, PID: 0x2513, path: /dev/bus/usb/001/003
[36m00000241[0m [139703017170880] hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x076B, PID: 0x3021, path: /dev/bus/usb/001/009
[36m00000011[0m [139703017170880] [34mhotplug_libudev.c:440:HPAddDevice() Adding USB device: OMNIKEY AG CardMan 3121[0m
[36m00000056[0m [139703017170880] [34mreaderfactory.c:1074:RFInitializeReader() Attempting startup of OMNIKEY AG CardMan 3121 00 00 using /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so[0m
[36m00000402[0m [139703017170880] [34mreaderfactory.c:950:RFBindFunctions() Loading IFD Handler 3.0[0m
[36m00000022[0m [139703017170880] [34mifdhandler.c:2025:init_driver() Driver version: 1.4.34[0m
[36m00000736[0m [139703017170880] [34mifdhandler.c:2042:init_driver() LogLevel: 0x0003[0m
[36m00000008[0m [139703017170880] [34mifdhandler.c:2053:init_driver() DriverOptions: 0x0000[0m
[36m00000238[0m [139703017170880] [34mifdhandler.c:2066:init_driver() LogLevel from LIBCCID_ifdLogLevel: 0x000F[0m
[36m00000007[0m [139703017170880] [34mifdhandler.c:110:CreateChannelByNameOrChannel() Lun: 0, device: usb:076b/3021:libudev:0:/dev/bus/usb/001/009[0m
[36m00000007[0m [139703017170880] ccid_usb.c:237:OpenUSBByName() Reader index: 0, Device: usb:076b/3021:libudev:0:/dev/bus/usb/001/009
[36m00000010[0m [139703017170880] ccid_usb.c:269:OpenUSBByName() interface_number: 0
[36m00000002[0m [139703017170880] ccid_usb.c:270:OpenUSBByName() usb bus/device: 1/9
[36m00000002[0m [139703017170880] [34mccid_usb.c:302:OpenUSBByName() Using: /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist[0m
[36m00000763[0m [139703017170880] [34mccid_usb.c:320:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)[0m
[36m00000008[0m [139703017170880] [34mccid_usb.c:321:OpenUSBByName() ifdProductString: Generic CCID driver[0m
[36m00000003[0m [139703017170880] [34mccid_usb.c:322:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.[0m
[36m00005817[0m [139703017170880] ccid_usb.c:406:OpenUSBByName() Try device: 1/9
[36m00000013[0m [139703017170880] ccid_usb.c:416:OpenUSBByName() vid/pid : 076B/3021
[36m00000005[0m [139703017170880] ccid_usb.c:482:OpenUSBByName() Checking device: 1/9
[36m00000002[0m [139703017170880] ccid_usb.c:553:OpenUSBByName() Trying to open USB bus/device: 1/9
[36m00208073[0m [139703017170880] [34mccid_usb.c:659:OpenUSBByName() Found Vendor/Product: 076B/3021 (OMNIKEY AG CardMan 3121)[0m
[36m00000017[0m [139703017170880] [34mccid_usb.c:661:OpenUSBByName() Using USB bus/device: 1/9[0m
[36m00000006[0m [139703017170880] ccid_usb.c:1275:ControlUSB() request: 0x03
[36m00015185[0m [139703017170880] receive: 00 2A 00 00 01 54 00 00 02 A8 00 00 05 50 01 00 0B A0 02 00 16 40 05 00 08 F8 01 00 0D 48 03 00 67 32 00 00 CE 64 00 00 9C C9 00 00 39 93 01 00 73 26 03 00 E7 4C 06 00 D6 5C 02 00 10 F0 03 00 00 1C 00 00 00 38 00 00 01 70 00 00 03 E0 00 00 07 C0 01 00 0E 80 03 00 09 30 02 00 9A 21 00 00 34 43 00 00 68 86 00 00 D1 0C 01 00 A2 19 02 00 44 33 04 00 00 15 00 00 04 FC 00 00 06 A4 01 00 33 19 00 00 6B 2E 01 00 80 1F 00 00 01 3F 00 00 02 7E 00 00 06 7A 01 00 0A 76 02 00 00 0E 00 00 04 18 01 00 CD 10 00 00 80 0A 00 00 03 D2 00 00 99 0C 00 00 35 97 00 00 C0 0F 00 00 03 BD 00 00 05 3B 01 00 66 08 00 00 14 0A 00 00 29 14 00 00 52 28 00 00 A5 50 00 00 4A A1 00 00 94 42 01 00 F7 78 00 00 84 1E 00 00 09 3D 00 00 12 7A 00 00 24 F4 00 00 48 E8 01 00 90 D0 03 00 36 6E 01 00 5A 62 02 00 9F 24 00 0 [... cut ...]
[36m00000007[0m [139703000380992] ifdhandler.c:1885:IFDHICCPresence() usb:076b/3021:libudev:0:/dev/bus/usb/001/009 (lun: 0)
[36m00000008[0m [139703000380992] -> 000000 65 00 00 00 00 00 1E 00 00 00 
[36m00000523[0m [139703000380992] <- 000000 81 00 00 00 00 00 1E 01 00 01 
[36m00000019[0m [139703000380992] ifdhandler.c:2005:IFDHICCPresence() Card present
[36m00000009[0m [139703000380992] [34mifdhandler.c:317:IFDHPolling() usb:076b/3021:libudev:0:/dev/bus/usb/001/009 (lun: 0) 600000 ms[0m
[36m00000006[0m [139703000380992] ccid_usb.c:1328:InterruptRead() before (0), timeout: 600000 ms
```

dangowrt, 2021-02-08:
Just to avoid confusion: never mind that Broadcom 5800 device which is also present, it's the internal reader of the laptop which used to work and then stopped (not related to any updates, it also doesn't respond to vendor firmware update tool in DOS, so I assume the hardware is gone). I'm using the external CardMan reader which works fine with the proprietary driver.

LudovicRousseau, 2021-02-09:
The APDU "00 47 81 00 00 00 02 B8 00 02 0F" is problematic and very strange for me. The reader answer with the error "Protocol not supported".
I suggest to report the problem on the OpenSC mailing list.

dangowrt, 2021-02-09:
The odd thing here is that it does work fine when using the proprietary driver (`ifdokccid.so`) instead of ccid.
See log
```
[36m00000011[0m [140114547267136] [34mAPDU: 00 47 81 00 00 00 02 B8 00 20 00 [0m
[36m00166482[0m [140114547267136] [34mSW: 7F 49 82 02 0A 81 82 02 00 9C CD 13 B8 EA 86 04 A3 90 78 58 CB 7B 0B 9A F8 E5 1A 06 4C DF E7 59 4F 7D BC 9F CC A7 CC 60 F2 FB E7 F6 7A 04 E6 DE CA CF D6 BB CE D3 42 17 86 FC CD B7 18 93 B3 BD 26 06 C7 43 F9 28 60 05 27 19 31 03 BB A6 BE 3B FA C4 D0 23 9D 60 CA 4F 86 EA 71 C4 6F 5D DC F6 25 91 F2 97 ED 74 84 52 76 90 C5 93 2A 6F 0A EE C6 76 3D 25 0C D8 C4 6F FC 39 DE 34 A9 E9 B0 FA 86 27 B3 B1 EA C5 61 C0 FF B5 4F 8C 2C 32 61 58 4C 75 28 6F 9A CE 0C 5F F1 F5 B2 59 46 A8 7F 81 55 93 02 69 B8 24 40 99 38 3D 7A CF 53 0E 7E C8 44 B8 67 F4 56 5B CB 81 5A 3A DD 44 38 7C 37 41 1E B2 7D B0 12 01 1B 1D F1 C6 ED 9D 2B 86 0F A4 43 19 CC E1 5A 7D 6E 46 DF 3D 29 B7 95 8E 56 45 29 4E CD 7A F1 49 2A B9 67 05 B1 AF 4C 5C 75 9A D9 27 F5 6C 12 8B EE 21 A7 1C 55 AC 24 E5 9C A3 E8 99 9D 14 2F 88 1C FA 4F 9F 07 55 F6 3B 70 9E AE F9 FE 47 5D F0 C5 EE E6 D3 D7 35 32 64 10 06 9A 39 B0 35 E3 A5 17 79 F0 EB 76 87 D5 CD F2 30 F5 A4 35 F5 ED 2F 1D BF 89 E3 DA 72 23 EF 3E 33 5B A7 69 C2 28 B6 3E 62 EC 33 CD 2F 8B B3 4D 39 E5 73 52 41 0C D3 0C 49 AF 36 87 4A 9C 56 1E 49 19 FB 0A 17 DE 2B 97 57 0D 89 B1 9B 61 3B 09 D3 52 2F 13 CE F5 4B 76 61 DD EA 1D C7 93 95 E5 29 1C 3B 4D E0 C1 58 8A EF FB DD E1 FE 13 D3 06 AA 1B 38 46 12 99 59 F3 F4 08 F8 87 1F 57 27 2F 12 80 B6 B7 0B F8 F4 0B C0 7D C0 D1 D4 4C A5 61 A9 7F 82 4C C8 AB CF 80 E7 7F E2 3D FB 7B 30 58 4E F0 2E A7 41 FA 86 41 EF A3 5F D5 79 A5 7A 90 F6 47 E4 79 F6 1E 8C 3D 33 B3 D3 49 C6 3C 82 02 33 00 EE 0B C5 93 CF CF F5 9D C2 4F 2A 6A 87 76 A7 DE 3C 8C 76 AB 50 8A D0 9C E9 65 74 BA 9D B0 99 B9 E7 99 5D C6 6F D6 45 67 82 04 00 01 00 01 90 00 [0m
```

LudovicRousseau, 2021-02-09:
This is an extended APDU https://ccid.apdu.fr/ccid_extended_apdu.html

Your "OMNIKEY AG CardMan 3121" reader declares it does NOT support extended APDU but short APDU only. https://ccid.apdu.fr/ccid/readers/CardMan3121.txt

I guess the proprietary driver uses proprietary commands to switch the reader to extended APDU mode or something like that. I don't think I can fix your problem in my driver. You should use a reader with extended APDU support.

closed 2021-02-09 by LudovicRousseau

dangowrt, 2021-02-09:
The reader does support TPDU when using the proprietary pcsc driver, as mentioned above.
If I can help with USB traces and/or reverse-engineering TPDU_T1R_Handle_R_Block function, let me know.

LudovicRousseau, 2021-02-09:
I don't have time and motivation to add support of non-compliant readers.
Sorry.