LudovicRousseau/CCID #82: Eletrand SmartCard & Storage reader does not work

kurthuwig, 2021-03-18:
Archlinux with package

community/ccid 1.4.34-1 (76.1 KiB 255.9 KiB) (Installed)
    A generic USB Chip/Smart Card Interface Devices driver

lsusb -vvv

```
Bus 001 Device 038: ID 0bda:0169 Realtek Semiconductor Corp. Mass Storage Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x0169 Mass Storage Device
  bcdDevice           61.23
  iManufacturer           1 Generic
  iProduct                2 USB2.0-CRW
  iSerial                 3 20070818000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0074
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 CARD READER
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass        11 Chip/SmartCard
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              6 Smart Card Reader Interface
      ChipCard Interface Descriptor:
        bLength                54
        bDescriptorType        33
        bcdCCID              1.10  (Warning: Only accurate for version 1.0)
        nMaxSlotIndex           0
        bVoltageSupport         7  5.0V 3.0V 1.8V 
        dwProtocols             3  T=0 T=1
        dwDefaultClock       3750
        dwMaxiumumClock      7500
        bNumClockSupported      0
        dwDataRate          10080 bps
        dwMaxDataRate      312500 bps
        bNumDataRatesSupp.      0
        dwMaxIFSD             254
        dwSyncProtocols  00000000 
        dwMechanical     00000000 
        dwFeatures       00010030
          Auto clock change
          Auto baud rate change
          TPDU level exchange
        dwMaxCCIDMsgLen       271
        bClassGetResponse      00
        bClassEnvelope         00
        wlcdLayout           none
        bPINSupport             0 
        bMaxCCIDBusySlots       1
[... 132 lines cut ...]
00000003 [140191530989120] readerfactory.c:338:RFAddReader() Driver is thread safe
00000003 [140191530989120] readerfactory.c:1074:RFInitializeReader() Attempting startup of Generic USB2.0-CRW [Bulk-In, Bulk-Out, Interface] (20070818000000000) 01 00 using /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000004 [140191530989120] readerfactory.c:863:RFLoadReader() Reusing already loaded driver for /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000006 [140191530989120] readerfactory.c:950:RFBindFunctions() Loading IFD Handler 3.0
00000008 [140191530989120] ifdhandler.c:110:CreateChannelByNameOrChannel() Lun: 10000, device: usb:0bda/0169:libudev:1:/dev/bus/usb/001/038
00000004 [140191530989120] ccid_usb.c:237:OpenUSBByName() Reader index: 1, Device: usb:0bda/0169:libudev:1:/dev/bus/usb/001/038
00000006 [140191530989120] ccid_usb.c:269:OpenUSBByName() interface_number: 1
00000002 [140191530989120] ccid_usb.c:270:OpenUSBByName() usb bus/device: 1/38
00000003 [140191530989120] ccid_usb.c:302:OpenUSBByName() Using: /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist
00000493 [140191530989120] ccid_usb.c:320:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000006 [140191530989120] ccid_usb.c:321:OpenUSBByName() ifdProductString: Generic CCID driver
00000004 [140191530989120] ccid_usb.c:322:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00000084 [140191530989120] ccid_usb.c:406:OpenUSBByName() Try device: 1/38
00000005 [140191530989120] ccid_usb.c:416:OpenUSBByName() vid/pid : 0BDA/0169
00000003 [140191530989120] ccid_usb.c:482:OpenUSBByName() Checking device: 1/38
00000003 [140191530989120] ccid_usb.c:553:OpenUSBByName() Trying to open USB bus/device: 1/38
00000043 [140191530989120] ccid_usb.c:636:OpenUSBByName() Found interface 0 but expecting 1
00000004 [140191530989120] ccid_usb.c:638:OpenUSBByName() Wrong interface for USB device 1/38. Checking next one.
00000345 [140191530989120] ccid_usb.c:800:OpenUSBByName() Device not found?
00000006 [140191530989120] ifdhandler.c:155:CreateChannelByNameOrChannel() failed
00000005 [140191530989120] readerfactory.c:1105:RFInitializeReader() Open Port 0x200001 Failed (usb:0bda/0169:libudev:1:/dev/bus/usb/001/038)
00000027 [140191530989120] readerfactory.c:376:RFAddReader() Generic USB2.0-CRW [Bulk-In, Bulk-Out, Interface] (20070818000000000) init failed.
00000005 [140191530989120] readerfactory.c:610:RFRemoveReader() UnrefReader() count was: 1
00000003 [140191530989120] readerfactory.c:1125:RFUnInitializeReader() Attempting shutdown of Generic USB2.0-CRW [Bulk-In, Bulk-Out, Interface] (20070818000000000) 01 00.
```

LudovicRousseau, 2021-03-18:
https://ludovicrousseau.blogspot.com/2014/03/level-1-smart-card-support-on-gnulinux.html

kurthuwig, 2021-03-18:
Thanks for the quick response. As you can see by the pcscd logs above, it has found the reader:

    00000090 [140191530989120] ccid_usb.c:659:OpenUSBByName() Found Vendor/Product: 0BDA/0169 (Generic USB2.0-CRW)

and correctly detects 1 slot:

    00000004 [140191530989120] ifdhandler.c:484:IFDHGetCapabilities() Reader supports 1 slot(s)

but then fails:

```
00000043 [140191530989120] ccid_usb.c:636:OpenUSBByName() Found interface 0 but expecting 1
00000004 [140191530989120] ccid_usb.c:638:OpenUSBByName() Wrong interface for USB device 1/38. Checking next one.
00000345 [140191530989120] ccid_usb.c:800:OpenUSBByName() Device not found?
00000006 [140191530989120] ifdhandler.c:155:CreateChannelByNameOrChannel() failed
00000005 [140191530989120] readerfactory.c:1105:RFInitializeReader() Open Port 0x200001 Failed (usb:0bda/0169:libudev:1:/dev/bus/usb/001/038)
```

As it is not detected by pcscd, the pcsc_scan detects no reader, so the first step in the url you referenced already fails.

What else can I do to provide more information?

LudovicRousseau, 2021-03-18:
From your logs your should see a reader "Generic USB2.0-CRW [Smart Card Reader Interface] (20070818000000000) 00 00" from pcsc_scan.

kurthuwig, 2021-03-18:
Thanks again for the quick help! pcsc_scan reported no readers found. BUT after this update today

[2021-03-18T21:23:09+0100] [ALPM] upgraded pcsclite (1.9.0-1 -> 1.9.1-1)

it works. pcsc_scan finds the reader and the card. Somehow I've got a buggy version just when I got the new reader.

closed 2021-03-18 by kurthuwig