LudovicRousseau/CCID #84: Problem with Alcor Micro AU9540

Mironenko, 2021-05-24:
Hello, Ludovic!

I have stumbled upon the problem with the Alcor Micro Corp. AU9540 reader built into Lenovo Thinkpad T14 laptop. I've got a card that provides TA1=97 and supports high frequencies. I am not able to use this card on Linux, because Alcor Micro AU9540 fails to communicate with the card on the baudrate agreed upon during PPS exchange, though the baudrate is reported as supported by the reader. As far as I know the same card works fine with the same reader on Windows (presumably because of some recovery mechanism implemented in Windows driver). The same card also works fine on Linux if TA1=96 is returned by the card. There is almost no doubt the card is alright; it works fine with another reader on the baudrate corresponding to TA1=97.

**Could there be any recovery mechanism provided with either PCSC or CCID specification to be implemented in libccid so that the card could work fine with Alcor Micro Corp. AU9540?**

I hope the logs attached to the report (see https://gist.github.com/Mironenko/4798626bcf5ea12ce684d89bfb8ec1d1) will help to better understand the problem described:
* [lsusb-alcor.txt](https://gist.github.com/Mironenko/4798626bcf5ea12ce684d89bfb8ec1d1#file-lsusb-alcor-txt): Alcor Micro Corp. AU9540 device descriptor
* [pcscd-alcor-ta1-97.txt](https://gist.github.com/Mironenko/4798626bcf5ea12ce684d89bfb8ec1d1#file-pcscd-alcor-ta1-97-txt): Result of `sudo LIBCCID_ifdLogLevel=0x000F pcscd -afd` with the card returning TA1=97
* [pcscd-alcor-ta1-96.txt](https://gist.github.com/Mironenko/4798626bcf5ea12ce684d89bfb8ec1d1#file-pcscd-alcor-ta1-96-txt): Result of `sudo LIBCCID_ifdLogLevel=0x000F pcscd -afd` with the card returning TA1=96
* [pcscd-rutoken-ta1-97.txt](https://gist.github.com/Mironenko/4798626bcf5ea12ce684d89bfb8ec1d1#file-pcscd-rutoken-ta1-97-txt): Result of `sudo LIBCCID_ifdLogLevel=0x000F pcscd -afd` with the card returning TA1=97 on another reader (to prove the higher baudrate is ok for the card).

Libccid version used:

```
> git describe
1.4.34-43-g676cc1f
> git rev-parse HEAD
676cc1f27b9460cec141c96abe629f6c1bb6155f
```

Thank you for the software you develop and support.

Best regards,
Eugene Mironenko

LudovicRousseau, 2021-05-24:
In your test with the "Aktiv Rutoken SCR 3001" you can see in the logs:
```
00000004 ifdhandler.c:2101:find_baud_rate() Card baud rate: 447500
00000003 ifdhandler.c:2110:find_baud_rate() Reader can do: 9600
00000003 ifdhandler.c:2110:find_baud_rate() Reader can do: 19531
00000002 ifdhandler.c:2110:find_baud_rate() Reader can do: 26881
00000002 ifdhandler.c:2110:find_baud_rate() Reader can do: 39062
00000002 ifdhandler.c:2110:find_baud_rate() Reader can do: 53763
00000002 ifdhandler.c:2110:find_baud_rate() Reader can do: 78125
00000003 ifdhandler.c:2110:find_baud_rate() Reader can do: 156250
00000002 ifdhandler.c:2110:find_baud_rate() Reader can do: 161290
00000002 ifdhandler.c:2110:find_baud_rate() Reader can do: 312500
00000006 ifdhandler.c:2110:find_baud_rate() Reader can do: 625000
00000004 ifdhandler.c:851:IFDHSetProtocolParameters() Reader does not support 447500 bauds
```
The reader does *not* support the speed requested by the card.
In this case the driver uses the default value TA1=0x11
You can see this value in the 10th byte of the PC_to_RDR_SetParameters command:
```
00000004 commands.c:2310:SetParameters() length: 7 bytes
00000009 -> 000001 61 07 00 00 00 00 0D 01 00 00 11 10 00 4D 00 40 00 
00018965 <- 000001 82 07 00 00 00 00 0D 00 00 01 11 10 00 4D 00 40 00 
```

I agree there is a problem with your card and the Alcor Micro AU9540.
There is no existing mechanism for this problem.

What I propose is to modify the CCID driver to replace the TA1 value by 0x96 in case a card with TA1=0x97 is inserted in the problematic reader.
You can insert this code at https://github.com/LudovicRousseau/CCID/blob/master/src/ifdhandler.c#L734

Mironenko, 2021-05-24:
> The reader does not support the speed requested by the card.
In this case the driver uses the default value TA1=0x11

Thank you for highlighting this; that's the other problem, and I assume it can be fixed in the reader itself.

> What I propose is to modify the CCID driver to replace the TA1 value by 0x96 in case a card with TA1=0x97 is inserted in the problematic reader.
> You can insert this code at https://github.com/LudovicRousseau/CCID/blob/master/src/ifdhandler.c#L734

Do I get it right that you will accept such a patch to be merged into the upstream?

LudovicRousseau, 2021-05-24:
It would be an ugly patch. I am not sure to want it upstream.
I would prefer if Alcor Micro could fix the reader firmware.

dmitrmax, 2021-11-29:
@LudovicRousseau would you consider an XML file with blacklisted features for certain readers? This is not as ugly as proposed quick solution. Any big peace of software dealing with a big number of hardware devices sooner or later would have such a blacklist. Or... a blocklist if you better like the new ethic terms.

Maybe even libccid_Info.plist could be extended in a backward compatible manner?

dmitrmax, 2021-11-29:
First of all this is hardcoded list. Second, it will indeed render the card reader completly unusable although it just needs some tweaking. I'm talking about external file which can be edited by system administrator without recompiling libccid.

LudovicRousseau, 2022-05-29:
The problem is in the reader firmware.

LudovicRousseau, 2022-09-20:
It looks like the same smart card reader with the same problem is also present in HP Zbook laptop. https://salsa.debian.org/rousseau/CCID/-/issues/13

dlee100, 2022-10-02:
I hope this will be still relevant to anyone looking but the only way I got my Alcor AU9540 SmartCard Reader to work on my Thinkpad T480 (running in Fedora 36) is through the following steps:

1. Download the ccid tar file from the listed page in this repo's About section: https://ccid.apdu.fr/
2. Extract the file and cd into it: `cd ccid-1.5.0`
3. Make sure you have `pcsc-lite`, `pcsc-lite-devel`, `libusb`, and ` libusb-devel `already installed in your system.
4. Run `./configure`
5. Run `make`
6. Run `make install` as root.

Right after that, the integrated smart card reader worked like a charm on Linux.

LudovicRousseau, 2022-10-02:
Fedora 36 _already_ provides the CCID driver in version 1.5.0 https://fedora.pkgs.org/36/fedora-x86_64/pcsc-lite-ccid-1.5.0-1.fc36.x86_64.rpm.html (with some [patches](https://src.fedoraproject.org/rpms/pcsc-lite-ccid/tree/f36))

What version of pcsc-lite-ccid had you installed on your Fedora 36 (before the ccid upgrade)?

dlee100, 2022-10-02:
`pcsc-lite-ccid-1.5.0-1`
But the sc reader only became capable when I built and installed ccid from source.

LudovicRousseau, 2022-10-03:
@dlee100 it may be a different problem in your case.

Can you generate 2 pcscd trace as described in https://ccid.apdu.fr/#support showing the problem in the two cases:
- using Fedora package
- using a local rebuild of ccid

In the 2 case please do the same operations and explain what you do to check if the reader is working or not.

brandon1024, 2022-12-02:
> It looks like the same smart card reader with the same problem is also present in HP Zbook laptop.

Apologies if this isn't relevant to the current thread, but I wanted to chime in because myself and my colleagues have HP ZBook laptops with these `AU9540` readers and they have given us so much trouble over the years.

On the HP ZBook laptops, the reader works fine most of the time. However, once or twice a day our card reader becomes unresponsive and doesn't respond until we restart the pcsc daemon. When I looked into this a year ago I concluded that this was likely occurring when the reader goes into power saving, but I'll admit I'm no expert when it comes to these sort of issues and I could be wrong.

Here's what I pulled from the logs:

```
00000391 winscard_svc.c:361:ContextThread() Received command: BEGIN_TRANSACTION from client 13                                                                                                                                                                        [187/1895]
00000018 readerfactory.c:852:RFReaderInfoById() RefReader() count was: 1                                                                                                                                                                                                        
00000007 winscard.c:1082:SCardBeginTransaction() Status: 0x00000000                                                                     
00000005 winscard.c:1085:SCardBeginTransaction() UnrefReader() count was: 2                              
00000005 winscard_svc.c:571:ContextThread() BEGIN_TRANSACTION rv=0x0 for client 13                       
00000087 winscard_svc.c:361:ContextThread() Received command: TRANSMIT from client 13                            
00000021 readerfactory.c:852:RFReaderInfoById() RefReader() count was: 1                                         
00000006 winscard.c:1595:SCardTransmit() Send Protocol: T=1                                                                             
00000007 ifdhandler.c:1332:IFDHTransmitToICC() usb:058f/9540:libudev:0:/dev/bus/usb/001/003 (lun: 0)
05000300 ccid_usb.c:858:WriteUSB() write failed (1/3): -7 LIBUSB_ERROR_TIMEOUT                                   
00000051 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612                                                                        
00000013 winscard.c:1620:SCardTransmit() Card not transacted: 0x80100016                      
00000008 winscard.c:1648:SCardTransmit() UnrefReader() count was: 2                                                                     
00000012 winscard_svc.c:691:ContextThread() TRANSMIT rv=0x80100016 for client 13
00000221 winscard_svc.c:361:ContextThread() Received command: CMD_WAIT_READER_STATE_CHANGE from client 13        
00000026 winscard_svc.c:840:MSGSendReaderStates() Send reader states: 13                                 
00001212 winscard_svc.c:361:ContextThread() Received command: CMD_STOP_WAITING_READER_STATE_CHANGE from client 13
00000027 winscard_svc.c:442:ContextThread() CMD_STOP_WAITING_READER_STATE_CHANGE rv=0x0 for client 13            
00000180 winscard_svc.c:361:ContextThread() Received command: CMD_WAIT_READER_STATE_CHANGE from client 13        
00000025 winscard_svc.c:840:MSGSendReaderStates() Send reader states: 13                                 
00000184 winscard_svc.c:361:ContextThread() Received command: CMD_STOP_WAITING_READER_STATE_CHANGE from client 13
00000026 winscard_svc.c:442:ContextThread() CMD_STOP_WAITING_READER_STATE_CHANGE rv=0x0 for client 13            
00000187 winscard_svc.c:361:ContextThread() Received command: CMD_WAIT_READER_STATE_CHANGE from client 13
00000025 winscard_svc.c:840:MSGSendReaderStates() Send reader states: 13                      
00001132 winscard_svc.c:361:ContextThread() Received command: CMD_STOP_WAITING_READER_STATE_CHANGE from client 13
00000014 winscard_svc.c:442:ContextThread() CMD_STOP_WAITING_READER_STATE_CHANGE rv=0x0 for client 13                                   
00000128 winscard_svc.c:361:ContextThread() Received command: CMD_WAIT_READER_STATE_CHANGE from client 13                               
00000009 winscard_svc.c:840:MSGSendReaderStates() Send reader states: 13                                 
00000052 winscard_svc.c:361:ContextThread() Received command: CMD_STOP_WAITING_READER_STATE_CHANGE from client 13
00000011 winscard_svc.c:442:ContextThread() CMD_STOP_WAITING_READER_STATE_CHANGE rv=0x0 for client 13                                   
00000187 winscard_svc.c:361:ContextThread() Received command: END_TRANSACTION from client 13                     
00000030 readerfactory.c:852:RFReaderInfoById() RefReader() count was: 1                                 
00000010 winscard.c:1234:SCardEndTransaction() Status: 0x00000000                                                                       
00000005 winscard.c:1237:SCardEndTransaction() UnrefReader() count was: 2                                        
00000007 winscard_svc.c:587:ContextThread() END_TRANSACTION rv=0x0 for client 13                     
00000197 winscard_svc.c:361:ContextThread() Received command: BEGIN_TRANSACTION from client 13                   
00000030 readerfactory.c:852:RFReaderInfoById() RefReader() count was: 1                                 
00000008 winscard.c:1082:SCardBeginTransaction() Status: 0x00000000                                                                     
[... cut ...]
00000024 winscard_svc.c:442:ContextThread() CMD_STOP_WAITING_READER_STATE_CHANGE rv=0x0 for client 13           
00000161 winscard_svc.c:361:ContextThread() Received command: END_TRANSACTION from client 13                     
00000027 readerfactory.c:852:RFReaderInfoById() RefReader() count was: 1
00000012 winscard.c:1234:SCardEndTransaction() Status: 0x00000000
00000008 winscard.c:1237:SCardEndTransaction() UnrefReader() count was: 2
00000009 winscard_svc.c:587:ContextThread() END_TRANSACTION rv=0x0 for client 13
```

This screams reader firmware issues to me. In any case, I hope this is helpful.

LudovicRousseau, 2022-12-02:
Thanks @brandon1024 for the information.

In your case the problem may NOT be that the reader does not work fine at the speed requested by the card.
For an unknown reason the reader fails to accept a command and we get a timeout:
```
05000300 ccid_usb.c:858:WriteUSB() write failed (1/3): -7 LIBUSB_ERROR_TIMEOUT
```
After that the USB communication fails again:
```
00003032 ccid_usb.c:897:ReadUSB() read failed (1/3): -8 LIBUSB_ERROR_OVERFLOW
```

I get many issue reports about the Alcor Micro AU9540 and AU9560 smart card readers. Maybe you understand why these readers are in the "[Unsupported or partly supported CCID readers](https://ccid.apdu.fr/ccid/unsupported.html#0x058F0x9540)" list :-)

LudovicRousseau, 2022-12-17:
Please see https://ludovicrousseau.blogspot.com/2022/12/alcormicro-au9560-reader-and-fast-smart.html

zayedalsaidi, 2022-12-18:
> Please see https://ludovicrousseau.blogspot.com/2022/12/alcormicro-au9560-reader-and-fast-smart.html

Thank you. I have a card reader with `TA(1) = 97`. However, when I tried to compile and run `./configure` I faced this error message: 
`configure: error: install pcsc-lite 1.8.3 or later, or use ./configure PCSC_CFLAGS=...`

I have Kubuntu 22.04.  I can't find `pcsc-lite` package, but `pcsc_scan` is running OK.  Furthermore, I try to search the internet about any help, but I can't find any.

bnavigator, 2022-12-18:
You have to install `libpcsclite-dev`

https://github.com/LudovicRousseau/CCID/blob/6ba85f0626cb00af18be7fd17ff1c0144d373405/.github/workflows/build.yml#L34-L45

zayedalsaidi, 2022-12-18:
Thanks. I managed to comply and install the new driver. It works OK for me. Me problem was that my government issue two version of ID cards. The old version is working fine with old driver (i.e. less than 1.5.1). But the new one failed to work with that driver.  Now, with ccid-1.5.1-3ac3a1a, they both are working fine. 
Thank you.

LudovicRousseau, 2022-12-18:
Thanks for the feedback @zayedalsaidi

fsakalos, 2022-12-20:
Hello Ludovic,

I did the test you mentioned on your blog with the following parameters:

1. Machine: Apple MacBook Pro 14-inch, 2021, M1
2. OS: macOS 13.1
3. Reader: Alcor Micro AU9540 (sold as AKASA AK-CR-03WHV2)
4. Card: 3B DF 96 FF 81 B1 FE 45 1F 87 00 31 B9 64 09 37 72 13 73 84 01 E0 00 00 00 00 (Slovak National Identity Card)
5. Driver: ccid-1.5.1-3ac3a1a

Works for accessing eGoverment services and electronically signing documents. Thus the test was **successful**.

LudovicRousseau, 2022-12-21:
Thanks @fsakalos for the feedback.
What was the status **before** the patch? Were you able to use your card or not?

fsakalos, 2022-12-21:
I wasn't able to use it, when the card was inserted the eGov application started to behave extremely weird. Sometimes it didn't detect the card at all, sometimes it did detect it, but never loaded the certificates. But it never actually worked as expected before the patch, it was impossible to login to government services.

The reader worked well with the old ID card I had until the beginning of December. But the state started to issue new one since 1st Dec 22.

LudovicRousseau, 2022-12-21:
Thanks @bnavigator for the feedback.
If your coworker wants to use GNU/Linux my new driver should also work on his HP Firefly 14 G8.

bnavigator, 2022-12-21:
Not him, but me and other's might use HPs on Linux in the future. Not sure which of those have the AU9540/AU9560. Thank you for your great work!

bnavigator, 2022-12-25:
Are you planning to make the 3ac3a1a version public and release a new version?

LudovicRousseau, 2022-12-25:
Yes, the goal is to include the patches (or a modified version) in an official release.

But before that, I would like to get more feedback from users. For example in https://salsa.debian.org/rousseau/CCID/-/issues/13

LudovicRousseau, 2023-01-31:
CCID 1.5.2 is now available.
https://ludovicrousseau.blogspot.com/2023/01/new-version-of-libccid-152.html

It will be included in the next Debian version (Debian 12) and in Ubuntu 23.04. I have no visibility for the other GNU/Linux distributions.

closed 2023-01-31 by LudovicRousseau