LudovicRousseau/CCID #92: IFDStatusICC returns SCARD_S_SUCCESS after the reader is removed

andrei-datcu, 2021-12-04:
Ever since [this patch](https://github.com/LudovicRousseau/CCID/commit/b0cb37f1aee46d04c31c19273f23967f4ae8601c#diff-25a6634263c1b1f6fc4697a04e2b9904ea4b042a89af59dc93ec1f5d44848a26) `IFDStatusICC` returns `SCARD_S_SUCESS` if the read fails. That's because `IFDHICCPresence` doesn't return `IFD_NO_SUCH_DEVICE` anymore. Upon a static code inspection, this means that nobody will ever set `SCARD_UKNOWN` because the status handler thread get a success from `IFDStatusICC`

LudovicRousseau, 2021-12-04:
I can't find `SCARD_UKNOWN` in the source code.

What is the problem exactly?

andrei-datcu, 2021-12-04:
Sorry, it was a typo: `SCARD_UNKNOWN`. Edited.

Basically a waiter for SCardGetStatus will not get a result when the reader is removed

LudovicRousseau, 2021-12-04:
`SCardGetStatusChange()` instead `SCardGetStatus()`?

Have you tried to reproduce the bug with a sample code?

LudovicRousseau, 2022-05-29:
@andrei-datcu any news about this problem?

LudovicRousseau, 2022-05-31:
I guess you are right.

What is the problem at the PC/SC API level? Do you have a sample code to show this problem?

LudovicRousseau, 2022-06-18:
The code https://github.com/LudovicRousseau/PCSC/blob/210efa525f50e99821a10d91b8c861825e470d8b/src/ifdwrapper.c#L369 is _not used_ only when you use the CCID driver. Other drivers may still report `IFD_NO_SUCH_DEVICE` when the reader is removed.

Unless you can show a real PC/SC application sample code with a problem I consider this as a non-issue and close it.

closed 2022-06-18 by LudovicRousseau