LudovicRousseau/CCID #94: Mixup of readers / add some "sanity check" for readers.txt?

martinpaljak, 2022-02-02:
I have a Feitian R502 reader that suddenly became Avtors. This might explain:

https://github.com/LudovicRousseau/CCID/blob/b2927edae089987a37c6fbdd6291c316c02c4500/readers/Avtor_SC_Reader_KP382.txt#L1-L4

https://github.com/LudovicRousseau/CCID/blob/b2927edae089987a37c6fbdd6291c316c02c4500/readers/Feitian_R502-CL.txt#L1-L4

I don't know if this is some kind of "OEM" re-branding, but at least mine is definitely Feitian-branded, Feitian-sourced.

LudovicRousseau, 2022-02-02:
Hello Martin,
"many" readers are in fact rebranded.
I fixed the problem in https://github.com/LudovicRousseau/CCID/commit/07aefeb8fd2f40eb5e8c7d9fd737c0a917a22721

You can see all the duplicates in the... "duplicates" section https://github.com/LudovicRousseau/CCID/blob/master/readers/supported_readers.txt#L1010

closed 2022-02-02 by LudovicRousseau

martinpaljak, 2022-02-02:
Thanks! Of course did not "see" the dups section... Interesting sidenote: this has also sneaked into usb.ids:
`Bus 001 Device 009: ID 096e:0608 Feitian Technologies, Inc. SC Reader KP382`