LudovicRousseau/CCID #102: How to display the actual reader name instead of numbers

FeitianSmartcardReader, 2022-10-26:
@LudovicRousseau Our customer using R502 Dual interface reader on macOS/Linux, and they found the reader name shows below:
[image]

is that passible to have the reader name? 
[image]

LudovicRousseau, 2022-10-26:
The reader name is not selected by the CCID driver but by the layer above.

- On GNU/Linux it is pcsc-lite (and the PC/SC name contains the interface name). See [What is in a PC/SC reader name?](https://ludovicrousseau.blogspot.com/2010/05/what-is-in-pcsc-reader-name.html)
- On macOS it is Crypto Token Kit (CTK)

CTK does not support multi interfaces CCID readers. The CCID driver has to fake a multislot reader instead.

You should ask Apple to:

1. support composite (multi interfaces) CCID readers
2. add the interface name in the reader name

LudovicRousseau, 2022-11-07:
Only Apple can fix your problem.

closed 2022-11-07 by LudovicRousseau

FeitianSmartcardReader, 2022-11-08:
Noted with thanks