LudovicRousseau/CCID #104: How can I add an unsupported reader (OS X)

JoepdeJong, 2022-12-22:
I am trying to add a reader that is not yet supported. However, editing `/usr/libexec/SmartCardServices/drivers/ifd-ccid.bundle/Contents/Info.plist` is not allowed. On the internet, they suggest booting in recovery mode. However, this feels a bit odd to do. Is there a more easy way to add support for readers that are not supported by default on OS X? 

Out of interest: how does the "selection" procedure for the supported readers work? Can everyone just request to be listed in this repo? Or do they need to satisfy certain security criteria?

The reader I am trying to add has the following info. According to the manufacturer, it should be CCID compliant.
```
wCopy Smart Reader:
              Product ID: 0xb030
              Vendor ID: 0x0416  (Winbond Electronics Corp.)
              Version: 1.03
              Speed: Up to 12 Mb/s
              Manufacturer: NSCCN
              Location ID: 0x14110000 / 10
              Current Available (mA): 500
              Current Required (mA): 400
              Extra Operating Current (mA): 0
              Media:
                Smart Reader:
                  Capacity: 51 KB (51.200 bytes)
                  Removable Media: Yes
                  BSD Name: disk2
                  Logical Unit: 0
                  Partition Map Type: Unknown
                  S.M.A.R.T. status: Verified
                  USB Interface: 1
```

LudovicRousseau, 2022-12-23:
This reader is not yet in my list.
The first step is to follow https://ccid.apdu.fr/#CCID_compliant

LudovicRousseau, 2023-01-06:
@JoepdeJong I have no news since 2 weeks.
Without your participation I can't do help.

JoepdeJong, 2023-01-06:
> @JoepdeJong I have no news since 2 weeks. Without your participation I can't do help.

I just performed the steps, but the output is as follows:

>  idVendor:  0x0416  iManufacturer: NSCCN
> idProduct: 0xB030  iProduct: wCopy Smart Reader
>  NOT a CCID/ICCD device

So I don't think the reader can be supported, right?

LudovicRousseau, 2023-01-06:
Exact. My driver will not help.

closed 2023-01-06 by LudovicRousseau