LudovicRousseau/CCID #105: Ingenico PC/SC support w/ CPS & Carte Vitale

bugeaud, 2022-12-29:
Hello,

Ingenico smartcard readers supports multiple apps, one of such app EI96S enables the use of PC/SC thru USB to read CPS (French ID for Doctors) or Carte Vitale (French Social Security Card) starting from version 4.0. Pre 4.0 version used a proprietary API that is now beeing replaced by PC/SC.

But running on jammy with pcsc_scan 1.6.0 on an Ingenico iWL250 with EI96S, I got those log :

```
00000235 [139629657261632] hotplug_libudev.c:667:HPEstablishUSBNotifications() USB Device add
00000468 [139629657261632] hotplug_libudev.c:300:get_driver() Looking for a driver for VID: 0x0B00, PID: 0x0090, path: /dev/bus/usb/001/010
```

Thus scanning is not detecting the reader beeing emulated.

I anticipate the problem will be the same for Set 2, Baladeur 2, Vital Act 3S, Prium 4, Vehis and Move 5000 readers, which means basically most of the reader used for CPS in France :
https://www.lecteur-sesam.com/img/cms/PCSC%20Gamme%20Olaqin%20oct2021%20(2).pdf

I tried the buid procedure of ccid, but I was unable to do get 1.5.1 thru as the configure failed with libusb.h not found error (see https://pastebin.com/BLkMaUfv) even if the libusv-dev is there. Any static build of parse/ccid 1.5.1 for x86 / linux / Debian or Ubuntu somewhere to get the logs for the reader ?

There are very little informations from the manufacturer about the EI96S app, so I was not able to confirm it emulates CCID as well. That is why I was just trying out to see.

There seem to exist a lib from Ingenico libIngenicoSVReader.so which look like a possible PC/SC reader specific driver : 

```
$ nm -D libIngenicoSVReader.so 
                 U BIO_free
                 U BIO_new_mem_buf
000000000021230c B __bss_start
                 U calloc@GLIBC_2.2.5
                 U clearerr@GLIBC_2.2.5
                 w __cxa_finalize@GLIBC_2.2.5
000000000021230c D _edata
00000000002137e8 B _end
                 U ERR_func_error_string
                 U ERR_get_error
                 U ERR_lib_error_string
                 U ERR_load_BIO_strings
                 U __errno_location@GLIBC_2.2.5
                 U ERR_reason_error_string
                 U EVP_DigestUpdate
                 U EVP_DigestVerifyFinal
                 U EVP_DigestVerifyInit
                 U EVP_MD_CTX_free
                 U EVP_MD_CTX_new
                 U EVP_PKEY_free
                 U EVP_sha256
                 U exit@GLIBC_2.2.5
                 U fclose@GLIBC_2.2.5
                 U feof@GLIBC_2.2.5
                 U ferror@GLIBC_2.2.5
                 U fgetc@GLIBC_2.2.5
                 U fileno@GLIBC_2.2.5
                 U fopen@GLIBC_2.2.5
                 U fprintf@GLIBC_2.2.5
                 U fread@GLIBC_2.2.5
                 U free@GLIBC_2.2.5
                 U fwrite@GLIBC_2.2.5
                 U getc@GLIBC_2.2.5
                 U getenv@GLIBC_2.2.5
                 U gettimeofday@GLIBC_2.2.5
                 w __gmon_start__
0000000000004820 T IFDHCloseChannel
0000000000003c80 T IFDHControl
0000000000005110 T IFDHCreateChannel
0000000000004a90 T IFDHCreateChannelByName
0000000000002590 T IFDHGetCapabilities
0000000000003f40 T IFDHICCPresence
00000000000032b0 T IFDHPowerICC
0000000000002ff0 T IFDHSetCapabilities
0000000000003120 T IFDHSetProtocolParameters
00000000000037f0 T IFDHTransmitToICC
                 U isatty@GLIBC_2.2.5
                 U __isoc99_sscanf@GLIBC_2.7
                 w _ITM_deregisterTMCloneTable
                 w _ITM_registerTMCloneTable
                 U libusb_bulk_transfer
                 U libusb_claim_interface
                 U libusb_close
                 U libusb_error_name
                 U libusb_exit
                 U libusb_free_config_descriptor
                 U libusb_free_device_list
                 U libusb_get_active_config_descriptor
                 U libusb_get_bus_number
                 U libusb_get_device
                 U libusb_get_device_address
                 U libusb_get_device_descriptor
                 U libusb_get_device_list
                 U libusb_get_string_descriptor_ascii
                 U libusb_init
                 U libusb_open
                 U libusb_release_interface
                 U malloc@GLIBC_2.2.5
                 U memcpy@GLIBC_2.14
                 U memset@GLIBC_2.2.5
                 U OPENSSL_init_crypto
                 U PEM_read_bio_X509_AUX
                 U pthread_cond_destroy@GLIBC_2.3.2
                 U pthread_cond_init@GLIBC_2.3.2
                 U pthread_cond_signal@GLIBC_2.3.2
                 U pthread_cond_timedwait@GLIBC_2.3.2
                 U pthread_cond_wait@GLIBC_2.3.2
                 U pthread_create@GLIBC_2.2.5
                 U pthread_mutex_destroy@GLIBC_2.2.5
                 U pthread_mutex_init@GLIBC_2.2.5
                 U pthread_mutex_lock@GLIBC_2.2.5
                 U pthread_mutex_unlock@GLIBC_2.2.5
                 U rand@GLIBC_2.2.5
                 U realloc@GLIBC_2.2.5
                 U snprintf@GLIBC_2.2.5
                 U sprintf@GLIBC_2.2.5
                 U stderr@GLIBC_2.2.5
                 U stdin@GLIBC_2.2.5
                 U stdout@GLIBC_2.2.5
                 U strcmp@GLIBC_2.2.5
                 U strdup@GLIBC_2.2.5
                 U strerror@GLIBC_2.2.5
                 U strlen@GLIBC_2.2.5
                 U strstr@GLIBC_2.2.5
                 U strtoul@GLIBC_2.2.5
                 U syslog@GLIBC_2.2.5
                 U usleep@GLIBC_2.2.5
                 U vsnprintf@GLIBC_2.2.5
                 U X509_check_ca
                 U X509_free
                 U X509_get_pubkey
                 U X509_STORE_add_cert
                 U X509_STORE_CTX_free
                 U X509_STORE_CTX_get_error
                 U X509_STORE_CTX_init
                 U X509_STORE_CTX_new
                 U X509_STORE_free
                 U X509_STORE_new
                 U X509_verify_cert
                 U X509_verify_cert_error_string

```

Not sure if it could fit in the CCID picture ...

Rgs,
JB

LudovicRousseau, 2022-12-30:
You need to install `libusb-1.0-0-dev` Ubuntu/Debian package for libusb 1.0.x (if you use Ubuntu/Debian).
`libusb-dev` is for libusb-0.1.x.

If you use a different GNU/Linux distribution I don't know the exact package name.

bugeaud, 2022-12-30:
Tnx, I thought any libusb-dev would do the trick (I had latest 2:0.1.12-32build3 on jammy), turn out that 1.0-0 was indeed required. 

Parse returned 
```
Parsing USB bus/device: 0B00:0090 (bus 1, device 11)
 idVendor:  0x0B00  iManufacturer: INGENICO
 idProduct: 0x0090  iProduct: Lecteur SESAM VITALE
  NOT a CCID/ICCD device
```
Looking bad ...

LudovicRousseau, 2022-12-30:
Yes bad. My CCID driver will not help.

But the manufacturer provides a driver for GNU/Linux at https://www.olaqin.fr/actualite-et-support-de-votre-vitalact3s/

The driver uses (from its Info.plist)
```
<key>Copyright</key>
<string>This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.</string>
```
 so it shall be possible to get the Olaqin's driver source code.

closed 2022-12-30 by LudovicRousseau

bugeaud, 2022-12-30:
No issues. I was just trying to check if the app emulation was bringging some CCID compliance too as this driver is still not working in my setup. Yes it is GPL 2.1 indeed but sources are yet to be seen as far as I know 😉 Tnx 👍