LudovicRousseau/CCID #106: 2ce3:9563 not working, problem with config?

mzatko, 2023-01-13:
[    3.877894] usb 1-9: new full-speed USB device number 5 using xhci_hcd
[    4.044881] usb 1-9: config index 0 descriptor too short (expected 93, got 85)
[    4.044893] usb 1-9: config 1 has an invalid descriptor of length 0, skipping remainder of the config
[    4.044896] usb 1-9: config 1 has 0 interfaces, different from the descriptor's value: 1
[    4.047253] usb 1-9: New USB device found, idVendor=2ce3, idProduct=9563, bcdDevice= 1.20
[    4.047264] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    4.047269] usb 1-9: Product: EMV Smartcard Reader
[    4.047274] usb 1-9: Manufacturer: Generic

LudovicRousseau, 2023-01-13:
This device is in the "Should work but untested by me" list
https://ccid.apdu.fr/ccid/shouldwork.html#0x2CE30x9563

First step is "[Level 1 smart card support on GNU/Linux](https://ludovicrousseau.blogspot.com/2014/03/level-1-smart-card-support-on-gnulinux.html)"
Then "[How to get support](https://ccid.apdu.fr/#support)"

mzatko, 2023-01-13:
I tried to do my best so here is result:

**Level1:**
====================
pcsc_scan
Using reader plug'n play mechanism
Scanning present readers...
Waiting for the first reader... 
-------------
scriptor
Can't get readers list

**Level2:**
====================
libacsccid1/kinetic,now 1.1.8-1 amd64 [installed]
  PC/SC driver for ACS USB CCID smart card readers
libccid/kinetic,now 1.5.0-2 amd64 [installed]
  PC/SC driver for USB CCID smart card readers
-------------
/usr/sbin/pcscd --version
pcsc-lite version 1.9.9.
Copyright (C) 1999-2002 by David Corcoran <corcoran@musclecard.com>.
Copyright (C) 2001-2022 by Ludovic Rousseau <ludovic.rousseau@free.fr>.
Copyright (C) 2003-2004 by Damien Sauveron <sauveron@labri.fr>.
Report bugs to <pcsclite-muscle@lists.infradead.org>.
Enabled features: Linux x86_64-pc-linux-gnu libsystemd serial usb libudev usbdropdir=/usr/lib/pcsc/drivers ipcdir=/run/pcscd filter configdir=/etc/reader.conf.d
MAX_READERNAME: 128, PCSCLITE_MAX_READERS_CONTEXTS: 16
-------------
cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.10
DISTRIB_CODENAME=kinetic
DISTRIB_DESCRIPTION="Ubuntu 22.10"
-------------
uname -a
Linux 5.19.0-29-generic #30-Ubuntu SMP PREEMPT_DYNAMIC Wed Jan 4 12:14:09 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
-------------
Lenovo ThinkPad P16 gen1
-------------
Bus 001 Device 005: ID 2ce3:9563 Generic EMV Smartcard Reader
it is built-in so probably/hopefully Alcor Link AK9563

[log.txt](https://github.com/LudovicRousseau/CCID/files/10411934/log.txt)

LudovicRousseau, 2023-01-13:
I do not see your reader 2ce3:9563 in the pcscd log.
What is the (complete) output of `lsusb -t`?

mzatko, 2023-01-13:
lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 10000M
        |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 10000M
            |__ Port 4: Dev 4, If 0, Class=Hub, Driver=hub/4p, 10000M
                |__ Port 1: Dev 5, If 0, Class=Hub, Driver=hub/4p, 5000M
                    |__ Port 4: Dev 7, If 0, Class=Hub, Driver=hub/4p, 5000M
                |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 20000M/x2
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 3: Dev 2, If 0, Class=Vendor Specific Class, Driver=, 12M
    |__ Port 4: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 4: Dev 3, If 2, Class=Application Specific Interface, Driver=, 480M
    |__ Port 4: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 6: Dev 4, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 4: Dev 9, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 4: Dev 10, If 0, Class=Hub, Driver=hub/4p, 480M
                |__ Port 1: Dev 11, If 0, Class=Hub, Driver=hub/4p, 480M
                    |__ Port 3: Dev 15, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
                    |__ Port 1: Dev 13, If 2, Class=Audio, Driver=snd-usb-audio, 12M
                    |__ Port 1: Dev 13, If 0, Class=Audio, Driver=snd-usb-audio, 12M
                    |__ Port 1: Dev 13, If 3, Class=Human Interface Device, Driver=usbhid, 12M
                    |__ Port 1: Dev 13, If 1, Class=Audio, Driver=snd-usb-audio, 12M
                    |__ Port 4: Dev 16, If 0, Class=Hub, Driver=hub/4p, 480M
                    |__ Port 2: Dev 14, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
                    |__ Port 2: Dev 14, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
                |__ Port 4: Dev 12, If 3, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 4: Dev 12, If 1, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 4: Dev 12, If 2, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 4: Dev 12, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 2: Dev 8, If 0, Class=, Driver=, 12M
        |__ Port 2: Dev 8, If 1, Class=Vendor Specific Class, Driver=, 12M
        |__ Port 1: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 14: Dev 6, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 14: Dev 6, If 1, Class=Wireless, Driver=btusb, 12M

there is none as well... ?
i can see it only in lsusb without any switch

lsusb
Bus 004 Device 006: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 004 Device 007: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 004 Device 005: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 004 Device 004: ID 17ef:30b8 Lenovo USB3.1 Hub             
Bus 004 Device 003: ID 17ef:30b6 Lenovo USB3.1 Hub             
Bus 004 Device 002: ID 8087:0b40 Intel Corp. USB3.0 Hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 2ce3:9563 **Generic EMV Smartcard Reader**
Bus 001 Device 012: ID 17ef:30bb Lenovo ThinkPad Thunderbolt 4 Dock USB Audio
Bus 001 Device 016: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 015: ID 1bcf:0005 Sunplus Innovation Technology Inc. Optical Mouse
Bus 001 Device 014: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 013: ID 0b0e:0300 GN Netcom Jabra EVOLVE 20 MS
Bus 001 Device 011: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 010: ID 17ef:30b9 Lenovo USB2.0 Hub             
Bus 001 Device 009: ID 17ef:30b7 Lenovo USB2.0 Hub             
Bus 001 Device 008: ID 17ef:30b5 Lenovo 40B0
Bus 001 Device 007: ID 17ef:30b4 Lenovo ThinkPad Thunderbolt 4 Dock MCU Controller2
Bus 001 Device 004: ID 17ef:30ba Lenovo V1003
Bus 001 Device 003: ID 13d3:5287 IMC Networks Integrated Camera
Bus 001 Device 002: ID 06cb:0126 Synaptics, Inc. 
Bus 001 Device 006: ID 8087:0033 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

mzatko, 2023-02-02:
I am sorry about very late answer... Is it possible that Device number is changed after boot or after some firmware updates or with different external usb devices attached?
Now it shows me as device 006 (previously 005) on bus 001
```
lsusb
Bus 001 Device 006: ID 2ce3:9563 Generic EMV Smartcard Reader
```
then 

```
lsusb -s 1:6
Bus 001 Device 006: ID 2ce3:9563 Generic EMV Smartcard Reader

lsusb -d 2ce3:9563
Bus 001 Device 006: ID 2ce3:9563 Generic EMV Smartcard Reader
```
still not listed using
`lsusb -t`

mrhpearson, 2023-02-02:
Thanks @LudovicRousseau for letting me know about this issue.

I've confirmed I'm seeing the same on my P16 - I see the reader under lsusb but not under lsusb -t; and pcsc_scan doesn't see the reader.

I've created internal ticket LO-2258 to track this and asked the team to follow up why this isn't working. The smartcard reader isn't part of our certification process - but it should work.

mrhpearson, 2023-02-03:
Turns out we'd been looking at this already, just before Christmas and I'd forgotten about it.

There's a patch that _might_ fix it for you but we've not proposed upstream yet as we were looking for feedback from the vendor. 
I'm going to check it out on my system but if you're able to build a kernel and try it out that would be awesome as a test case. 
If you're not comfortable building a kernel let me know what you're running and I'll try and get you something

```
diff --git a/drivers/usb/core/quirks.c b/drivers/usb/core/quirks.c
index e457c08a855d..815973bea0b2 100644
--- a/drivers/usb/core/quirks.c
+++ b/drivers/usb/core/quirks.c
@@ -544,6 +544,9 @@ static const struct usb_device_id usb_quirk_list[] = {
 	/* Fibocom L850, aka Intel XMM 7360, Fibocom WWAN modem productive firmware */
 	{ USB_DEVICE(0x2cb7, 0x0007), .driver_info = USB_QUIRK_RESET_RESUME | USB_QUIRK_DELAY_INIT | USB_QUIRK_NO_LPM | USB_QUIRK_NO_LTM },
 
+	/* Alcor Link AK9563 SC Reader used in 2022 Lenovo ThinkPads */
+	{ USB_DEVICE(0x2ce3, 0x9563), .driver_info = USB_QUIRK_NO_LPM | USB_QUIRK_NO_LTM },
+
 	{ }  /* terminating entry must be last */
 };
```

mzatko, 2023-02-03:
@mrhpearson, I am sorry, but building kernel is beyond my skills :(

mrhpearson, 2023-02-03:
OK - let me see if I can get you a deb package to try.
Mark

mrhpearson, 2023-02-06:
Hi @mzatko - would you be able to see if these work:

https://drive.google.com/file/d/1vNxxhfE6A9cHtXJeYsc9s7Umbg6lSMJk/view?usp=share_link
https://drive.google.com/file/d/1ESTI1xrlmm6nNFqCtrut69C3BHBzldSw/view?usp=share_link

This is the headers and kernel .deb packages. It is for 5.17.15 kernel because that's what I'm running on my P16 with Ubuntu - but I think it will work on your system. Sorry it's a bit older but it will hopefully help confirm the fix works and then I can get it pushed upstream.

To install just 'sudo dpkg -i <the two deb files>'. Let me know if any problems.

On my system the device is now showing up under lsusb -t which is good. I don't seem to be able to install pcsc-tools to check that as I'm having some issues with the Ubuntu repo's (looks like they are done right now which is...weird)....but I'm hoping it will work. I'll try again later today

mzatko, 2023-02-06:
hi @mrhpearson, after I installed both packages with `sudo dpkg -i`, mu notebook is still booting to 
```
uname -a
Linux ubuntu-01 5.19.0-29-generic #30-Ubuntu SMP PREEMPT_DYNAMIC Wed Jan 4 12:14:09 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```
There are probably some additional steps needed...

mzatko, 2023-02-06:
hi @mrhpearson, after boot using the kernel it is listed both with `lsusb`
`Bus 001 Device 005: ID 2ce3:9563 Generic EMV Smartcard Reader`

and with `lsusb -t`
```
.....
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 9: Dev 5, If 0, Class=Chip/SmartCard, Driver=, 12M
.....
```
```
pcsc_scan
Using reader plug'n play mechanism
Scanning present readers...
0: Alcor Link AK9563 00 00
 
Tue Feb  7 00:33:43 2023
 Reader 0: Alcor Link AK9563 00 00
  Event number: 0
  Card state: Card removed, 
```

I was even able to authenticate using my smartcard

The patch seems to be fixing the issue!

mrhpearson, 2023-02-07:
Awesome - thanks for the testing and confirming! I'll get the patch upstream.

If you'd like to be added as a Tested-by tag to the patch to get credit for your help let me know (along with an email address I should use for that). If you're not sure what this is - check https://www.kernel.org/doc/html/v4.17/process/submitting-patches.html (section 13)

mrhpearson, 2023-02-09:
Patch pushed upstream and (after a couple of stupid mistakes on my side) awaiting review. It was accepted for the stable backports so it should make it's way into the distro's fairly quickly via the normal processes.

mzatko, 2023-02-09:
Thank you @mrhpearson, great to read it! I am closing the issue here.

closed 2023-02-09 by mzatko

AeyDay, 2023-11-14:
I have corrected version with fixed side-by-side config error.

How to swipe dumps using x2 emv software full pack method ATM cashout CARDPEEK, JCOP, X2A, ATR TOOL, ARQC GEN, BP TOOLS, MRSX & EMV FOUNDRY [LESS THAN A COUPLE DAYS ASSURED CASHOUT] PLEASE PM on Telegram @AnnahAey