LudovicRousseau/CCID #107: Installation

Cartagines, 2023-02-03:
Hello fellas!

I'm trying to find a driver for  a  rt-scr1 for Arch, at the moment this looks like my last hope.
I was looking into the installation instructions, but in this file you are talking about a file that i can not find and im not aware of how can i make this package. 

i was looking into the Actions part looking for an artifact or something but i found nothing.

Am I being and idiot or there is missing information?

Sorry if creating a issue for this is not correct.

Looking forward for your response and thanks! Amazing job guys!

LudovicRousseau, 2023-02-03:
Maybe you could be more specific and tell me exactly what instructions you try to follow and what is wrong.
I can't guess your problem.

Cartagines, 2023-02-03:
I'm trying just to install the driver. I was reading the installation file I found this:

`Installation from source:
"""""""""""""""""""""""""

get the ccid-x.y.z.tar.gz archive and do:

$ tar xzvf ccid-x.y.z.tar.gz
$ cd ccid-x.y.z
$ ./configure
$ make
$ sudo make install`

I dont know which file is it. If i need to generate it or what.

I'm sorry if this is clear in another part of the proyect but i couldnt find it

LudovicRousseau, 2023-02-03:
I updated the documentation.
Does it work for you now?

closed 2023-02-03 by LudovicRousseau

Cartagines, 2023-02-04:
Yes. I got the file now. I will try to install it and test it.  Thanks!