LudovicRousseau/CCID #112: MacOS Ventura M2 linking problems (1.5.2 + latest readers)

asterbini, 2023-06-25:
I am trying to build current version on my M1 mac Ventura.
(because you are now supporting the GHI reader)

I have unzipped the ccid-1.5.2 compressed source and copied the drivers from here to there.
After configuration (./MacOS/configure), when I make I get this error:

make
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in readers
make[2]: Nothing to be done for `all'.
Making all in examples
make[2]: Nothing to be done for `all'.
Making all in contrib
Making all in Kobil_mIDentity_switch
make[3]: Nothing to be done for `all'.
Making all in RSA_SecurID
make[3]: Nothing to be done for `all'.
make[3]: Nothing to be done for `all-am'.
Making all in src
  CCLD     parse
Undefined symbols for architecture arm64:
  "_gzclose", referenced from:
      _main in parse-parse.o
  "_gzfwrite", referenced from:
      _main in parse-parse.o
  "_gzopen", referenced from:
      _main in parse-parse.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [parse] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

Is there any way to solve this?
Thanks

LudovicRousseau, 2023-06-25:
It looks like your zlib library is not available for arm64.

I guess you are not using version 1.5.2 but the version from git. Exact?

LudovicRousseau, 2023-06-26:
I was able to reproduce your problem.
Solution:
```
./bootstrap
./MacOSX/configure
make
```

closed 2023-06-26 by LudovicRousseau