LudovicRousseau/CCID #121: DISCONNECT scenario for threads with pcscd

rubygoldbergmachineinc, 2023-11-05:
Scenario:
On Tap Screen, stop system service pcscd.service which will restart on its own (via socket), then enter wrong password, wait for screen timeout, Checking for NFC Reader screen will appear

Logs (with comments)
--Fresh start of system, Two thread available in logs, one active at time (combination of ESTABLISH_CONTEXT at one and RELEASE_CONTEXT for other)

--after stopping pcscd once, pcscd recovers, still see two active threads

Nov  4 22:14:11 device11 pcscd[23232322]: 00000007 winscard.c:258:SCardConnect() Attempting Connect to Identiv uTrust 3700 F CL Reader [uTrust 3700 F CL Reader] (55022243215264) 00 00 using protocol: 3
Nov  4 22:14:11 device11 pcscd[23232322]: 00000029 readerfactory.c:821:RFReaderInfo() RefReader() count was: 1
Nov  4 22:14:11 device11 pcscd[23232322]: 00000015 winscard.c:312:SCardConnect() Card Not Inserted
Nov  4 22:14:11 device11 pcscd[23232322]: 00000012 winscard.c:518:SCardConnect() UnrefReader() count was: 2
Nov  4 22:14:11 device11 pcscd[23232322]: 00000013 winscard_svc.c:513:ContextThread() CONNECT rv=0x8010000C for client 15
Nov  4 22:14:11 device11 pcscd[23232322]: 00000020 winscard_svc.c:1063:MSGCleanupClient() Freeing SCONTEXT @0x564a177d0190
Nov  4 22:14:12 device11 pcscd[23232322]: 01001108 winscard_msg_srv.c:256:ProcessEventsServer() Common channel packet arrival
Nov  4 22:14:12 device11 pcscd[23232322]: 00000037 winscard_msg_srv.c:267:ProcessEventsServer() ProcessCommonChannelRequest detects: 14
Nov  4 22:14:12 device11 pcscd[23232322]: 00000007 pcscdaemon.c:133:SVCServiceRunLoop() A new context thread creation is requested: 14
Nov  4 22:14:12 device11 pcscd[23232322]: 00000290 winscard_svc.c:340:ContextThread() Authorized PC/SC client
Nov  4 22:14:12 device11 pcscd[23232322]: 00000010 winscard_svc.c:343:ContextThread() Thread is started: dwClientID=14, threadContext @0x564a177d0190
Nov  4 22:14:12 device11 pcscd[23232322]: 00000014 winscard_svc.c:361:ContextThread() Received command: CMD_VERSION from client 14
Nov  4 22:14:12 device11 pcscd[23232322]: 00000008 winscard_svc.c:373:ContextThread() Client is protocol version 4:4
Nov  4 22:14:12 device11 pcscd[23232322]: 00000006 winscard_svc.c:396:ContextThread() CMD_VERSION rv=0x0 for client 14
Nov  4 22:14:12 device11 pcscd[23232322]: 00000053 winscard_svc.c:361:ContextThread() Received command: ESTABLISH_CONTEXT from client 14
Nov  4 22:14:12 device11 pcscd[23232322]: 00000013 winscard.c:215:SCardEstablishContext() Establishing Context: 0x575A5A4E
Nov  4 22:14:12 device11 pcscd[23232322]: 00000004 winscard_svc.c:461:ContextThread() ESTABLISH_CONTEXT rv=0x0 for client 14
Nov  4 22:14:12 device11 pcscd[23232322]: 00000097 winscard_svc.c:361:ContextThread() Received command: RELEASE_CONTEXT from client 15
Nov  4 22:14:12 device11 pcscd[23232322]: 00000011 winscard.c:229:SCardReleaseContext() Releasing Context: 0x4C08EC5
Nov  4 22:14:12 device11 pcscd[23232322]: 00000005 winscard_svc.c:476:ContextThread() RELEASE_CONTEXT rv=0x0 for client 15
Nov  4 22:14:12 device11 pcscd[23232322]: 00000035 winscard_svc.c:354:ContextThread() Client die: 15
Nov  4 22:14:12 device11 pcscd[23232322]: 00000023 winscard_svc.c:1055:MSGCleanupClient() Thread is stopping: dwClientID=15, threadContext @0x564a1778f370
Nov  4 22:14:12 device11 pcscd[23232322]: 00000007 winscard_svc.c:1063:MSGCleanupClient() Freeing SCONTEXT @0x564a1778f370
Nov  4 22:14:12 device11 pcscd[23232322]: 00000053 winscard_svc.c:361:ContextThread() Received command: CONNECT from client 14
Nov  4 22:14:12 device11 pcscd[23232322]: 00000033 winscard_svc.c:499:ContextThread() Authorized client for 'Identiv uTrust 3700 F CL Reader [uTrust 3700 F CL Reader] (55022243215264) 00 00'
Nov  4 22:14:12 device11 pcscd[23232322]: 00000006 winscard.c:258:SCardConnect() Attempting Connect to Identiv uTrust 3700 F CL Reader [uTrust 3700 F CL Reader] (55022243215264) 00 00 using protocol: 3
Nov  4 22:14:12 device11 pcscd[23232322]: 00000005 readerfactory.c:821:RFReaderInfo() RefReader() count was: 1
Nov  4 22:14:12 device11 pcscd[23232322]: 00000005 winscard.c:312:SCardConnect() Card Not Inserted
Nov  4 22:14:12 device11 pcscd[23232322]: 00000005 winscard.c:518:SCardConnect() UnrefReader() count was: 2
Nov  4 22:14:12 device11 pcscd[23232322]: 00000005 winscard_svc.c:513:ContextThread() CONNECT rv=0x8010000C for client 14
Nov  4 22:14:13 device11 pcscd[23232322]: 01001656 winscard_msg_srv.c:256:ProcessEventsServer() Common channel packet arrival
Nov  4 22:14:13 device11 pcscd[23232322]: 00000046 winscard_msg_srv.c:267:ProcessEventsServer() ProcessCommonChannelRequest detects: 15
Nov  4 22:14:13 device11 pcscd[23232322]: 00000009 pcscdaemon.c:133:SVCServiceRunLoop() A new context thread creation is requested: 15
Nov  4 22:14:13 device11 pcscd[23232322]: 00000237 winscard_svc.c:340:ContextThread() Authorized PC/SC client
Nov  4 22:14:13 device11 pcscd[23232322]: 00000010 winscard_svc.c:343:ContextThread() Thread is started: dwClientID=15, threadContext @0x564a

--I go to enter password screen,  only one thread is waiting for transmit, one is definitely gone due to action related daemon stop in previous step

-Nov  4 22:14:15 device11 pcscd[23232322]: 00000004 ifdhandler.c:1302:IFDHTransmitToICC() usb:04e6/5790:libudev:0:/dev/bus/usb/001/002lun: 0)
Nov  4 22:14:15 device11 pcscd[23232322]: 00003051 winscard.c:1640:SCardTransmit() UnrefReader() count was: 2
Nov  4 22:14:15 device11 pcscd[23232322]: 00000011 winscard_svc.c:685:ContextThread() TRANSMIT rv=0x0 for client 15
Nov  4 22:14:15 device11 pcscd[23232322]: 00000379 winscard_svc.c:361:ContextThread() Received command: TRANSMIT from client 15
Nov  4 22:14:15 device11 pcscd[23232322]: 00000016 readerfactory.c:848:RFReaderInfoById() RefReader() count was: 1
Nov  4 22:14:15 device11 pcscd[23232322]: 00000004 winscard.c:1595:SCardTransmit() Send Protocol: T=1
Nov  4 22:14:15 device11 pcscd[23232322]: 00000004 ifdhandler.c:1302:IFDHTransmitToICC() usb:04e6/5790:libudev:0:/dev/bus/usb/001/002 (lun: 0)
Nov  4 22:14:15 device11 pcscd[23232322]: 00003265 winscard.c:1640:SCardTransmit() UnrefReader() count was: 2
Nov  4 22:14:15 device11 pcscd[23232322]: 00000016 winscard_svc.c:685:ContextThread() TRANSMIT rv=0x0 for client 15
Nov  4 22:14:15 device11 pcscd[23232322]: 00000542 winscard_svc.c:361:ContextThread() Received command: TRANSMIT from client 15
Nov  4 22:14:15 device11 pcscd[23232322]: 00000135 readerfactory.c:848:RFReaderInfoById() RefReader() count was: 1
Nov  4 22:14:15 device11 pcscd[23232322]: 00000012 winscard.c:1595:SCardTransmit() Send Protocol: T=1
Nov  4 22:14:15 device11 pcscd[23232322]: 00000015 ifdhandler.c:1302:IFDHTransmitToICC() usb:04e6/5790:libudev:0:/dev/bus/usb/001/002 (lun: 0)
Nov  4 22:14:15 device11 pcscd[23232322]: 00002710 winscard.c:1640:SCardTransmit() UnrefReader() count was: 2
Nov  4 22:14:15 device11 pcscd[23232322]: 00000020 winscard_svc.c:685:ContextThread() TRANSMIT rv=0x0 for client 15
Nov  4 22:14:15 device11 pcscd[23232322]: 00000292 winscard_svc.c:361:ContextThread() Received command: TRANSMIT from client 15
Nov  4 22:14:15 device11 pcscd[23232322]: 00000018 readerfactory.c:848:RFReaderInfoById() RefReader() count was: 1
Nov  4 22:14:15 device11 pcscd[23232322]: 00000004 winscard.c:1595:SCardTransmit() Send Protocol: T=1
Nov  4 22:14:15 device11 pcscd[23232322]: 00000004 ifdhandler.c:1302:IFDHTransmitToICC() usb:04e6/5790:libudev:0:/dev/bus/usb/001/002 (lun: 0)
Nov  4 22:14:15 device11 pcscd[23232322]: 00009696 winscard.c:1640:SCardTransmit() UnrefReader() count was: 2
Nov  4 22:14:15 device11 pcscd[23232322]: 00000015 winscard_svc.c:685:ContextThread() TRANSMIT rv=0x0 for client 15
Nov  4 22:14:15 device11 pcscd[23232322]: 00000566 winscard_svc.c:361:ContextThread() Received command: TRANSMIT from client 15
Nov  4 22:14:15 device11 pcscd[23232322]: 00000018 readerfactory.c:848:RFReaderInfoById() RefReader() count was: 1
Nov  4 22:14:15 device11 pcscd[23232322]: 00000004 winscard.c:1595:SCardTransmit() Send Protocol: T=1
Nov  4 22:14:15 device11 pcscd[23232322]: 00000005 ifdhandler.c:1302:IFDHTransmitToICC() usb:04e6/5790:libudev:0:/dev/bus/usb/001/002 (lun: 0)

--as soon as I come back from password failure back to tap screen, above Thread  receives DISCONNECT, no thread is left in pool, logs stop, 
application screen goes to Checking Reader or frozen

Nov  4 22:14:23 device11 pcscd[23232322]: 06641311 winscard_svc.c:361:ContextThread() Received command: DISCONNECT from client 15
[... cut ...]
Nov  4 22:14:24 device11 pcscd[23232322]: 00000079 winscard_svc.c:361:ContextThread() Received command: RELEASE_CONTEXT from client 15
Nov  4 22:14:24 device11 pcscd[23232322]: 00000008 winscard.c:229:SCardReleaseContext() Releasing Context: 0x3A330449
Nov  4 22:14:24 device11 pcscd[23232322]: 00000004 winscard_svc.c:476:ContextThread() RELEASE_CONTEXT rv=0x0 for client 15
Nov  4 22:14:24 device11 pcscd[23232322]: 00000025 winscard_svc.c:354:ContextThread() Client die: 15
Nov  4 22:14:24 device11 pcscd[23232322]: 00000016 winscard_svc.c:1055:MSGCleanupClient() Thread is stopping: dwClientID=15, threadContext @0x564a1778f370

rubygoldbergmachineinc, 2023-11-05:
Please suggest references to understand thread model for pcscd.

LudovicRousseau, 2023-11-05:
Please describe what you think is the bug here.
What results do you expect instead of what you get?

LudovicRousseau, 2023-11-24:
No feedback.
Closing

closed 2023-11-24 by LudovicRousseau