LudovicRousseau/CCID #122: ccid and architectures other than intel (powerpc32/64). Switch from HID to CCID mode

sergione78, 2023-11-05:
Hello and forgive my lack of technicality. I'll explain the problem briefly.
At work I have an old powermac G5, on which I installed debian sid 12. Everything works great except my smartcard.
I would like to point out that I have installed all the software from synaptic, which is also present for the powerpc platform, and in fact the smartcard (ak910 usb stick) is seen in lsusb as AK910 Hkey (2021:0002).
Unfortunately it's not what I need to work. I need the key in CCID mode (2021:0001), which happens automatically under Windows, while not under Linux. And in fact, in Linux (Intel architectures) I have to manually open a terminal and run a little program (in my case it's convert.64.exe) which carries out some operations and switches the stick to CCID mode (2021:0001).
Below I report the procedure performed on an x64 machine with Ubuntu 22.04 LTS.

`sergio@sergio-X540NA:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 004: ID 13d3:5a01 IMC Networks USB2.0 VGA UVC WebCam
Bus 001 Device 003: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 008: ID 2021:0002 AK910 HKey
Bus 001 Device 007: ID 048d:1167 Integrated Technology Express, Inc. USB Mass Storage Device
Bus 001 Device 006: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 002: ID 10c4:8108 Silicon Labs USB OPTICAL MOUSE
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
sergio@sergio-X540NA:~$ cd /media/sergio/ARUBAKEY/L
Launcher/ LOST.DIR/ 
sergio@sergio-X540NA:~$ cd /media/sergio/ARUBAKEY/ArubaKeyLinux/Main_lnx/drivers/
sergio@sergio-X540NA:/media/sergio/ARUBAKEY/ArubaKeyLinux/Main_lnx/drivers$ ls
convert.32.exe      convert.64.exe      convert-ak.exe  start_hid
convert.32.exe.svc  convert.64.exe.svc  inst_hid.bat
sergio@sergio-X540NA:/media/sergio/ARUBAKEY/ArubaKeyLinux/Main_lnx/drivers$ ./convert-ak.exe 
+ dirname ./convert-ak.exe
+ curdir=.
+ cd .
+ uname -m
+ echo 64
+ BITS=64
+ ./convert.64.exe -c
/bin/mkdir: impossibile creare la directory "/home/sergio/.hsvc_1000/": File già esistente
STARTUP: Closed open FD 3
STARTUP: Closed open FD 4
console log enabled
STARTUP: arguments:
0: '/home/sergio/.hsvc_1000/convert.64.exe.svc'
1: '-p'
2: '/tmp/pcsch/pcsc_hid_socket'
STARTUP: grace time: 120
Initial random sleep...

connectSocket(): connect socket error: 2
STARTUP: no service was running on '/tmp/pcsch/pcsc_hid_socket' 
DAEMONIZING pid=15353...
Reader: 'AK910 HKey 00'
waiting PCSC reader 10000 ms...
.....
PCSC readers changed!
converted readers: 1
sergio@sergio-X540NA:/media/sergio/ARUBAKEY/ArubaKeyLinux/Main_lnx/drivers$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 004: ID 13d3:5a01 IMC Networks USB2.0 VGA UVC WebCam
Bus 001 Device 003: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 009: ID 2021:0001 AK910 CKey
Bus 001 Device 007: ID 048d:1167 Integrated Technology Express, Inc. USB Mass Storage Device
Bus 001 Device 006: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 002: ID 10c4:8108 Silicon Labs USB OPTICAL MOUSE
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
sergio@sergio-X540NA:/media/sergio/ARUBAKEY/ArubaKeyLinux/Main_lnx/drivers$ 
`
Now, the million dollar question is: is it possible to do the same thing without having to run that proprietary software that only works with Intel processors?
Before posting here I searched everywhere on the net.
ah, I forgot! If I run pcsc_scan on the G5 it doesn't find any readers, exactly like here if I don't follow the procedure described above.
I hope you can help me
Sergio

sergione78, 2023-11-05:
I apologize in advance if I don't use code tags correctly. I'm a noob.

LudovicRousseau, 2023-11-05:
You have to use an equivalent of `convert-ak.exe` on your powermac G5.
Either you find the source code of `convert-ak.exe` or you reverse engineer the program.
Maybe the program is simple enough and does just send one or two commands to the device to switch it. You can use [wireshark](https://wiki.wireshark.org/CaptureSetup/USB) to spy/capture the USB bus.

Or maybe it is possible to permanently switch the device to CCID mode.

sergione78, 2023-11-06:
ok, I tried to follow your suggestion. Since there is no equivalent of convert-ak for the ppc64 architecture I tried using wireshark software to sniff USB traffic. Well, it seems to have registered some activity, but for me it's still rocket science. Maybe you can interpret what's happening?

sergione78, 2023-11-06:
the only clue I can give about the convert.64.exe file is that it is an executable written in Qt. I don't have the skills needed to reverse engineer it.

LudovicRousseau, 2023-11-07:
I can help you. Contact me by email. See https://blog.apdu.fr/articles/about_me/

LudovicRousseau, 2023-11-12:
@sergione78 I did not received an email from you.
Closing now.

closed 2023-11-12 by LudovicRousseau