LudovicRousseau/CCID #123: Support of Lenovo CARDREADER CS21 Smartcardreader-Highstar for specific smart card

Mato-Z, 2023-11-17:
Hello,

I have interesting problem. Laptop Lenovo ThinkPad P14s Gen4 model 21HF001ACK with Ubuntu 22.04 LTS, kernel 6.2.0-36-generic, libccid version: 1.5.0-2  has integrated smart card reader `CARDREADER CS21 Smartcardreader-Highstar`. I need use this reader for my work with several smart cards. One of them does not work correctly. Pcsc detect card, but it can not read information form card correctly - `Card state: Card inserted, Unresponsive card.` This smart card is Slovak eID card. 
```
Possibly identified card (using smartcard_list.txt):
3B DF 96 FF 81 B1 FE 45 1F 87 00 31 B9 64 09 37 72 13 73 84 01 E0 00 00 00 00
[34m	National Identity Card of Slovakia (eID) (eID)[0m
[34m	https://en.wikipedia.org/wiki/Slovak_identity_card[0m
```

If I use other external smart card reader all works OK. Other smart cards work ok with integrated smart card reader too... 

Supported smart card readers from official national authority for Slovak eID: 
```
Gemalto IDBridge CT30
Bit4id Minilector EVO
ACS ACR39U
HID Omnikey 3021
HID Omnikey 3121   
```

My question is, where is the problem? Has some problem integrated card reader, or driver or...?

Ubuntu 22.04 see integrated smart card reader like: Alcor Micro AU9540 01 00 CCID is used. 

Official requirements for card readers, which can work with Slovak eID are:
```
Supported standards: ISO 7816 Class A, B, C, FCC Class B part 15, CE, VCCI, USB-IF, Microsoft WHQL
SC interface: PC/SC (2.01)
Supported protocols: T=0, T=1
Dimensions of smart card: ID-1
Transmissions rate: with PC: 12Mbps (USB 2.0), with smart card: to 344Kbps
Supported OS: Windows 11, Windows 10, Windows 8, Windows 7, Windows Vista, Windows XP, MacOS, Linux 32 bit / 64 bit
Power: USB
```
Unfortunately technical parameters of CARDREADER CS21 Smartcardreader-Highstar not available

Is there any possibility for use of this type of smart card with integrated smart card reader and with CCID with some way please?

Thank you.

LudovicRousseau, 2023-11-17:
Please follow https://ccid.apdu.fr/#CCID_compliant

Mato-Z, 2023-11-17:
@LudovicRousseau thanks a lot for your quick reply. So relevant output:

```
Parsing USB bus/device: 058F:9540 (bus 1, device 5)
 idVendor:  0x058F  iManufacturer: Generic
 idProduct: 0x9540  Can't get iProduct string: LIBUSB_ERROR_IO
  Found a CCID/ICCD device at interface 0
  Can't get iManufacturer string: LIBUSB_ERROR_IO
----------------------------------------------------------------

 idVendor: 0x058F
 idProduct: 0x9540
  iProduct: EMV Smartcard Reader
 bcdDevice: 1.20 (firmware release?)
 bLength: 9
 bDescriptorType: 4
 bInterfaceNumber: 0
 bAlternateSetting: 0
 bNumEndpoints: 3
  bulk-IN, bulk-OUT and Interrupt-IN
 bInterfaceClass: 0x0B [Chip Card Interface Device Class (CCID)]
 bInterfaceSubClass: 0
 bInterfaceProtocol: 0
  bulk transfer, optional interrupt-IN (CCID)
 Can't get iInterface string: LIBUSB_ERROR_INVALID_PARAM
 CCID Class Descriptor
  bLength: 0x36
  bDescriptorType: 0x21
  bcdCCID: 1.10
  bMaxSlotIndex: 0x00
  bVoltageSupport: 0x07
   5.0V
   3.0V
   1.8V
  dwProtocols: 0x0000 0x0003
   T=0
   T=1
  dwDefaultClock: 3.700 MHz
  dwMaximumClock: 12.000 MHz
  bNumClockSupported: 3
   Support 3700 kHz
   Support 4800 kHz
   Support 12000 kHz
  dwDataRate: 9946 bps
  dwMaxDataRate: 688172 bps
  bNumDataRatesSupported: 138
   Support 1807 bps
   Support 1989 bps
   Support 2409 bps
   Support 2487 bps
   Support 3315 bps
   Support 3613 bps
   Support 3978 bps
   Support 4818 bps
   Support 4973 bps
   Support 6631 bps
   Support 7227 bps
   Support 7957 bps
   Support 9635 bps
   Support 9946 bps
   Support 13262 bps
   Support 14453 bps
   Support 15914 bps
   Support 19271 bps
   Support 19892 bps
   Support 21680 bps
   Support 23871 bps
   Support 26523 bps
   Support 28906 bps
   Support 29839 bps
   Support 31828 bps
   Support 36133 bps
   Support 38542 bps
   Support 39785 bps
   Support 43359 bps
   Support 48177 bps
   Support 49731 bps
   Support 53047 bps
   Support 57813 bps
   Support 59677 bps
   Support 63656 bps
   Support 66308 bps
   Support 72266 bps
   Support 77083 bps
   Support 79570 bps
   Support 86719 bps
   Support 96354 bps
   Support 99462 bps
   Support 106093 bps
   Support 115625 bps
   Support 119355 bps
   Support 132616 bps
   Support 144531 bps
   Support 154167 bps
   Support 159140 bps
   Support 198925 bps
   Support 212186 bps
   Support 231250 bps
   Support 318280 bps
   Support 2344 bps
   Support 2581 bps
   Support 3125 bps
   Support 3226 bps
   Support 4301 bps
   Support 4688 bps
   Support 5161 bps
   Support 5859 bps
   Support 6250 bps
   Support 6452 bps
   Support 7813 bps
   Support 8065 bps
   Support 8602 bps
   Support 9375 bps
   Support 10323 bps
   Support 10753 bps
   Support 11719 bps
   Support 12500 bps
   Support 12903 bps
   Support 15625 bps
   Support 16129 bps
   Support 17204 bps
   Support 18750 bps
   Support 20645 bps
   Support 21505 bps
   Support 23438 bps
   Support 25000 bps
   Support 25806 bps
   Support 28125 bps
   Support 30968 bps
   Support 31250 bps
   Support 32258 bps
   Support 34409 bps
   Support 37500 bps
   Support 38710 bps
   Support 41290 bps
   Support 43011 bps
   Support 46875 bps
   Support 50000 bps
   Support 51613 bps
   Support 56250 bps
   Support 62500 bps
   Support 64516 bps
   Support 68817 bps
   Support 70313 bps
   Support 75000 bps
   Support 77419 bps
   Support 82581 bps
   Support 86022 bps
   Support 93750 bps
   Support 96774 bps
   Support 100000 bps
   Support 103226 bps
   Support 112500 bps
   Support 117188 bps
   Support 125000 bps
   Support 127312 bps
   Support 129032 bps
   Support 137634 bps
   Support 150000 bps
   Support 156250 bps
   Support 161290 bps
   Support 165161 bps
   Support 172043 bps
   Support 187500 bps
   Support 200000 bps
   Support 206452 bps
   Support 215054 bps
   Support 250000 bps
   Support 258065 bps
   Support 275269 bps
   Support 300000 bps
   Support 308333 bps
   Support 344086 bps
   Support 375000 bps
   Support 400000 bps
   Support 412903 bps
   Support 424373 bps
   Support 462500 bps
   Support 500000 bps
   Support 516129 bps
   Support 550538 bps
   Support 600000 bps
   Support 636559 bps
   Support 688172 bps
  dwMaxIFSD: 254
  dwSynchProtocols: 0x00000007
   2-wire protocol
   3-wire protocol
   I2C protocol
  dwMechanical: 0x00000000
   No special characteristics
  dwFeatures: 0x000404BE
   ....02 Automatic parameter configuration based on ATR data
   ....04 Automatic activation of ICC on inserting
   ....08 Automatic ICC voltage selection
   ....10 Automatic ICC clock frequency change according to parameters
   ....20 Automatic baud rate change according to frequency and Fi, Di params
   ....80 Automatic PPS made by the CCID
   ..04.. Automatic IFSD exchange as first exchange (T=1)
   04.... Short and Extended APDU level exchange
  dwMaxCCIDMessageLength: 272 bytes
  bClassGetResponse: 0xFF
   echoes the APDU class
  bClassEnvelope: 0xFF
   echoes the APDU class
  wLcdLayout: 0x0000
  bPINSupport: 0x00
  bMaxCCIDBusySlots: 1
```

LudovicRousseau, 2023-11-17:
You have the [AlcorMicro AU9560](https://ccid.apdu.fr/ccid/readers/AlcorMicro_AU9560.txt) reader.
I fixed a bug with this reader in CCID version 1.5.2 https://blog.apdu.fr/posts/2023/01/new-version-of-libccid-152/
Try to update your [libccid](https://packages.ubuntu.com/search?keywords=libccid) package.

Mato-Z, 2023-11-17:
@LudovicRousseau great, it works. Many thanks.

closed 2023-11-18 by Mato-Z