LudovicRousseau/CCID #129: ./configure syntax error

JTrotta, 2024-03-13:
Got this error:

`./configure: line 13541: syntax error near unexpected token `newline'
./configure: line 13541: `      AX_PTHREAD('
`

LudovicRousseau, 2024-03-13:
Install the dependencies and try again
https://ccid.apdu.fr/#git

closed 2024-03-13 by LudovicRousseau

JTrotta, 2024-03-13:
Compiled! thank you.
now : "Failed adding USB device: HID Global veriCLASS Reader"
ID 076b:5400 OmniKey AG veriCLASS Reader

'00000926 [1934619584] hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x05E3, PID: 0x0610, path: /dev/bus/usb/001/015
10235157 [1934619584] hotplug_libudev.c:655:HPEstablishUSBNotifications() USB Device add
00001014 [1934619584] hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x1AC2, PID: 0x0134, path: /dev/bus/usb/001/016
00000419 [1934619584] hotplug_libudev.c:655:HPEstablishUSBNotifications() USB Device add
00000906 [1934619584] hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x1AC2, PID: 0x0134, path: /dev/bus/usb/001/016
02541859 [1934619584] hotplug_libudev.c:655:HPEstablishUSBNotifications() USB Device add
00000516 [1934619584] hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x076B, PID: 0x5400, path: /dev/bus/usb/001/017
00000057 [1934619584] hotplug_libudev.c:440:HPAddDevice() Adding USB device: HID Global veriCLASS Reader
00000187 [1934619584] readerfactory.c:1079:RFInitializeReader() Attempting startup of HID Global veriCLASS Reader (00010100534241373131014B45311557) 00 00 using /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000796 [1934619584] readerfactory.c:954:RFBindFunctions() Loading IFD Handler 3.0
00000118 [1934619584] ifdhandler.c:2111:init_driver() Driver version: 1.5.4
00004505 [1934619584] ifdhandler.c:2133:init_driver() LogLevel: 0x0003
00000057 [1934619584] ifdhandler.c:2144:init_driver() DriverOptions: 0x0000
00001230 [1934619584] ifdhandler.c:111:CreateChannelByNameOrChannel() Lun: 0, device: usb:076b/5400:libudev:0:/dev/bus/usb/001/017
00000097 [1934619584] ccid_usb.c:327:OpenUSBByName() Using: /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist
00004805 [1934619584] ccid_usb.c:345:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000053 [1934619584] ccid_usb.c:346:OpenUSBByName() ifdProductString: Generic CCID driver
00000023 [1934619584] ccid_usb.c:347:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00020903 [1934619584] ccid_usb.c:699:OpenUSBByName() Found Vendor/Product: 076B/5400 (HID Global veriCLASS Reader)
00000038 [1934619584] ccid_usb.c:701:OpenUSBByName() Using USB bus/device: 1/17
00000022 [1934619584] ccid_usb.c:774:OpenUSBByName() bNumDataRatesSupported is 0
00112724 [1934619584] ifdhandler.c:389:IFDHGetCapabilities() tag: 0xFB3, usb:076b/5400:libudev:0:/dev/bus/usb/001/017 (lun: 0)
00000044 [1934619584] readerfactory.c:400:RFAddReader() Using the reader polling thread
00002917 [1934619584] ifdhandler.c:389:IFDHGetCapabilities() tag: 0xFAE, usb:076b/5400:libudev:0:/dev/bus/usb/001/017 (lun: 0)
00000028 [1934619584] ifdhandler.c:496:IFDHGetCapabilities() Reader supports 2 slot(s)
00005692 [1934619584] ifdhandler.c:389:IFDHGetCapabilities() tag: 0xFAC, usb:076b/5400:libudev:0:/dev/bus/usb/001/017 (lun: 0)
00000033 [1934619584] readerfactory.c:1079:RFInitializeReader() Attempting startup of HID Global veriCLASS Reader (00010100534241373131014B45311557) 00 01 using /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000017 [1934619584] readerfactory.c:867:RFLoadReader() Reusing already loaded driver for /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000041 [1934619584] readerfactory.c:954:RFBindFunctions() Loading IFD Handler 3.0
00000067 [1934619584] ifdhandler.c:111:CreateChannelByNameOrChannel() Lun: 1, device: usb:076b/5400:libudev:0:/dev/bus/usb/001/017
00000043 [1934619584] ccid_usb.c:327:OpenUSBByName() Using: /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist
00004518 [1934619584] ccid_usb.c:345:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000020 [1934619584] ccid_usb.c:346:OpenUSBByName() ifdProductString: Generic CCID driver
00000016 [1934619584] ccid_usb.c:347:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00000398 [1934619584] ccid_usb.c:571:OpenUSBByName() Opening slot: 1
00119294 [1934619584] ifdhandler.c:389:IFDHGetCapabilities() tag: 0xFB3, usb:076b/5400:libudev:0:/dev/bus/usb/001/017 (lun: 1)
00000073 [1934619584] readerfactory.c:567:RFAddReader() Using the reader polling thread
00002273 [1934619584] hotplug_libudev.c:655:HPEstablishUSBNotifications() USB Device add
00000604 [1934619584] hotplug_libudev.c:299:get_driver() Looking for a driver for VID: 0x076B, PID: 0x5400, path: /dev/bus/usb/001/017
00000073 [1934619584] hotplug_libudev.c:440:HPAddDevice() Adding USB device: HID Global veriCLASS Reader
00002032 [1934619584] ifdhandler.c:389:IFDHGetCapabilities() tag: 0xFAF, usb:076b/5400:libudev:0:/dev/bus/usb/001/017 (lun: 0)
00000066 [1934619584] ifdhandler.c:389:IFDHGetCapabilities() tag: 0xFAF, usb:076b/5400:libudev:0:/dev/bus/usb/001/017 (lun: 1)
00000041 [1934619584] readerfactory.c:787:RFSetReaderName() Driver /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so does not support more than 1 reader(s). Maybe the driver should support TAG_IFD_SIMULTANEOUS_ACCESS
00000041 [1934619584] hotplug_libudev.c:526:HPAddDevice() Failed adding USB device: HID Global veriCLASS Reader'

LudovicRousseau, 2024-03-13:
It looks like the test for pthread fails.
It is also very strange that you rebuild the driver from source code but the driver version is 1.5.4 while the latest version is 1.5.5.

JTrotta, 2024-03-13:
My fault. I downloaded the old 1.5.4.
The last 1.5.5 works!