LudovicRousseau/CCID #135: libnfc not able to connect to device on macOS

ajmas, 2024-04-29:
I am wanting to use my NFC reader via a node.js project, but I am running into an issue with libnfc based tools:

```
LIBNFC_LOG_LEVEL=3 nfc-list -v                                                                                        ajmas@ghostwalker-echo
debug	libnfc.config	key: [allow_intrusive_scan], value: [yes]
debug	libnfc.config	Unable to open directory: /usr/local/Cellar/libnfc/1.8.0/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc 1.8.0
debug	libnfc.general	0 device(s) found using arygon driver
debug	libnfc.general	0 device(s) found using pn532_uart driver
debug	libnfc.general	0 device(s) found using ACR122S driver
debug	libnfc.general	0 device(s) found using acr122_usb driver
debug	libnfc.driver.acr122_pcsc	PCSC device [Circle CIR315(1)] is not NFC capable or not supported by libnfc.
debug	libnfc.general	0 device(s) found using acr122_pcsc driver
debug	libnfc.general	1 device(s) found using pcsc driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.pcsc	Attempt to open Circle CIR315(1)
debug	libnfc.driver.pcsc	PCSC connect failed
debug	libnfc.general	Unable to open "pcsc:Circle CIR315(1)".
nfc-list: ERROR: Unable to open NFC device: pcsc:Circle CIR315(1)
```

I am using the forked version of the project [here](https://www.abcircle.co.jp/en/product/4/CIR315A/contactless-smart-card-reader/#), but I'm hoping this is close enough to the original CCID that I can get some pointers here? I am a newbie here, so I don't totally know whether libnfc is using CCID as a system driver or a bound library?

I did try "NFC Tools for Desktop" which did recognise the reader.

BTW is there any interest in merging the CIR315A support into the main project?

OS: macOS 14.4.1

LudovicRousseau, 2024-04-29:
Ask AB Circle for help if you use their driver.

ajmas, 2024-04-30:
I have now uninstalled the vendor driver, and  [built and installed ](https://github.com/LudovicRousseau/CCID/issues/136) from source, though I am still having issues. 

The [device](https://store.gototags.com/gototags-cir315a-nfc-usb-reader/) [should be supported](https://ccid.apdu.fr/ccid/shouldwork.html), but I am still getting the same error output as above. 

The version of libnfc I have is 1.8.0, which appears to be the [latest](https://github.com/nfc-tools/libnfc/releases).

Output of `sudo ./src/parse` gives:

```
idVendor: 0x31AA
  iManufacturer: Circle
 idProduct: 0x3001
  iProduct: CIR315
 bcdDevice: 2.01 (firmware release?)
 bLength: 9
 bDescriptorType: 4
 bInterfaceNumber: 0
 bAlternateSetting: 0
 bNumEndpoints: 3
  bulk-IN, bulk-OUT and Interrupt-IN
 bInterfaceClass: 0x0B [Chip Card Interface Device Class (CCID)]
 bInterfaceSubClass: 0
 bInterfaceProtocol: 0
  bulk transfer, optional interrupt-IN (CCID)
 iInterface: CIR315 PICC
 CCID Class Descriptor
  bLength: 0x36
  bDescriptorType: 0x21
  bcdCCID: 1.10
  bMaxSlotIndex: 0x00
  bVoltageSupport: 0x07
   5.0V
   3.0V
   1.8V
  dwProtocols: 0x0000 0x0003
   T=0
   T=1
  dwDefaultClock: 4.800 MHz
  dwMaximumClock: 4.800 MHz
  bNumClockSupported: 0 (will use whatever is returned)
   IFD does not support GET CLOCK FREQUENCIES request: LIBUSB_ERROR_PIPE
  dwDataRate: 12903 bps
  dwMaxDataRate: 412903 bps
  bNumDataRatesSupported: 0 (will use whatever is returned)
   IFD does not support GET_DATA_RATES request: LIBUSB_ERROR_PIPE
  dwMaxIFSD: 512
  dwSynchProtocols: 0x00000000
  dwMechanical: 0x00000000
   No special characteristics
  dwFeatures: 0x000404BA
   ....02 Automatic parameter configuration based on ATR data
   ....08 Automatic ICC voltage selection
   ....10 Automatic ICC clock frequency change according to parameters
   ....20 Automatic baud rate change according to frequency and Fi, Di params
   ....80 Automatic PPS made by the CCID
   ..04.. Automatic IFSD exchange as first exchange (T=1)
   04.... Short and Extended APDU level exchange
  dwMaxCCIDMessageLength: 522 bytes
  bClassGetResponse: 0x00
  bClassEnvelope: 0x00
  wLcdLayout: 0x0000
  bPINSupport: 0x00
  bMaxCCIDBusySlots: 1
 idVendor: 0x31AA
  iManufacturer: Circle
 idProduct: 0x3001
  iProduct: CIR315
 bcdDevice: 2.01 (firmware release?)
 bLength: 9
 bDescriptorType: 4
 bInterfaceNumber: 1
 bAlternateSetting: 0
 bNumEndpoints: 3
  bulk-IN, bulk-OUT and Interrupt-IN
 bInterfaceClass: 0x0B [Chip Card Interface Device Class (CCID)]
 bInterfaceSubClass: 0
 bInterfaceProtocol: 0
  bulk transfer, optional interrupt-IN (CCID)
 iInterface: CIR315 SAM
 CCID Class Descriptor
  bLength: 0x36
  bDescriptorType: 0x21
  bcdCCID: 1.10
  bMaxSlotIndex: 0x00
  bVoltageSupport: 0x01
   5.0V
  dwProtocols: 0x0000 0x0003
   T=0
   T=1
  dwDefaultClock: 0.250 MHz
  dwMaximumClock: 0.250 MHz
  bNumClockSupported: 0 (will use whatever is returned)
   IFD does not support GET CLOCK FREQUENCIES request: LIBUSB_ERROR_PIPE
  dwDataRate: 10752 bps
  dwMaxDataRate: 129032 bps
  bNumDataRatesSupported: 0 (will use whatever is returned)
   IFD does not support GET_DATA_RATES request: LIBUSB_ERROR_PIPE
  dwMaxIFSD: 254
  dwSynchProtocols: 0x00000000
  dwMechanical: 0x00000000
   No special characteristics
  dwFeatures: 0x000200BA
   ....02 Automatic parameter configuration based on ATR data
   ....08 Automatic ICC voltage selection
   ....10 Automatic ICC clock frequency change according to parameters
   ....20 Automatic baud rate change according to frequency and Fi, Di params
   ....80 Automatic PPS made by the CCID
   02.... Short APDU level exchange
  dwMaxCCIDMessageLength: 271 bytes
  bClassGetResponse: 0x00
  bClassEnvelope: 0x00
  wLcdLayout: 0x0000
  bPINSupport: 0x00
  bMaxCCIDBusySlots: 1
```

LudovicRousseau, 2024-04-30:
To check a reader works fine follow https://blog.apdu.fr/posts/2014/03/level-1-smart-card-support-on-mac-os-x/

LudovicRousseau, 2024-04-30:
So the reader and driver work fine.

If you have an issue with libnfc you should ask for help there.

LudovicRousseau, 2024-06-16:
No problem in CCID driver.
Closing.

closed 2024-06-16 by LudovicRousseau