LudovicRousseau/CCID #141: MacOSX installation not working on macOS Sonoma 14.5

danielmorozoff, 2024-06-15:
# Attempted Commands
```
 ./MacOSX/configure
```
yields `error: ./MacOSX/configure: line 130: ./configure: No such file or directory`

Code in `./configure` @ line 130

```
set -x
./configure \
	CFLAGS="$CFLAGS" \
	PCSC_CFLAGS="$PCSC_CFLAGS" \
	PCSC_LIBS="$PCSC_LIBS" \
	LIBUSB_CFLAGS="$LIBUSB_CFLAGS" \
	LIBUSB_LIBS="$LIBUSB_LIBS" \
	LDFLAGS="$LDFLAGS" \
	--enable-usbdropdir="$DROPDIR" \
	$CONFIGURE_ARGS \
	"$@"
```

Seems  `./configure`  is the culprit.

LudovicRousseau, 2024-06-16:
libccid uses meson now. See https://blog.apdu.fr/posts/2024/06/libccid-now-uses-meson-build-tool/
`configure` is no more generated and distributed.

See also https://ccid.apdu.fr/#MacOSX

LudovicRousseau, 2024-08-12:
You can also regenerate the `configure` script yourself using the `./bootstrap`  script.

LudovicRousseau, 2024-10-17:
No feedback.
Closing

closed 2024-10-17 by LudovicRousseau