LudovicRousseau/CCID #142: Card init fails

danielkucera, 2024-06-17:
this is the error when I run pcscd as system service:
```
jún 17 09:49:26 silverhorse pcscd[1353810]: 01403464 commands.c:1571:CCID_Receive Procedure byte conflict
jún 17 09:49:26 silverhorse pcscd[1353810]: 00000023 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612
jún 17 09:49:26 silverhorse pcscd[1353810]: 00000004 winscard.c:1618:SCardTransmit() Card not transacted: 0x80100016
```
debug log is attached
[log.txt](https://github.com/user-attachments/files/15872281/log.txt)

the card works normally with `Gemalto PC Twin Reader (74558F51) 00 00` but doesn't work with `Alcor Micro AU9540 00 00`

danielkucera, 2024-06-17:
The chip markings are AU9580

[image]
[image]

LudovicRousseau, 2024-06-17:
Install version CCID 1.5.2 or more and try again.
See https://blog.apdu.fr/posts/2023/01/new-version-of-libccid-152/

danielkucera, 2024-06-17:
Thank you very much, that worked!

closed 2024-06-17 by danielkucera